fwupd/fwupd #4839: Version numbers are inconsistent

kroeckx, 2022-07-17:
I currently get to see:
```
 fwupdmgr get-updates
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVLB1T0HBLR-000L7
11EVCTO1WW
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Summary:            UEFI ESRT device
  │   Current version:    65607
  │   Minimum Version:    65607
  │   Vendor:             Lenovo (DMI:LENOVO)
  │   Update State:       Success
  │   GUIDs:              5deafc1a-69b6-4a95-8572-2c888cabf533
  │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts:
        New version:      65615
        Remote ID:        lvfs
        Release ID:       13217
        Summary:          System Firmware for ThinkCentre M90ts_ThinkCentre M80ts_ThinkCentre M70ts
        License:          Proprietary
        Size:             16.9 MB
        Created:          2022-04-29
        Urgency:          Low
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        This stable release fixes the following issues:
        
        • Enhancement to address security vulnerability.
      
root@discovery:~# fwupdmgr upgrade
Devices with no available firmware updates: 
 • UEFI Device Firmware
 • UEFI dbx
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • MZVLB1T0HBLR-000L7
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 65607 to 65615?                                 ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This stable release fixes the following issues:                              ║
║                                                                              ║
║ • Enhancement to address security vulnerability.                             ║
║                                                                              ║
║ 11EVCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
```

But sometimes the version numbers are also shown differently. 65607 is then shown as 1.71 and 1.79. It's unclear why it sometimes shows it in a different format.

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.18.0-2-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.1
runtime   org.freedesktop.gusb          0.3.10
```

I'm using Debian testing.

This seems to be a regression from a previous version, where it showed 1.71 and 1.79 in more cases. I think version 1.7.7-1 still showed 1.71 in more cases, but also sometimes showed it as 65607.

hughsie, 2022-07-17:
Odd indeed! Can you grab me the output of `sudo fwupdtool get-devices --plugins uefi_capsule -v` please.

kroeckx, 2022-07-17:
```
11:41:50:0241 FuDebug              Verbose debugging enabled (on console 1)
11:41:50:0241 FuMain               locked /run/lock/fwupdtool
11:41:50:0243 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
11:41:50:0243 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
11:41:50:0243 FuCommon             reading /etc/fwupd/daemon.conf with 2080 bytes
11:41:50:0243 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
11:41:50:0243 FuConfig             using autodetected max archive size 18.4 EB
11:41:50:0244 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
11:41:50:0244 FuRemoteList         loading /usr/share/fwupd/metainfo
11:41:50:0244 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
11:41:50:0244 XbSilo               file: bea95e58-80fb-f9b2-fe97-256d058a9987, current:bea95e58-80fb-f9b2-fe97-256d058a9987, cached: (null)
11:41:50:0244 XbSilo               loading silo with file contents
11:41:50:0244 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
11:41:50:0244 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
11:41:50:0244 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
11:41:50:0244 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
11:41:50:0244 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
11:41:50:0244 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
11:41:50:0244 FuRemoteList         ordering lvfs-testing=lvfs+1
11:41:50:0245 FuRemoteList         ignoring unfound remote fwupd
11:41:50:0245 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:41:50:0245 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
11:41:50:0245 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
11:41:50:0245 Jcat                 loaded 1 certificates
11:41:50:0245 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
11:41:50:0245 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
11:41:50:0245 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
11:41:50:0245 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
11:41:50:0245 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
11:41:50:0245 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
11:41:50:0245 Jcat                 loaded 1 certificates
11:41:50:0245 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2459 bytes
11:41:50:0245 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
11:41:50:0245 FuEngine             client certificate now exists: nothing to do
11:41:50:0245 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
11:41:50:0245 FuHistory            got schema version of 7
11:41:50:0247 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_wacom_raw.so)
11:41:50:0247 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_rts54hid.so)
11:41:50:0248 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_fastboot.so)
11:41:50:0248 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_redfish.so)
11:41:50:0250 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dell.so)
11:41:50:0250 FuPluginDell         Using libsmbios 2.4
11:41:50:0250 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_nordic_hid.so)
11:41:50:0251 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_genesys.so)
11:41:50:0251 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_synaptics_cxaudio.so)
11:41:50:0253 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dfu.so)
11:41:50:0253 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_elantp.so)
11:41:50:0254 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_jabra.so)
11:41:50:0254 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_corsair.so)
11:41:50:0255 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_pci_bcr.so)
11:41:50:0255 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_dell_dock.so)
11:41:50:0255 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_synaptics_mst.so)
11:41:50:0256 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_vli.so)
11:41:50:0259 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_gpio.so)
11:41:50:0259 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_superio.so)
11:41:50:0261 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_ccgx.so)
11:41:50:0261 FuPlugin             load(/usr/lib/x86_64-linux-gnu/fwupd-plugins-6/libfu_plugin_parade_lspcon.so)
[... cut ...]
  Plugin:               uefi_capsule
  Protocol:             org.uefi.capsule
  Flags:                internal|updatable|require-ac|registered|needs-reboot|usable-during-update
  VendorId:             DMI:LENOVO
  Version:              458837
  VersionLowest:        458837
  VersionFormat:        number
  VersionRaw:           0x00070055
  VersionLowestRaw:     0x00070055
  Created:              2022-07-17
  UpdateState:          success

11:41:50:0355 FuPlugin             destroy(uefi_capsule)
```

kroeckx, 2022-07-17:
After an upgrade, it now showed:
```
Devices that have been updated successfully:

 • System Firmware (1.71 → 65615)
```

The report it send also contains:
```
                               "VersionOld" : "1.71",
                               "VersionNew" : "65615",
```

hughsie, 2022-07-17:
Lenovo changed the version format mid-stream. I've fixed up the firmware. Can you do `fwupdmgr refresh --force` in about 4 hours and try again please.

kroeckx, 2022-07-18:
It now shows:
├─System Firmware:
│ │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│ │   Summary:            UEFI ESRT device
│ │   Current version:    0.1.79
│ │   Minimum Version:    0.1.79
│ │   Vendor:             Lenovo (DMI:LENOVO)
│ │   Update State:       Success
│ │   GUIDs:              5deafc1a-69b6-4a95-8572-2c888cabf533
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a ← main-system-firmware
│ │   Device Flags:       • Internal device
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update

Kurt

hughsie, 2022-07-18:
Yup, that's expected. Thanks for the report.

closed 2022-07-18 by hughsie

kroeckx, 2022-07-18:
The mail with this seems to have been lost:
```
# fwupdmgr get-history
11EVCTO1WW
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   1.71
  │   Update State:       Success
  │   Last modified:      2022-07-17 11:47
  │   GUID:               5deafc1a-69b6-4a95-8572-2c888cabf533
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      65615
        Remote ID:        lvfs
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.
```

hughsie, 2022-07-18:
> failed to verify System Firmware: No stored checksums for 0.1.79

This is expected, Lenovo didn't supply them. You can store these locally using `fwupdmgr verify-update`

kroeckx, 2022-10-11:
I just noticed this:
# fwupdmgr get-history
11EVCTO1WW
│
└─System Firmware:
  │   Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
  │   Previous version:   1.71
  │   Update State:       Success
  │   Last modified:      2022-07-17 11:47
  │   GUID:               5deafc1a-69b6-4a95-8572-2c888cabf533
  │   Device Flags:       • Internal device
  │                       • Updatable
  │                       • System requires external power source
  │                       • Supported on remote server
  │                       • Needs a reboot after installation
  │                       • Reported to remote server
  │                       • Cryptographic hash verification is available
  │                       • Device is usable for the duration of the update
  │ 
  └─  New version:      65615
        Remote ID:        lvfs
        License:          Unknown
        Description:      
        The vendor did not supply any release notes.

Kurt

kroeckx, 2022-10-11:
I guess the mail made it 4 months later.