fwupd/fwupd #4843: `device_test` return false positive results for composition devices

d4s, 2022-07-18:
**Describe the bug**
In case if JSON file lists 2 or more devices, the `device-test` return false positive result even if some devices are not available and wasn't tested.

Checked wireless mouses and receivers "Corsair KATAR PRO WIRELESS" and "Steelseries Aerox 3 Wireless" with the same result.

Both mices are mixing the update steps of the receiver and mouse in JSON test file, so right after update of the receiver it is needed some time for mouse reconnection -- until that the receiver will return "disconnected" status.

**Steps to Reproduce**

```shell
# fwupdmgr device-test /home/lxdhome/d4s/work/fwupd/fwupd/data/device-tests/corsair-katar-pro-wireless.json --json
{
  "results" : [
    {
      "filename" : "/home/lxdhome/d4s/work/fwupd/fwupd/data/device-tests/corsair-katar-pro-wireless.json",
      "name" : "Corsair KATAR PRO WIRELESS",
      "interactive" : false,
      "steps" : [
        {
          "url" : "https://fwupd.org/downloads/03c478a9248c658d985267c6a67da878ab8c856e71bbd49105424d48e7fbf98c-corsair-mandarin-receiver-2.3.22.cab",
          "protocol" : "com.corsair.bp",
          "guids" : [
            "8bf1f03f-3c6f-53a1-9169-800fddf0952c"
          ],
          "version" : "2.3.22",
          "success" : true
        },
        {
          "url" : "https://fwupd.org/downloads/713ec6eace214d6646c788fdbaa24b24c1077dc52d8e9e404bf545db63940d00-corsair-mandarin-mouse-2.5.32.cab",
          "info" : "No supported devices found"
        },
        {
          "url" : "https://fwupd.org/downloads/e89d2bda6e66c1af342ffe307e313c49a16371a83a7e9a5f87dcdeadba61f4ad-corsair-mandarin-receiver-2.3.23.cab",
          "protocol" : "com.corsair.bp",
          "guids" : [
            "8bf1f03f-3c6f-53a1-9169-800fddf0952c"
          ],
          "version" : "2.3.23",
          "success" : true
        },
        {
          "url" : "https://fwupd.org/downloads/7cccf09f717ae6eb8d5d23fb089aa07f8279edc63ec1dfff9bf2f74d46e498fc-corsair-mandarin-mouse-2.5.33.cab",
          "info" : "No supported devices found"
        }
      ]
    }
  ]
}

# echo $?
0
```

For the Aerox 3 mice the error status is more interesting:
```
        {
          "url" : "https://fwupd.org/downloads/35883a1f60f9a171451ccd8ca9e3d89cfaedd99cc1325410465e30d180bcb47a-SteelSeries_Aerox_3_Wireless_Mouse_1.8.1.cab",
          "error" : "Device Aerox 3 Wireless Mouse via USB Receiver [54a3b20cf19588066894b220ec88741a4406fbe1] has no firmware version"
        },
```

**Expected behavior**

1. The error code should not be 0 (Ok) if any step has been skipped for any reason
2. In case of composite devices -- allow to bring up the parent device (receiver) after update **and** establish connection to the child device.

**fwupd version information**

Built from the recent `main` branch.

```shell
# ./fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.2
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.7
compile   com.hughsie.libjcat           0.1.8
runtime   org.kernel                    5.15.52-un-def-alt1
compile   org.freedesktop.fwupd         1.8.2
runtime   org.freedesktop.gusb          0.3.7
```

**fwupd device information**

I confirm that mentioned devices are listed and could be updated to any version separately, the issue is only in update order without delays during the test.

hughsie, 2022-07-19:
For corsair specifically, I'm a bit confused how to fix `fu_corsair_ensure_mode()` -- it's a bit different to the other plugins. Can you grab me a `fwupdtool --plugins corsair --verbose get-devices` please -- I think the subdevice is a child of the receiver. I suspect this is going to be a common problem and we want to do something like FU_DEVICE_INTERNAL_FLAG_USE_PARENT_FOR_OPEN

d4s, 2022-07-19:
The verbose output is in attach [corsair-katar-pro-wireless-20220719-1.log](https://github.com/fwupd/fwupd/files/9140268/corsair-katar-pro-wireless-20220719-1.log)

```
Intel(R) Client Systems NUC8i7HVK
│
└─KATAR PRO WIRELESS receiver:
  │   Device ID:          1f914e619b232dc4da68ec3389b98b20163003a2
  │   Current version:    2.3.23
  │   Bootloader Version: 1.0.8
  │   Vendor:             USB:0x1B1C
  │   GUIDs:              8bf1f03f-3c6f-53a1-9169-800fddf0952c ← USB\VID_1B1C&PID_1B94
  │                       d887d9d9-09c6-5852-9d27-c99b36c627bb ← USB\VID_1B1C&PID_1B94&REV_0203
  │   Device Flags:       • Updatable
  │                       • Unsigned Payload
  │ 
  └─KATAR PRO WIRELESS mouse:
        Device ID:        f0f9c1d567dfdf1b10391e7d3469e242ebd6459c
        Current version:  2.5.33
        Bootloader Version:2.0.9
        Vendor:           USB:0x1B1C
        Battery:          46% (threshold 30%)
        GUID:             7235bc76-2e5a-5312-b848-882871503152 ← USB\VID_1B1C&PID_1B94&WIRELESS
        Device Flags:     • Updatable
                          • Unsigned Payload
```

closed 2022-07-19 by superm1