fwupd/fwupd #4864: Windows v1.8.3 `fwupdtool` fails to run with `/proc/self/exe` error

cwsystem76, 2022-07-23:
**Describe the bug**
After installing the v1.8.3 version of `fwupd` on Windows, the `fwupdtool` tool does not run properly. It produces the following:

```
(process:15660): FuDebug-WARNING **: 16:26:35.143: RegCreateKeyExA failed [0]
```

and with command verbs, additionally a stream of the following error, about 150 times:

```
21:15:56:0581 FuQuirks             failed to build silo: Failed to open file "/proc/self/exe": No such file or directory
```

Command verbs like `get-devices` fail to find supported hardware like a Logitech Unified Receiver or a System76 Launch keyboard, instead resulting in the output `No detected devices` at the end of the above errors. Using a Windows `sudo` command to run as an administrator doesn't change the behavior.

**Steps to Reproduce**

* Download `https://github.com/fwupd/fwupd/releases/download/1.8.3/fwupd-1.8.3-setup-x86_64.msi` from the releases page.
* Run the installer.
* Open up PowerShell inside Windows Terminal (although the same thing happens with just `cmd.exe`).
* `cd 'C:\Program Files (x86)\fwupd\bin'` to switch to the program directory.
* `.\fwupdtool.exe get-devices`

I've attached a copy of the logs generated with the `-v` parameter below:

[fwupd windows log.txt](https://github.com/fwupd/fwupd/files/9174692/fwupd.windows.log.txt)

**Expected behavior**
I would expect `fwupdtool` to provide a list of updateable devices.

**fwupd version information**
Please provide the version of the daemon and client.

```
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe --version

(process:15660): FuDebug-WARNING **: 16:26:35.143: RegCreateKeyExA failed [0]
Loading…                 [*                                      ]21:26:35:0174 FuContext            Failed to load quirks: Failed to open file "/proc/self/exe": No such file or directory
Loading…                 [-                                      ]
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.3
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.gusb          0.3.10
```

**Additional questions**

- Operating system and version:

Windows 10 21H2

- Have you tried rebooting?

Yes.

- Is this a regression?

I believe the same thing was happening with v1.8.2 but I didn't write down what it was doing.

superm1, 2022-07-23:
Does it happen if you run with admin rights too?

cwsystem76, 2022-07-23:
Yes, I used the `sudo` from Scoop in order to test it with both normal and administrator privs, like I said above. I also just went back and opened up a traditional `cmd.exe` session as admin, and the same thing happens.

superm1, 2022-07-23:
> Yes, I used the `sudo` from Scoop in order to test it with both normal and administrator privs, like I said above. I also just went back and opened up a traditional `cmd.exe` session as admin, and the same thing happens.

Sorry I wasn't familiar with sudo from scoop on Windows.

What about if you stop the fwupd service? That help?

cwsystem76, 2022-07-24:
The Windows installer doesn't actually set up a service, so there's nothing to stop. I assumed that the service wasn't really ready for prime time and wasn't enabled by default, so I've just been testing with `fwupdtool.exe` instead.

However, running the `fwupd.exe` service executable with admin privs in one terminal window produces the same error output as `fwuptool`, and then running `fwupmgr.exe get-devices` in another terminal window while `fwupd.exe` is still running gives the same `No detected devices` result as running just `fwupdtool`. (Running `fwupmgr` without `fwupd` produces the expected "Failed to connect to daemon" error.) This appears to be the intended configuration with a service plus client (I think?) and it does the same thing.

[image]

(Left side is `sudo .\fwupd.exe`, right side is as shown.)

cwsystem76, 2022-07-24:
Just noticed that the `get-plugins` command verb on Windows shows a lot fewer plugins than on Linux. It's not loading the Logitech module, but it's loading the System76 module, so that doesn't quite explain why the Launch isn't being seen when it's connected.

cwsystem76, 2022-07-24:
I was going to grab the installer from the CI but it looks like the [build failed](https://app.circleci.com/pipelines/github/fwupd/fwupd/10784/workflows/7e34349b-4913-4271-9ce3-969a0cd44b32/jobs/38790).

cwsystem76, 2022-07-24:
Well that was quick. Have downloaded [the new MSI file from the CI system](https://output.circle-artifacts.com/output/job/0bb0cb53-e3e4-4fda-8816-2e02b3a27a56/artifacts/0/dist/setup/fwupd-1.8.4-setup-x86_64.msi) and:

```
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe

(process:9312): FuDebug-WARNING **: 19:43:11.061: RegCreateKeyExA failed [0]
Command not found

Use fwupdtool --help for help
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe --version

(process:11784): FuDebug-WARNING **: 19:43:46.253: RegCreateKeyExA failed [0]
Loading…                 [-                                      ]
WARNING: This package has not been validated, it may not work properly.
compile   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.gusb          0.3.10
runtime   org.freedesktop.fwupd         1.8.4
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.gusb          0.3.10
PS C:\Program Files (x86)\fwupd\bin> .\fwupdtool.exe get-updates

(process:9564): FuDebug-WARNING **: 19:43:13.398: RegCreateKeyExA failed [0]
Loading…                 [-                                      ]
WARNING: This package has not been validated, it may not work properly.
Devices with no available firmware updates:
 • Launch Configurable Keyboard (launch 1)
No updates available for remaining devices
```

(The `RegCreateKeyExA` error appears to be what happens when you don't run it as admin.)

So we're still not seeing the Logitech receiver, but it doesn't appear to be building/installing the module on Windows so that makes sense at least. Additionally, manually running `fwupd.exe` as admin to set it up as a service and then `fwupdmgr.exe` in another terminal window works as expected:

```
PS C:\Program Files (x86)\fwupd\bin> .\fwupdmgr.exe get-devices
WARNING: This package has not been validated, it may not work properly.
ASUS All Series
│
└─Launch Configurable Keyboard (launch 1):
      Device ID:          b43e0d166325bdb94f19c5b7971726f10d4ac65b
      Current version:    0.7.103-4011-g708ed5
      Vendor:             System76 (USB:0x3384)
      GUIDs:              b2cfb86e-dacd-5663-83e6-58059831ab6d
                          af6272df-30f2-5067-ae05-201e69cd08a5 ← USB\VID_3384&PID_0001
                          7549de75-189b-5fa6-911a-c9f5d127dc28 ← USB\VID_3384&PID_0001&REV_0001
      Device Flags:       • Updatable
                          • Unsigned Payload
```

So while `fwupd.exe` is built to work as a service, and does so when you run it manually, the MSI installer doesn't seem to be automatically be setting up the actual Windows service (i.e. the equivalent of having the executables but not the init scripts in place on a Linux machine). Might be related to this (https://stackoverflow.com/questions/1560407/windows-service-not-appearing-in-services-list-after-install?rq=1) but this probably belongs in a different issue, if it's an actual problem.

Similarly, testing to see if the firmware update actually works for the Launch on Windows will probably belong in another issue if it fails, so feel free to close this one if you want.

closed 2022-07-24 by superm1

superm1, 2022-07-28:
> MSI installer doesn't seem to be automatically be setting up the actual Windows service (i.e. the equivalent of having the executables but not the init scripts in place on a Linux machine)

Yeah this is something separate.  I don't know if it was intentional by @hughsie to not actually install the Windows service yet.  But I think it's just WIX packaging work to enable it: 
https://wixtoolset.org/documentation/manual/v3/howtos/general/install_windows_service.html

commit 1f6667fe (2022-07-24): trivial: don't use /proc/self/exe on Windows
Fixes: #4864
PR #4865 (2022-07-24): trivial: don't use /proc/self/exe on Windows
Fixes: #4864

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 4a02a154 (2022-07-24): trivial: don't use /proc/self/exe on Windows
Fixes: #4864