fwupd/fwupd #4871: libfwupd should never iterate a GMainContext it did not create itself (unless documented)

mcatanzaro, 2022-07-26:
The synchronous FwupdClient APIs are unsafe to use because FwupdClient iterates a GMainContext that libfwupd did not create. Specifically, see the `g_main_loop_run (helper->loop)` calls in fwupd-client-sync.c. Iterating a GMainContext that you do not own may cause callbacks not created by libfwupd to be unexpectedly dispatched at an unexpected time, which can cause all sorts of difficult-to-debug issues ([contrived example here](https://gitlab.gnome.org/GNOME/gnome-software/-/issues/1113#note_999813)). By default, libfwupd iterates the thread-default main context, which is especially dangerous. Only applications should do this.

Applications can avoid this issue by creating a special GMainContext just for libfwupd and calling fwupd_client_set_main_context(), but libfwupd ought to be safe by default. The normal way to do this would be to create *your own* GMainContext, push thread default, iterate it, then pop thread default. I'm not sure why that wouldn't work for libfwupd?

Anyway, I'm not sure why fwupd_client_set_main_context() needs to exist, but I suppose it would be OK if its documentation would indicate that the GMainContext will be iterated by libfwupd. Currently there is no indication that it will do this.

I previously warned about this [here](https://github.com/fwupd/fwupd/pull/2638#issuecomment-756419155), but didn't get much attention.

hughsie, 2022-07-27:
> Only applications should do this

Like fwupdmgr?

> I'm not sure why that wouldn't work for libfwupd?

Because it changes _behaviour_. Apps that expected it work with the default GMainContext expect things to not stop working.

> suppose it would be OK if its documentation would indicate that the GMainContext will be iterated by libfwupd

Can you do a PR with the changes you had in mind please?

> I previously warned about this https://github.com/fwupd/fwupd/pull/2638#issuecomment-756419155, but didn't get much attention.

I'm not sure what you were expecting. We can't just change behaviour for a library that's being used by dozens of consumers -- even if it's technically a correct change.

mcatanzaro, 2022-07-27:
> Like fwupdmgr?

Yes.

> I'm not sure why that wouldn't work for libfwupd?
> Because it changes behaviour. Apps that expected it work with the default GMainContext expect things to not stop working.

It's true that using your own GMainContext will break any applications that depend on libfwupd to iterate the default main context; however, that would be a bug in the application. Applications should not assume that a library will iterate the default main context as this would normally be unexpected.

> suppose it would be OK if its documentation would indicate that the GMainContext will be iterated by libfwupd
> Can you do a PR with the changes you had in mind please?

Maybe... wasn't planning to work on this myself, but I suppose it's probably easy enough.

> I previously warned about this https://github.com/fwupd/fwupd/pull/2638#issuecomment-756419155, but didn't get much attention.
> I'm not sure what you were expecting. We can't just change behaviour for a library that's being used by dozens of consumers -- even if it's technically a correct change.

Problem is if you _don't_ change this, libfwupd's sync APIs will cause all sorts of unpredictable bugs and security vulnerabilities in applications. If you really have real-world applications depending on the existing incorrect behavior, then an alternative would be to deprecate the existing sync APIs and try again with a brand new sync API, but that's probably overkill?

hughsie, 2022-07-27:
>> Like fwupdmgr?
> Yes.

That's reading the DBus percentage property attribute as it changes and updating a progressbar.

> security vulnerabilities in applications

I think citation needed on that one dude.

mcatanzaro, 2022-07-27:
> I'm not sure what you were expecting. We can't just change behaviour for a library that's being used by dozens of consumers -- even if it's technically a correct change.

Ah well good news: I went to investigate changing this, and it seems you already fixed this in https://github.com/fwupd/fwupd/pull/2741, two days after I complained. That's exactly what I wanted to see. Nice.

Accordingly, I'm going to close this issue. Sorry for not checking the current code first: I was staring at https://github.com/fwupd/fwupd/pull/2638 when I reported this bug and didn't think to check the latest code. My bad.

> I think citation needed on that one dude.

I don't have any concrete examples here, besides the contrived ones I already presented. I'm about to publish a blog post with a bunch of other contrived examples of use-after-free issues when callbacks are executed at the wrong time. In fairness, in this case the most likely problem would be code running too *soon* rather than code running too *late*, so use-after-free is less likely. Anyway, should be clear enough that code running when it shouldn't is often bad.