fwupd/fwupd #4874: 1.8.3: gresource_quirks=enabled is sensitive to location of source directory relative to build directory

mkszuba, 2022-07-27:
**Describe the bug**

When gresource_quirks is enabled, meson invokes the script generating the new gresource XML file with literal 'org.freedesktop.fwupd.xml' the first element of the input array. As a result, the actual argument received by the script, and subsequently injected into the new gresource XML file, is '../fwupd-1.8.3/src/org.freedesktop.fwupd.xml' (nb. from what I can see the current Git master is still affected).

Given that glib-compile-resources parses non-absolute paths in its input file as relative to that file, this effectively breaks out-of-tree builds of fwupd - org.freedesktop.fwupd.xml cannot be found. See https://bugs.gentoo.org/861521 for a build log showing this.

A fix for this is to use an explicit absolute path to org.freedesktop.fwupd.xml. Probably the simplest way of achieving this is to replicate the way absolute paths are already generated for quirk XML files: `join_paths(meson.current_source_dir(), 'org.freedesktop.fwupd.xml'),`.

**Steps to Reproduce**

1. Unpack the fwupd-1.8.3 source tarball
2. Create an out-of-tree build directory for fwupd
3. From that directory, run meson on the source dir with -Dgresource_quirks=enabled
4. Start building fwupd

**Expected behavior**

fwupd configured with -Dgresource_quirks=enabled should build successfully.

**fwupd version information**

1.8.3 source tarball from this GitHub project.

**fwupd device information**

N/A

**Additional questions**

- Operating system and version: Gentoo Linux, up-to-date ~amd64
- Have you tried rebooting? yes
- Is this a regression? no, gresource_quirks is a new feature

superm1, 2022-07-28:
Would you mind opening a PR with your proposed solution?

eli-schwartz, 2022-07-28:
Wouldn't it make more sense for the script itself to only use the basename, or else recompute the correct path using os.path.relpath?

hughsie, 2022-07-28:
> or else recompute the correct path using os.path.relpath

The files are all coming from different locations.

closed 2022-07-28 by hughsie