fwupd/fwupd #4887: Supported CPU: Failed

fhuberts, 2022-07-29:
**Describe the bug**
I'm on a Lenovo P14s Gen 2 AMD with an AMD Ryzen 7 PRO 5850U with Radeon Graphics.
When I run ```fwupdmgr security  --force``` it reports ```Supported CPU: Failed```.
Most unexpected.

It also reports ```Encrypted RAM: Unencrypted``` while I have TSME enabled in the BIOS.
I suspect this is related.

**Steps to Reproduce**
Configure the laptop with proper security options, boot Fedora 36 x64 and run ```fwupdmgr security  --force``` as root.

**Expected behavior**
Everything should be green, the system should be at HSI-4.

**fwupd version information**

```shell
# rpm -q -a |grep -i fwup
fwupd-efi-1.3-1.fc36.x86_64
fwupd-plugin-flashrom-1.8.3-1.fc36.x86_64
fwupd-plugin-modem-manager-1.8.3-1.fc36.x86_64
fwupd-plugin-uefi-capsule-data-1.8.3-1.fc36.x86_64
fwupd-1.8.3-1.fc36.x86_64
```

```shell
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.10
runtime   org.kernel                    5.18.13-200.fc36.x86_64
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
# fwupdmgr get-devices --show-all-devices
LENOVO 21A0CTO1WW
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─Integrated Camera:
│     Device ID:          301046452a49d84af6356d23e43a684b8f10660f
│     Current version:    59.18
│     Vendor:             Chicony Electronics Co.,Ltd. (USB:0x04F2)
│     Serial Number:      0001
│     GUIDs:              f8d483f9-9284-51b6-9363-7e6d59c40a97 ← USB\VID_04F2&PID_B6D0
│                         17d214a8-3e31-5f50-94b5-45bf35278459 ← USB\VID_04F2&PID_B6D0&REV_5918
│     Device Flags:       • Updatable
│   
├─Prometheus:
│ │   Device ID:          96a0be68fa5cf5d60d3d2d392797c6844ab64d33
│ │   Summary:            Fingerprint reader
│ │   Current version:    10.01.3478575
│ │   Vendor:             Synaptics (USB:0x06CB)
│ │   Install Duration:   2 seconds
│ │   Serial Number:      59441008266472
│ │   GUIDs:              8088f861-6318-5b1e-9ce4-fbddbedb09ac ← USB\VID_06CB&PID_00BD
│ │                       09e8bf16-3e69-50f5-bb66-c7a040248352 ← USB\VID_06CB&PID_00BD&REV_0000
│ │   Device Flags:       • Updatable
│ │                       • Supported on remote server
[... 174 lines cut ...]
HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✔ Pre-boot DMA protection:       Enabled
✔ Suspend-to-idle:               Enabled
✔ Suspend-to-ram:                Disabled

HSI-4
✘ Encrypted RAM:                 Unencrypted

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Encrypted
✔ fwupd plugins:                 Untainted
```

superm1, 2022-07-29:
No, the 5850U doesn't provide the needed information from the PSP.  The statement is correct.

closed 2022-07-29 by superm1

peterhoeg, 2022-08-29:
I have a U5750 but `fwupdmgr` did at least at some point earlier this month think the CPU was valid (refer to Host Security Events):

```console
HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✘ Supported CPU:                 Failed

...snip...

Host Security Events
  2022-08-04 16:47:44:  ✔ Kernel is tainted
  2022-08-04 16:47:44:  ✔ Supported CPU appeared: Valid
  2022-08-02 00:09:25:  ✘ Kernel is no longer tainted
  2022-08-02 00:09:25:  ✔ Supported CPU appeared: Valid
  2022-07-27 08:08:25:  ✔ Kernel is tainted
  2022-07-27 08:08:25:  ✔ Supported CPU appeared: Valid
  2022-07-25 06:27:59:  ✘ Kernel is no longer tainted
  2022-07-25 06:27:59:  ✔ Supported CPU appeared: Valid
  2022-06-29 14:57:31:  ✔ Kernel is tainted
  2022-06-29 14:57:31:  ✔ Supported CPU appeared: Valid
  2022-06-27 20:13:32:  ✘ Kernel is no longer tainted
  2022-06-27 20:13:32:  ✔ Supported CPU appeared: Valid
  2022-06-18 14:50:04:  ✔ Kernel is tainted
  2022-06-18 14:50:04:  ✔ Supported CPU appeared: Valid
  2022-06-17 08:49:46:  ✘ Kernel is no longer tainted
  2022-06-17 08:49:46:  ✔ Supported CPU appeared: Valid
  2022-06-02 09:07:15:  ✔ Supported CPU appeared: Valid
  2022-06-02 09:07:15:  ✘ Suspend-to-ram changed: Disabled → Enabled
```

blitz, 2022-11-28:
> No, the 5850U doesn't provide the needed information from the PSP. The statement is correct.

Do you know what specifically is missing?

hughsie, 2022-11-28:
The kernel driver that talks to the PSP, IIRC. @superm1 might be able to share more.

blitz, 2022-11-30:
@superm1 Thanks for the background info!

IPlayZed, 2023-05-29:
Would it be possible to flag devices where this is a known issue? I would otherwise have HSI-3, but  this causes failure because of the mentioned problem.

hughsie, 2023-06-02:
> I would otherwise have HSI-3

Right, but if we don't have the data that allows us to check the system is indeed secure, isn't it more correct to tell the truth? Otherwise I could run fwupd on a SoC we have zero tests for and get HSI-4.

superm1, 2023-06-07:
If changing the error messaging at all I think it's best not to call out the SOC but instead the more generic term "platform". Then this can scale better for other situations that can arise in other architectures too.

IPlayZed, 2023-06-08:
Yes, so something like "Platform did not provide the needed information from the PSP".

superm1, 2023-06-08:
I've adjusted it: https://github.com/fwupd/fwupd/commit/432aa86783b4fed4f35364b28f40a5b04b540b7b

superm1, 2024-05-29:
Here is a [patch series](https://lore.kernel.org/linux-crypto/oCZE3H8ZnqlQsatOdb4HH7_0t7E9fsRqVv1QodA3IeUSDCdx9DP1bWsLOVq6T5iGffMjzMequirUJSM4UpcA1y2LZVUvMsffulaArsPy2cs=@hughsie.com/T/#t) for the kernel that enables the feature on the older SoCs.

IPlayZed, 2024-05-29:
> Here is a [patch series](https://lore.kernel.org/linux-crypto/oCZE3H8ZnqlQsatOdb4HH7_0t7E9fsRqVv1QodA3IeUSDCdx9DP1bWsLOVq6T5iGffMjzMequirUJSM4UpcA1y2LZVUvMsffulaArsPy2cs=@hughsie.com/T/#t) for the kernel that enables the feature on the older SoCs.

Hey, happy to see this. Which models does this fix enable the features related to this issue (from-to)?

superm1, 2024-05-29:
I personally tested it on Renoir and Vermeer, but it's supposed to cover Matisse and Cezanne as well.

IPlayZed, 2024-06-23:
> I personally tested it on Renoir and Vermeer, but it's supposed to cover Matisse and Cezanne as well.

@superm1 I have a Vermeer family CPU, I would like to check if this works in my case, which kernel and fwupd version would I need to do so?

superm1, 2024-06-23:
6.11-rc1 will have it. You can backport manually right now if you want.