fwupd/fwupd #4899: Add translatable strings for all Lenovo BIOS attributes

superm1, 2022-08-02:
Lenovo BIOS attributes don't actually have a description in `display_name`.  Instead they just duplicate the value that is in current_value.  To make BIOS settings more user friendly, we should add translated descriptions for all these attributes.

To fix this, please populate this area: https://github.com/fwupd/fwupd/blob/main/src/fu-util-bios-attr.c#L34

superm1, 2022-08-24:
Just FYI, that file moved in `main` to `src/fu-util-bios-settings.c`, so if you started working on this already you should rebase to the latest.

hughsie, 2024-11-11:
@superm1 is this still true? If so, maybe @mrhpearson can open an internal Lenovo issue?

superm1, 2024-11-11:
> @superm1 is this still true? If so, maybe @mrhpearson can open an internal Lenovo issue?

Yeah

hughsie, 2024-11-11:
@mrhpearson please do your LO thing. Thanks :)

mrhpearson, 2024-11-11:
Just to check the intent of this ticket, as the title is confusing - the aim is to have a description of what each attribute does  - not a translation.

We're looking to add that feature, but it will come from the BIOS itself. In my opinion that is better as it will continue for new updates without me having to do anything.

I haven't yet dug into what that will mean in the kernel, as we haven't agreed the spec with the BIOS team yet (and realistically it's going to take some time to deliver - we're looking to make naming consistent across all portfolio's). 

Assuming that there is a description string from the BIOS that is available by the firmware-attributes class - that would cover the intent of this ticket right?

hughsie, 2024-11-11:
Ya, it's so fwupd can translate it, rather than asking the BIOS to provide translatable strings. At the moment the `display_name` is super unhelpful.

mrhpearson, 2024-11-11:
I'm still confused I'm afraid. I am looking for clarification on the meaning of 'translate' :) 

By 'translate' do we mean into different languages, or do we just mean a useful description (i.e convert a name into something meaningful to users).
I think we're going to fix the 'useful description' part of it, but not the different languages.

hughsie, 2024-11-11:
Ignore _translate_ completely :) What we need is the US English short summary of the firmware option in the `display_name` attribute.

hughsie, 2025-05-04:
@mrhpearson did you have any updates from the firmware team? Thanks.

mrhpearson, 2025-05-05:
Somewhat. It's part of a bigger exercise to improve the WMI interface that is still under review. Afraid that will take time as we're also aiming to make it common across all the different BU's.