fwupd/fwupd #4901: Building fwupd for certain distro's fails

Sean-StarLabs, 2022-08-02:
`ubuntu:focal` and `ubuntu:jammy` images fail to build due to the lack of `bash-completion`. Add that to the `dependencies.xml` for Ubuntu allows to it get past that step.

Then, building `ubuntu:focal` will fail due to only having meson 53. Backport that, it then fails at:

```
Status: Downloaded newer image for ubuntu:focal
 ---> 3bc6e9f30f51
Step 2/7 : ENV OS ubuntu-x86_64
 ---> Running in e289ef6af650
Removing intermediate container e289ef6af650
 ---> aa476725ef9b
Step 3/7 : ENV CI_NETWORK true
 ---> Running in 615d797d351e
Removing intermediate container 615d797d351e
 ---> ef943e99db0f
Step 4/7 : RUN echo fubar > /etc/machine-id
 ---> Running in c1e58f240294
Removing intermediate container c1e58f240294
 ---> bcfc0f764fae
Step 5/7 : RUN apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	abigail-tools 	bash-completion 	ca-certificates 	clang 	clang-tools 	clang-format 	libcairo-dev 	libcairo-gobject2 	libjson-glib-dev 	libftdi1-dev 	libpci-dev 	fonts-noto 	devscripts 	debhelper 	dbus-x11 	dh-strip-nondeterminism fakeroot 	libfreetype6-dev 	fontconfig 	libflashrom-dev 	gcab 	gettext 	gnu-efi 	git 	libglib2.0-doc 	libglib2.0-dev gobject-introspection 	gnome-desktop-testing 	gnutls-dev 	gnutls-bin 	jq 	libxmlb-dev 	libjcat-dev 	liblzma-dev 	libarchive-dev 	libcbor-dev 	libefivar-dev 	libefiboot-dev 	libgcab-dev 	libgirepository1.0-dev libgudev-1.0-dev 	libgusb-dev 	libsmbios-dev 	libsoup2.4-dev 	libcurl4-gnutls-dev 	libtool-bin 	libumockdev-dev 	lintian 	locales meson 	mingw-w64-tools 	gir1.2-pango-1.0 	pkg-config 	policykit-1 	libmm-glib-dev 	libqmi-glib-dev 	libmbim-glib-dev 	libpolkit-gobject-1-dev 	python3-gi-cairo 	gi-docgen 	python3-pygments 	python3-typogrify 	python3-toml 	python3-markdown 	python3-jinja2 	python3-requests 	python3-packaging 	libsqlite3-dev 	systemd libsystemd-dev 	shared-mime-info 	texlive-fonts-recommended 	udev 	umockdev 	valac 	valgrind 	libtss2-dev 	shellcheck 	libprotobuf-c-dev 	protobuf-c-compiler 	sudo 	python3-venv
 ---> Running in 700718bb5f1c

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

All packages are up to date.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package gi-docgen
```

If I remove `gi-docgen` from the `dependencies.xml`, it'll build but the `deb` creation fails due to missing files for fwupd-docs.

`ubuntu:jammy` will build successfully.

`debian:latest` will fail for lots of things, these modifications will hack it together: https://github.com/StarLabsLtd/fwupd/commit/3557a34af0ba9e4c018f5a7b4375519fa3efe2e0

superm1, 2022-08-02:
> Step 5/7 : RUN apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	abigail-tools 	bash-completion 	ca-certificates 	clang 	clang-tools 	clang-format [...]

Your install step is very different than the fwupd one.

https://github.com/fwupd/fwupd/blob/main/contrib/ci/Dockerfile-ubuntu.in#L8

Do you have something out of date?

superm1, 2022-08-02:
specifically missing https://github.com/fwupd/fwupd/commit/b3014c2ebfbb1147fc30982330fbf5e331e799b2

Sean-StarLabs, 2022-08-02:
Modifying `debian.sh` is simpler for the deps :)

Does the same with the Ubuntu one:
```
Step 5/7 : RUN apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	abigail-tools 	bash-completion 	ca-certificates 	clang 	clang-tools 	clang-format 	libcairo-dev 	libcairo-gobject2 	libjson-glib-dev 	libftdi1-dev 	libpci-dev 	fonts-noto 	devscripts 	debhelper 	dbus-x11 	dh-strip-nondeterminism 	fakeroot libfreetype6-dev 	fontconfig 	libflashrom-dev 	gcab 	gettext 	gnu-efi 	git 	libglib2.0-doc 	libglib2.0-dev 	gobject-introspection 	gnome-desktop-testing 	gnutls-dev 	gnutls-bin 	jq 	libxmlb-dev 	libjcat-dev 	liblzma-dev 	libarchive-dev 	libcbor-dev 	libefivar-dev 	libefiboot-dev 	libgcab-dev 	libgirepository1.0-dev 	libgudev-1.0-dev 	libgusb-dev 	libsmbios-dev 	libsoup2.4-dev 	libcurl4-gnutls-dev 	libtool-bin 	libumockdev-dev 	lintian 	locales 	meson 	mingw-w64-tools 	gir1.2-pango-1.0 	pkg-config 	policykit-1 	libmm-glib-dev 	libqmi-glib-dev 	libmbim-glib-dev 	libpolkit-gobject-1-dev 	python3-gi-cairo 	gi-docgen 	python3-pygments 	python3-typogrify 	python3-toml 	python3-markdown 	python3-jinja2 	python3-requests 	python3-packaging 	libsqlite3-dev 	systemd 	libsystemd-dev 	shared-mime-info 	texlive-fonts-recommended 	udev 	umockdev 	valac 	valgrind 	libtss2-dev 	shellcheck 	libprotobuf-c-dev 	protobuf-c-compiler 	sudo 	python3-venv
 ---> Running in 3777945b2d18

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

All packages are up to date.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package gi-docgen
The command '/bin/sh -c apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	abigail-tools 	bash-completion 	ca-certificates 	clang 	clang-tools 	clang-format 	libcairo-dev 	libcairo-gobject2 	libjson-glib-dev 	libftdi1-dev 	libpci-dev 	fonts-noto 	devscripts 	debhelper 	dbus-x11 	dh-strip-nondeterminism 	fakeroot 	libfreetype6-dev 	fontconfig 	libflashrom-dev 	gcab 	gettext 	gnu-efi 	git 	libglib2.0-doc 	libglib2.0-dev 	gobject-introspection 	gnome-desktop-testing 	gnutls-dev 	gnutls-bin 	jq 	libxmlb-dev 	libjcat-dev 	liblzma-dev 	libarchive-dev 	libcbor-dev 	libefivar-dev 	libefiboot-dev 	libgcab-dev 	libgirepository1.0-dev 	libgudev-1.0-dev 	libgusb-dev 	libsmbios-dev 	libsoup2.4-dev 	libcurl4-gnutls-dev 	libtool-bin 	libumockdev-dev lintian 	locales 	meson 	mingw-w64-tools 	gir1.2-pango-1.0 	pkg-config 	policykit-1 	libmm-glib-dev 	libqmi-glib-dev 	libmbim-glib-dev 	libpolkit-gobject-1-dev 	python3-gi-cairo 	gi-docgen 	python3-pygments 	python3-typogrify 	python3-toml 	python3-markdown 	python3-jinja2 	python3-requests 	python3-packaging 	libsqlite3-dev 	systemd 	libsystemd-dev 	shared-mime-info 	texlive-fonts-recommended 	udev 	umockdev 	valac 	valgrind 	libtss2-dev 	shellcheck 	libprotobuf-c-dev 	protobuf-c-compiler 	sudo 	python3-venv' returned a non-zero code: 100

```

superm1, 2022-08-02:
Yeah but my point is that we made a change to not install the deps in the container that way anymore to make it more dynamic.  I think you're missing that in your containers.

Sean-StarLabs, 2022-08-02:
Gotcha - killed the container and recreated, that's now happy - thanks :)

debian:latest is still the same though

superm1, 2022-08-02:
Can I see the logs for the Debian?

Sean-StarLabs, 2022-08-02:
Sure:
```
sudo OS=debian-x86_64 contrib/ci/generate_docker.py build
Sending build context to Docker daemon  53.04MB
Step 1/7 : FROM debian:latest
latest: Pulling from library/debian
001c52e26ad5: Already exists 
Digest: sha256:82bab30ed448b8e2509aabe21f40f0607d905b7fd0dec72802627a20274eba55
Status: Downloaded newer image for debian:latest
 ---> 07d9246c53a6
Step 2/7 : ENV OS debian-x86_64
 ---> Running in abae3b2f4b12
Removing intermediate container abae3b2f4b12
 ---> a9c81b8673a8
Step 3/7 : ENV CI_NETWORK true
 ---> Running in befdafbef3ee
Removing intermediate container befdafbef3ee
 ---> ecc9dcdbbaaa
Step 4/7 : RUN echo fubar > /etc/machine-id
 ---> Running in 52372972fbcb
Removing intermediate container 52372972fbcb
 ---> 3cf891be8fc0
Step 5/7 : RUN apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	bash-completion 	ca-certificates 	clang-format 	libcairo-dev 	libcairo-gobject2 	libjson-glib-dev 	libftdi1-dev 	libpci-dev 	fonts-noto 	devscripts 	debhelper 	dbus-x11 	dh-strip-nondeterminism 	fakeroot 	libfreetype6-dev 	fontconfig 	libflashrom-dev 	gcab 	gettext 	gnu-efi 	git 	libglib2.0-doc 	libglib2.0-dev 	gobject-introspection 	gnome-desktop-testing 	gnutls-dev 	gnutls-bin 	libxmlb-dev 	libjcat-dev 	liblzma-dev 	libarchive-dev 	libcbor-dev 	libefivar-dev 	libefiboot-dev 	libgcab-dev 	libgirepository1.0-dev 	libgudev-1.0-dev 	libgusb-dev 	libsmbios-dev 	libcurl4-gnutls-dev 	libtool-bin 	libumockdev-dev 	lintian 	locales 	meson 	mingw-w64-tools gir1.2-pango-1.0 	pkg-config 	policykit-1 	libmm-glib-dev 	libqmi-glib-dev 	libmbim-glib-dev 	libqrtr-glib-dev 	libpolkit-gobject-1-dev 	python3-gi-cairo 	gi-docgen 	python3-typogrify 	python3-toml 	python3-jinja2 	python3-requests 	python3-packaging 	libsqlite3-dev 	systemd 	libsystemd-dev 	shared-mime-info 	texlive-fonts-recommended 	udev 	umockdev 	valac 	valgrind 	libtss2-dev 	libprotobuf-c-dev 	protobuf-c-compiler 	python3-venv
 ---> Running in 2e4330e1f738

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

All packages are up to date.

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

Reading package lists...
Building dependency tree...
Reading state information...
E: Unable to locate package libqrtr-glib-dev
E: Unable to locate package gi-docgen
The command '/bin/sh -c apt update -qq && 	DEBIAN_FRONTEND=noninteractive apt install -yq --no-install-recommends	bash-completion 	ca-certificates 	clang-format 	libcairo-dev 	libcairo-gobject2 	libjson-glib-dev 	libftdi1-dev 	libpci-dev 	fonts-noto 	devscripts 	debhelper 	dbus-x11 	dh-strip-nondeterminism 	fakeroot 	libfreetype6-dev 	fontconfig 	libflashrom-dev 	gcab 	gettext 	gnu-efi 	git 	libglib2.0-doc 	libglib2.0-dev 	gobject-introspection 	gnome-desktop-testing 	gnutls-dev 	gnutls-bin 	libxmlb-dev 	libjcat-dev 	liblzma-dev 	libarchive-dev 	libcbor-dev 	libefivar-dev 	libefiboot-dev 	libgcab-dev 	libgirepository1.0-dev 	libgudev-1.0-dev 	libgusb-dev 	libsmbios-dev 	libcurl4-gnutls-dev 	libtool-bin 	libumockdev-dev 	lintian 	locales 	meson 	mingw-w64-tools 	gir1.2-pango-1.0 	pkg-config 	policykit-1 	libmm-glib-dev 	libqmi-glib-dev 	libmbim-glib-dev 	libqrtr-glib-dev 	libpolkit-gobject-1-dev 	python3-gi-cairo 	gi-docgen 	python3-typogrify 	python3-toml 	python3-jinja2 	python3-requests 	python3-packaging 	libsqlite3-dev 	systemd 	libsystemd-dev 	shared-mime-info 	texlive-fonts-recommended 	udev 	umockdev 	valac 	valgrind 	libtss2-dev 	libprotobuf-c-dev 	protobuf-c-compiler 	python3-venv' returned a non-zero code: 100
Traceback (most recent call last):
  File "/home/sean/Documents/fwupd/contrib/ci/generate_docker.py", line 97, in <module>
    subprocess.check_call(args)
  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['docker', 'build', '-t', 'fwupd-debian-x86_64', '-f', './Dockerfile', '.']' returned non-zero exit status 100.
```

superm1, 2022-08-03:
Does latest map to stable I guess?

Sean-StarLabs, 2022-08-03:
They seem to exist separately, both with the same date and labelled as Bullseye.

superm1, 2022-08-03:
Okay I guess we need to do the same thing as the Ubuntu container change then.

commit 5180e909 (2022-08-03): trivial: ci: debian: Use helper script to install dependencies instead.

Should fix building Debian stable containers
Fixes: #4901
PR #4906 (2022-08-05): trivial: ci: debian: Use helper script to install dependencies instead.
Should fix building Debian stable containers
Fixes: #4901

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit a19b6f23 (2022-08-05): trivial: ci: debian: Use helper script to install dependencies instead. (#4906)

* trivial: ci: debian: Use helper script to install dependencies instead.

Should fix building Debian stable containers
Fixes: #4901

* trivial: debian: ci: only populate fwupd-doc if dependencies are met

* trivial: ci: debian: generate control file using fwupd_setup_helpers
closed 2022-08-05 by Sean-StarLabs