fwupd/fwupd #4902: fwupdmgr fails to update UEFI on my Lenovo P52 - UEFI capsule updates not available or enabled in firmware setup

rugubara, 2022-08-03:
PF16W6Y2 ~ # fwupdmgr update
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Devices with no available firmware updates:
 • Corsair MP400
 • Corsair MP400
No updatable devices

rugubara, 2022-08-03:
[image]

superm1, 2022-08-03:
Did your not install in UEFI mode?

rugubara, 2022-08-03:
I use uefi-only boot mode

superm1, 2022-08-03:
Do you have an esrt directory in sysfs?

superm1, 2022-08-03:
I mean specifically: `/sys/firmware/efi/esrt`

rugubara, 2022-08-03:
Nothing.

PF16W6Y2 ~ # ls -l /sys/firmware/efi/
total 0
-r--r--r-- 1 root root 4096 авг  3 19:57 config_table
dr-xr-xr-x 2 root root    0 авг  3 09:11 efivars
-r--r--r-- 1 root root 4096 авг  3 19:57 fw_platform_size
-r--r--r-- 1 root root 4096 авг  3 19:57 fw_vendor
-r--r--r-- 1 root root 4096 авг  3 19:57 runtime
-r-------- 1 root root 4096 авг  3 19:57 systab

superm1, 2022-08-03:
Then either your kernel isn't compiled with support or the BIOS isn't properly turning it on.  Nothing fwupd can do about either of these problems.

rugubara, 2022-08-04:
Definetly, my kernel is compiled with esrt support:
PF16W6Y2 /usr/src/linux # zcat /proc/config.gz | grep -i esrt
CONFIG_EFI_ESRT=y

So is this issue better suited  for Lenovo firmware?