fwupd/fwupd #4909: "fwupdate -a <guid> firmware.cap" fails: could not open "<random garbage>": No such file or directory

berolinux, 2022-08-04:
**Describe the bug**
Trying to update the system firmware on a Synquacer (https://www.96boards.org/product/developerbox/) using `fwupdate -a` fails.
`fwupdate -l` correctly prints the firmware GUID and recognizes the presently installed version 31, but trying to apply the update from the capsule file fails claiming `could not open "[random garbage characters]": No such file or directory`

I haven't had the time to look into this deeply, but it looks like something overwrites the filename of the capsule file (strace doesn't show any attempt to open it).

**Steps to Reproduce**
`fwupdate -a '50b94ce5-8b63-4849-8af4-ea479356f0e3' DeveloperBox.Cap`

**Expected behavior**
Firmware is updated

**fwupd version information**
fwupd 1.8.3
fwupd-efi 1.3

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.8.1-g09971062c-dirty
compile   com.hughsie.libjcat           0.1.8
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`dnf`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
Socionext SynQuacer E-series DeveloperBox
│
├─Unknown Device:
│     Device ID:          23ea95870be8a9e633dd65350704f9a1b08b1b0d
│     GUID:               11a12870-d8d9-5d0f-9a56-74be3eaedaa1 ← GPIO\ID_51000000.gpio
│   
├─ASM1062 Serial ATA Controller:
│     Device ID:          5987c136e0024e0f45943927a31c5c4c711f3eb0
│     Current version:    02
│     Vendor:             ASMedia Technology Inc. (PCI:0x1B21)
│     GUIDs:              73bea05e-44cc-5213-a22f-a9ee6302015a ← PCI\VEN_1B21&DEV_0612
│                         bd9f0355-e194-5d42-8077-40561fb32384 ← PCI\VEN_1B21&DEV_0612&REV_02
│                         9811d16d-27dc-5557-a933-ce269dd401e2 ← PCI\VEN_1B21&DEV_0612&SUBSYS_1B211060
│                         e9da2173-5b9b-5e2d-ba1d-0a96c06094a9 ← PCI\VEN_1B21&DEV_0612&SUBSYS_1B211060&REV_02
│                         87070086-f4bb-57cb-958c-24f791fd3a41 ← PCI\VEN_1B21&DEV_1184
│                         1503cdda-2a49-59a5-9602-d79e79a00b19 ← PCI\VEN_1B21&DEV_1184&REV_00
│                         1fe1c557-6d2e-54eb-ba56-2d149cf5c833 ← PCI\VEN_1B21&DEV_1184&SUBSYS_1B21118F
│                         f2594781-cd5f-5253-990a-47a3d87eb40b ← PCI\VEN_1B21&DEV_1184&SUBSYS_1B21118F&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─GK208B [GeForce GT 710] (GT710-SL-1GD5):
│     Device ID:          1fc7521dd6359039b991ad649c7ffc77f4a81fad
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              6d6da4a3-b929-5e6c-a532-099cef09041e ← PCI\VEN_10DE&DEV_128B
│                         de901f49-2a62-5eb5-9c28-3e8ce27263b4 ← PCI\VEN_10DE&DEV_128B&REV_A1
│                         416925d3-cc72-5f53-9ba8-20b6b2780102 ← PCI\VEN_10DE&DEV_128B&SUBSYS_104385F7
│                         519b7dd2-874b-5a4b-820a-2ebebdb61594 ← PCI\VEN_10DE&DEV_128B&SUBSYS_104385F7&REV_A1
│                         f52ff75f-542f-5227-86e2-adcb77448706 ← PCI\VEN_10DE&DEV_128B&REV_00
│                         a5779ae6-d6c8-5a3d-b25a-31c0584ba724 ← PCI\VEN_10DE&DEV_128B&SUBSYS_104385F7&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─M32508:
│     Device ID:          0506acf884bc36bd5ebeba68c3e03e3c1a996d37
│     Current version:    0x5200000000000000
│     Vendor:             Kingston (EMMC:0x000070)
│     Serial Number:      0x11437899
│     GUIDs:              70d7765a-f6c5-549c-9613-fa9e12fca051 ← EMMC\NAME_M32508
│                         7e598bab-95c2-546a-b17a-8396c6d44d7d ← EMMC\NAME_M32508&REV_0x5200000000000000
│                         5486249f-ae3e-5b15-b2b5-3afe68439b46 ← EMMC\MAN_0070&OEM_0100
│                         35a8e4a9-3c51-5cb6-8356-74bad5e9285a ← EMMC\MAN_0070&OEM_0100&NAME_M32508
│                         6960c360-9e19-583c-8170-ef773a873ffe ← EMMC\MAN_0070&NAME_M32508&REV_0x5200000000000000
│                         c27a3d87-e36b-56f0-9920-909a1d4d7a56 ← EMMC\MAN_0070&OEM_0100&NAME_M32508&REV_0x5200000000000000
│     Device Flags:       • Internal device
│                         • Updatable
│   
└─WDC WD10EZEX-22MFCA0:
      Device ID:          3fd299d1970278eb390d05c9c3f3af5dad19514d
      Summary:            ATA drive
      Current version:    01.01A01
      Vendor:             Western Digital (ATA:0x101C, OUI:0014ee)
      Serial Number:      WD-WCC6Y4CAF7AK
      GUIDs:              e1879707-d706-5f17-ad95-814cd27d82c3 ← IDE\WDC_WD10EZEX-22MFCA0____________________01.01A01
                          d9344506-64b8-5365-9a62-f8a679965146 ← IDE\0WDC_WD10EZEX-22MFCA0____________________
                          c5d86e7a-19e3-50b1-92ef-3681be3648e1 ← WDC WD10EZEX-22MFCA0
      Device Flags:       • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
```

**System UEFI configuration**
Please provide the output of the following commands:

```shell
efibootmgr -v
BootCurrent: 0004
Timeout: 5 seconds
BootOrder: 0002,0000,0009,000A,000B,000C,000D,000E,000F,0010,0001,0003,0004
Boot0000* UiApp FvVol(89cc2ab6-b847-475f-93e2-819603c3d15a)/FvFile(462caa21-7614-4503-836e-8ab6f4662331)
Boot0001* UEFI WDC WD10EZEX-22MFCA0 WD-WCC6Y4CAF7AK     PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x1,0x0)/Pci(0x0,0x0)/Sata(1,65535,0)N.....YM....R,Y.
Boot0002* openmandriva  HD(1,GPT,05f6d99b-e131-4a12-9082-9a9317c05f1c,0x800,0x80000)/File(\EFI\openmandriva\grubaa64.efi)
Boot0003* Install Debian Stretch (Linaro ERP) over HTTP MAC(e0d55e65a68f,1)/IPv4(0.0.0.00.0.0.0,0,0)/Uri(http://people.linaro.org/~ard.biesheuvel/devbox/mini.iso)
Boot0004* UEFI USB Flash Disk SCY0000000002101  PcieRoot(0x0)/Pci(0x0,0x0)/Pci(0x5,0x0)/Pci(0x0,0x0)/USB(1,0)N.....YM....R,Y.
Boot0009* UEFI PXEv4 (MAC:E0D55E65A68F) MAC(e0d55e65a68f,1)/IPv4(0.0.0.00.0.0.0,0,0)N.....YM....R,Y.
Boot000A* UEFI PXEv6 (MAC:E0D55E65A68F) MAC(e0d55e65a68f,1)/IPv6([::]:<->[::]:,0,0)N.....YM....R,Y.
Boot000B* UEFI HTTPv4 (MAC:E0D55E65A68F)        MAC(e0d55e65a68f,1)/IPv4(0.0.0.00.0.0.0,0,0)/Uri()N.....YM....R,Y.
Boot000C* UEFI HTTPv6 (MAC:E0D55E65A68F)        MAC(e0d55e65a68f,1)/IPv6([::]:<->[::]:,0,0)/Uri()N.....YM....R,Y.
Boot000D* UEFI Shell    FvVol(89cc2ab6-b847-475f-93e2-819603c3d15a)/FvFile(7c04a583-9e3e-4f1c-ad65-e05268d0b4d1)
Boot000E* UEFI eMMC Device      VenHw(0d51905b-b77e-452a-a2c0-eca0cc8d514a,000030520000000000)/eMMC(0)/Ctrl(0x0)N.....YM....R,Y.
Boot000F* UEFI eMMC Device 2    VenHw(0d51905b-b77e-452a-a2c0-eca0cc8d514a,000030520000000000)/eMMC(0)/Ctrl(0x1)N.....YM....R,Y.
Boot0010* UEFI eMMC Device 3    VenHw(0d51905b-b77e-452a-a2c0-eca0cc8d514a,000030520000000000)/eMMC(0)/Ctrl(0x2)N.....YM....R,Y.
```

```shell
efivar -l | grep fw
[blank output]
```

```shell
tree /boot
/boot
├── actions
│   ├── s700-cubieboard7.dtb
│   └── s900-bubblegum-96.dtb
├── efi
│   └── EFI
│       ├── BOOT
│       │   ├── bootaa64.efi
│       │   ├── fwupdaa64.efi
│       │   └── grub.cfg
│       └── openmandriva
│           ├── fw
│           │   └── DeveloperBox.Cap
│           ├── fwupaa64.efi
│           └── grub.efi
├── freescale
│   ├── imx8mm-evk.dtb
│   ├── imx8mn-ddr4-evk.dtb
│   ├── imx8mn-evk.dtb
│   ├── imx8mp-evk.dtb
│   ├── imx8mq-evk.dtb
│   ├── imx8mq-hummingboard-pulse.dtb
│   ├── imx8mq-librem5-devkit.dtb
│   ├── imx8mq-nitrogen.dtb
│   ├── imx8mq-phanbell.dtb
│   ├── imx8mq-pico-pi.dtb
│   ├── imx8mq-thor96.dtb
│   ├── imx8mq-zii-ultra-rmb3.dtb
│   ├── imx8mq-zii-ultra-zest.dtb
│   ├── imx8qxp-ai_ml.dtb
│   ├── imx8qxp-colibri-eval-v3.dtb
│   ├── imx8qxp-mek.dtb
│   └── s32v234-evb.dtb
├── grub2
│   ├── grub.cfg
│   └── themes
├── initrd-5.8.1-g09971062c-dirty.img
├── intel
│   ├── socfpga_agilex_socdk.dtb
│   └── socfpga_agilex_socdk_nand.dtb
└── vmlinuz-5.8.1-g09971062c-dirty
```

**Additional questions**

- Operating system and version:
Cooker

- Have you tried rebooting?
yes

- Is this a regression?
no

- Are you using an NVMe disk?
no

- Is secure boot enabled?
no

- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
no

superm1, 2022-08-04:
Would you please share that strace output from your attempt?  Would you also please add `--verbose` to your invocation?
Also, can this reproduce with `fwupdtool install-blob`?

superm1, 2022-08-04:
> fwupdtool install-blob doesn't seem to recognize 50b94ce5-8b63-4849-8af4-ea479356f0e3

Can you try just `fwupdtool install-blob foo.bin` and then pick the device from the picker?

> openat(AT_FDCWD, "\351R\256\252\n", O_RDONLY) = -1 ENOENT (No such file or directory)

Yeah that's the most concerning to me too.  Maybe your ESP is not being detected properly and we don't have a clear message for it..  Do you have udisks installed?  

fwupdtool is better about being verbose for all this stuff, that's why I'm asking for a test with it.

berolinux, 2022-08-04:
`fwupdtool` doesn't see the device, not if I specify the guid and not if I leave it out.
```
# fwupdtool install-blob DeveloperBox.Cap 
Loading…                 [*                                      ]18:42:53:0606 FuContext            Failed to load SMBIOS: structure larger than available data
Loading…                 [******                                 ]18:42:54:0480 FuUdevDevice         reading subsystem_vendor for 0xffff was invalid: value 65535 was above maximum 65534
18:42:54:0480 FuUdevDevice         reading subsystem_device for 0xffff was invalid: value 65535 was above maximum 65534
Loading…                 [-                                      ]
Writing…                 [************                           ]Choose a device:
0.      Cancel
1.      0506acf884bc36bd5ebeba68c3e03e3c1a996d37 (M32508)
2.      3fd299d1970278eb390d05c9c3f3af5dad19514d (WDC WD10EZEX-22MFCA0)
```

Since I don't want to flash the mainboard firmware to a harddisk, no option but to select cancel.

udisks is installed.
Specifying the ESP manually with `-p /boot/efi` doesn't change anything.

superm1, 2022-08-04:
I don't understand why no device showing up. Can we see verbose output for fwupdtool?

berolinux, 2022-08-04:
I've figured out the problem with `fwupdate`.

https://github.com/rhboot/fwupdate/blob/security/linux/fwupdate.c#L306
`filename` gets set to `poptGetArg(optcon)` -- a pointer to somewhere in popt's context
https://github.com/rhboot/fwupdate/blob/security/linux/fwupdate.c#L327
The popt context is freed -- which at least in current versions of popt also frees the pointers inside it
https://github.com/rhboot/fwupdate/blob/security/linux/fwupdate.c#L385
`filename` is referenced after the popt context has been freed and something else was free to overwrite the memory.

The fix is to add a `strdup` in https://github.com/rhboot/fwupdate/blob/security/linux/fwupdate.c#L306 (and while I haven't had a problem with it, https://github.com/rhboot/fwupdate/blob/security/linux/fwupdate.c#L296 as well -- something could overwrite the guid between free and use as well).

Is it worth sending an MR to the dead project to fix it?

In the mean time, yes, it would be nice to have `fwupdtool` working as well, so let's keep this bug open... Will take a look at that next

superm1, 2022-08-04:
Wait a minute; you're using the `fwupdate` from the fwupdate project!?  I figured you were using the `fwupdate` that we distribute with _fwupd_'s `fwupdate` compatibility tool..  It's not worth doing anything with the dead `fwupdate`.  If that bug reproduces with fwupd's `fwupdate` please submit a PR.  But yeah, we should focus on fwupdtool and why it doesn't populate an ESRT device for you.

superm1, 2022-08-05:
> Loading…                 [*                                      ]18:42:53:0606 FuContext            Failed to load SMBIOS: structure larger than available data

That is worrying.  When you provide the full `fwupdtool get-devices --verbose` output, I would like to also see `fwupdtool hwids` output too.

We might need a copy of  `/sys/firmware/dmi/tables` from your system to investigate further.

closed 2023-02-20 by hughsie