fwupd/fwupd #4913: Support asmedia usb4 format

superm1, 2022-08-08:
There are patches on the mailing list that will be adding support for the NVM format used by ASMedia. We should make sure that the matching code for fwupd is also updated for this.

https://lore.kernel.org/linux-usb/202208071845.cTQlPQto-lkp@intel.com/T/#t

hughsie, 2022-08-10:
Got a hint? Would this be a variant of libfwupdplugin/fu-intel-nvm.c as added in wip/hughsie/FuIntelNvm?

superm1, 2022-08-10:
Yeah it should definitely come with or after that.

I was just raising it now in case it makes sense to fold into the same branch.

superm1, 2022-08-10:
> variant of libfwupdplugin/fu-intel-nvm.c

I haven't looked closely; but I don't have any expectation that it shares same characteristics as this.  It should be "another" kind of NVM, not subclass that.

superm1, 2022-08-15:
They submitted a v3, here's the updated link for it:
https://lore.kernel.org/linux-usb/202208152107.P6jLM6te-lkp@intel.com/T/#m1a8b1478fad56317e8f2c0895e996876a3aafca4

I left some comments with the goal that they can re-use the existing sysfs files so userspace doesn't need to change how the version string parses significantly.  Depending on what is accepted we will want to at _least_ update our unit tests.

superm1, 2022-08-31:
With the changes in 1.8.5 I suspect this is no work unless we get more details from ASMedia, so I'm going to close it.  If we to end up needing changes will revisit then.

closed 2022-08-31 by superm1