fwupd/fwupd #4922: Could not apply capsules: volume full

rogueai, 2022-08-15:
**Describe the bug**
I'm running `fwmgr update` on my ThinkPad P53 from Arch Linux, dual booting, when rebooting the update process starts correctly, but I'm getting:
```
Could not apply capsules: volume full
```

**Steps to Reproduce**
- `fwupdmgr update`
- reboot

**Expected behavior**
Firmware updates correctly.

**fwupd version information**

```shell
❯ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.16-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10

```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
Installed through `pacman -S fwupd`

**fwupd device information**

```shell
❯ fwupdmgr get-devices --show-all-devices
LENOVO 20QNCTO1WW
│
├─Unknown Device:
│     Device ID:          4a4907dbb1b96c6a1177dfd1b95eb41c781d1265
│     GUID:               1199a818-4c52-5137-b536-d59e2e2cada9 ← GPIO\ID_INT3450:00
│   
├─0000:00:1f.5:
│     Device ID:          b04e387fb80d2b91f37a4d0c7b21461c451775e1
│     Summary:            Memory Technology Device
│     Vendor:             DMI:LENOVO
│     GUIDs:              5f93d7e7-e282-59b9-b663-0146e382f8f6 ← MTD\NAME_0000:00:1f.5
│                         1d5d8ab8-c56e-59b0-9567-d4d306907af8 ← MTD\VENDOR_LENOVO&NAME_0000:00:1f.5
│                         a6962606-1b89-53ff-8561-fe5b1f82d16b ← MTD\VENDOR_LENOVO&PRODUCT_20QNCTO1WW&NAME_0000:00:1f.5
│     Device Flags:       • Internal device
│                         • Updatable
│                         • Needs a reboot after installation
│                         • Cryptographic hash verification is available
│   
├─CoffeeLake-H GT2 [UHD Graphics 630]:
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Vendor:             Intel Corporation (PCI:0x8086)
│     GUIDs:              db677673-1102-5481-a11e-6f397e123ced ← PCI\VEN_8086&DEV_3E9B
│                         12c60b98-52cd-500a-b99e-087aa6b47f0b ← PCI\VEN_8086&DEV_3E9B&REV_00
│                         50512ae6-69eb-58e3-8ac0-b3e10d4bad7e ← PCI\VEN_8086&DEV_3E9B&SUBSYS_17AA2297
│                         c7b07ad5-a672-58fd-94cb-e0b50c9399ec ← PCI\VEN_8086&DEV_3E9B&SUBSYS_17AA2297&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
├─Core™ i7-9750H CPU @ 2.60GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
[... 284 lines cut ...]
```shell
❯ efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-9d97f767-ea06-474f-9141-7e2430d97fc1-0
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-533b2fa2-90be-4e2b-942f-f9b6950e310f-0
```

```shell
❯ tree /boot
/boot
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── intel-ucode.img
└── vmlinuz-linux
```
Note: I'm using rEFInd as a boot loader, booting into a UEFI image created from the above kernel + initrd + ucode + kernel parameters, if that's relevant info.

**Additional questions**

- Operating system and version: Linux arch-linux 5.18.16-arch1-1 #1 SMP PREEMPT_DYNAMIC Wed, 03 Aug 2022 11:25:04 +0000 x86_64 GNU/Linux
- Have you tried rebooting? Yes
- Is this a regression? I don't know
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS?
It is a Lenovo, not quite sure about the Boot Order Lock.

superm1, 2022-08-15:
Is your ESP full?

rogueai, 2022-08-15:
There is plenty of space as far as I can tell:
```
/dev/nvme0n1p1           1.1G  275M  782M  26% /efi
```

I'm not sure if it's relevant, but there are actually a few files in the efi partition that look related to fwupd:
```
├── EFI
│   ├── arch
│   │   ├── fw
│   │   │   ├── fwupd-533b2fa2-90be-4e2b-942f-f9b6950e310f.cap
│   │   │   └── fwupd-9d97f767-ea06-474f-9141-7e2430d97fc1.cap
│   │   └── fwupdx64.efi
```

superm1, 2022-08-15:
Yeah fwupd stages the capsules there, which is why I was suggesting if it's full that could be the problem.
Maybe this is a problem with the NVRAM being full then.

We don't have that error string in the fwupd codebase, so it must be coming from the firmware.

I think there isn't much more we can do from fwupd side, let's see if Lenovo has any thoughts on it.