fwupd/fwupd #4929: Bad output for update details

nchiapol, 2022-08-19:
**Describe the bug**

```
# fwupdmgr get-releases
Choose a device:
0.      Cancel
1.      2292ae5236790b47884e37cf162dcf23bfcd1c60 (Embedded Controller)
2.      349bb341230b1a86e5effe7dfe4337e1590227bd (Intel Management Engine)
3.      f2759da7fe8e0388c5f3601cb072f837b1070b03 (MZVLB512HAJQ-000L7)
4.      a45df35ac0e948ee180fe216a5f703f32dda163f (System Firmware)
2
LENOVO 20L6S09K00
│
├─ThinkPad T480:
│     New version:        184.92.4222
│     Remote ID:          lvfs
│     Release ID:         12646
│     Summary:            Lenovo ThinkPad T480 Consumer ME Firmware
│     Licence:            Proprietary
│     Size:               2.2 MB
│     Created:            2016-07-08
│     Urgency:            High
│     Details:            https://pcsupport.lenovo.com/de/en/search?query=N24RN20W
│     Vendor:             Lenovo
│     Update Message:     The computer will be restarted automatically after updating BIOS completely . Do NOT turn  off your computer or remove the AC adaptor while update is in progress.  
│     Description:        
│     • 8 Intel Platform Update 2022.1
│     
│     Version 11.8.92.4222 (LVFS: 184.92.4222)
│     
│     Problem Fixes
│     
│     • Mitigated the following security vulnerabilities:
│   
├─ThinkPad T480:
│     New version:        184.90.3987
```
shows at least two strange things:
  1) `Created:            2016-07-08` does not make much sense 
  2) The list with `the following secuirty vulnerabilities` seemst to be empty.

It's unclear to me if I do not understand the output, if this is a problem in `fwupd` or if Lenovo is providing bad releases.

Further things I noticed:
  * https://fwupd.org/lvfs/devices/com.lenovo.ThinkPadN24RN.firmware shows a useful Release date and includes the single fixed CVE.
  * All further releases in the list show the same `Created` date
  * The output in https://github.com/fwupd/fwupd/issues/3591 shows the same `Created` date

**fwupd version information**
```
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.0
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.18.0-4-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
```
installed via apt

**Additional questions**
- Operating system and version:
Debian testing
- Have you tried rebooting?
No
- Is this a regression?
Don't think so.

hughsie, 2022-08-22:
>  Created: 2016-07-08 does not make much sense

That's indeed what Lenovo wrote in the update metadata:

    <release version="184.92.4222" date="2016-07-08" urgency="high" tag="N24RN20W">

> The list with the following secuirty vulnerabilities seemst to be empty.

Hmm, that was specified by Lenovo. Can you grab me the output of `fwupdmgr get-updates --json` please.

nchiapol, 2022-08-22:
Concerning `Created`: I feared something like this since the same date is shown for all releases.
Concerning `fwupdmgr get-updates --json`: I have applied all updates, so its empty:
```
 {
  "Devices" : []
}
```

I guess this is another problem in the data provided by Lenovo. Somehow there seems to be two sources in use, one by the website and one by `fwupdmgr`. Can you inform Lenovo and/or the other relevant parties about the problem?

hughsie, 2022-08-22:
@mrhpearson would you be okay if I checked the release date of uploaded updates on the LVFS was < 3 years from the current date? That seems like a sane check to me.

mrhpearson, 2022-08-22:
That seems totally sane to me too.

QQ related to the 2nd item. I noticed that you specifically prevent CVE's being mentioned in the description (I saw the code with extra checking for 'sneaky vendors' in the site source. What's the reason for that? I noted that for the X1 Carbon 9 recently they ended up removing the CVE information because it wasn't allowed to be uploaded with it - but I wasn't sure why.

hughsie, 2022-08-22:
> I noticed that you specifically prevent CVE's being mentioned in the description

Ya, we force vendors to add them as tags that get stored in "issues" -- this lets us link together CVEs using VINCE and also means we can show the user a *link* and *long description* for each CVE, rather than the cryptic numbers and letters.

hughsie, 2022-08-23:
> That seems totally sane to me too.

https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1359

superm1, 2022-08-30:
the fwupd half of this was fixed by https://github.com/fwupd/fwupd/pull/4931, so I'm going to close this.

closed 2022-08-30 by superm1