fwupd/fwupd #4936: Limit get-devices usage to one device, preventing disconnection of other same devices in host OS

semihalf-hajec-lukasz, 2022-08-22:
**Is your feature request related to a problem? Please describe.**
We want to use fwupd to update Genesys gl3590 hub FW in Google servo devices -[ this one.](https://github.com/fwupd/fwupd/blob/main/plugins/genesys/genesys.quirk#L7)
The most common use case for these devices is using few of them with one host computer at the same time (connected via USB).
The problem is when we want update only one of devices. When using `fwupdtool get-devices` it disconnects all devices in host OS for a moment, which disrupts functioning of devices not intended to be updated. I am not seeing any options to somehow limit/ isolate `get-devices` to specific device/bus/sys path.  

**Describe the solution you'd like**
Best solution for us would be adding extra argument that describes the USB path (or device path, in general) to work with and have fwupd ignore everything else. Something like `fwupdtool --device usb:01:00:06 get-devices`. I believe making it somehow generic can be useful for other users. 

**Describe alternatives you've considered**
If it would be not possible to add this kind of feature to fwupd in general, maybe it would be possible to implement it somehow plugin specific?

**Additional context**
nothing for know, but can attach some logs if needed

hughsie, 2022-08-22:
What fwupd version? iirc Genesys sent a patch to stop the disconnection fairly recently.

hughsie, 2022-08-22:
The fix was 58102313bcbab983845d26a15e893f680366012d and that's in 1.8.2 and newer.

semihalf-hajec-lukasz, 2022-08-22:
I was using 1.8.1, not aware of this, thanks for letting me know! 
I would verify it soon.

semihalf-hajec-lukasz, 2022-08-22:
I have just tested 1.8.3 and it works, so problem was more related to plugin and not having newest version. Thanks a lot for such a quick help!

closed 2022-08-22 by semihalf-hajec-lukasz