fwupd/fwupd #4948: Authentication support for BIOS settings

superm1, 2022-08-24:
The Linux kernel firmware-attributes API is supposed to offer authentication for various password and certificate types for BIOS settings.
In trying to plan out how to use this with fwupd, I tried to set it by hand on a Z13, but it's not working.  Weirdly enough I'm getting different error numbers for different attempts too.

```
root@Z13-Gen-1:/sys/class/firmware-attributes/thinklmi/authentication# echo "test12345" | tee Admin/new_password
test12345
tee: Admin/new_password: Permission denied
root@Z13-Gen-1:/sys/class/firmware-attributes/thinklmi/authentication# echo "test12345" | tee System/new_password
test12345
tee: System/new_password: Invalid argument
root@Z13-Gen-1:/sys/class/firmware-attributes/thinklmi/authentication# echo "test12345" | tee Admin/new_password
test12345
tee: Admin/new_password: Invalid argument
```

I don't think I'm doing it wrong, but I would like to check as however this is supposed to work will impact the design of how to do authentication for BIOS settings with fwupd.  This is with thinklmi on 6.0-rc2, BIOS `N3GET38W (1.19 )`.

@mrhpearson comments?

superm1, 2022-08-25:
Looking through the source I found this comment

> *  - A password cannot be set using this method when one does not already
>  *    exist. Passwords can only be updated or cleared.

So perhaps that's the issue.  If so, how do I program a password to start?

ljrk0, 2022-11-01:
I think the initial password must indeed be set in the firmware settings.

However, in order to support authentication, `fwupdmgr` needs to be able to unlock the BIOS settings.  This is required for various features such as setting `com.thinklmi.SleepState` in `fwupdmgr security`.

From the specification:

>		Devices support various authentication mechanisms which can be exposed
>		as a separate configuration object.
>
>		For example a "BIOS Admin" password and "System" Password can be set,
>		reset or cleared using these attributes.
>
>		- An "Admin" password is used for preventing modification to the BIOS
>		  settings.
>		- A "System" password is required to boot a machine.
>
>		Change in any of these two authentication methods will also generate an
>		uevent KOBJ_CHANGE.
>
>		is_enabled:
>					A file that can be read to obtain a 0/1 flag to see if
>					<attr> authentication is enabled.
>					This attribute is mandatory.
>
>		[...]		
>
>		current_password:
>					A write only value used for privileged access such as
>					setting	attributes when a system or admin password is set
>					or resetting to a new password
>
>					This attribute is mandatory when mechanism == "password".
>
>		Note, password management is session specific. If Admin password is set,
>		same password must be written into current_password file (required for
>		password-validation) and must be cleared once the session is over.
>		For example::
>
>			echo "password" > current_password
>			echo "disabled" > TouchScreen/current_value
>			echo "" > current_password
>
>		[...]
>
>		On Dell and Lenovo systems, if Admin password is set, then all BIOS attributes
>		require password validation.
>		On Lenovo systems if you change the Admin password the new password is not active until
>		the next boot.

Indeed, it is required to "log in" before changing any settings if the Admin password is set:

```
# echo 'Windows10' > /sys/class/firmware-attributes/thinklmi/attributes/SleepState/current_value 
bash: echo: write error: Invalid argument
```

After logging in, the same thing succeeds:
```
# cat > /sys/class/firmware-attributes/thinklmi/authentication/Admin/current_password 
my-super-secret-password
# echo 'Windows10' > /sys/class/firmware-attributes/thinklmi/attributes/SleepState/current_value 
```

Afterwards we need to log out in order to protect the settings again:
```
# echo '' > /sys/class/firmware-attributes/thinklmi/authentication/Admin/current_password
```

mrhpearson, 2022-11-01:
Sorry - I somehow never saw this thread (I think it must have been during a week of summer vacation?).

I've definitely tested this on some platforms so it should work and the instructions above look correct to me (thanks @ljrk0).

One important point - after a failure to authenticate - you may have to reboot. it only allows a very limited number of failures. It's a pain and drove me batty during testing because if you make a typo you can't do anything but swear at the wretched thing.
Also check some of the other fields under the Authentication directory - particularly the kbdlang. For users in non-US countries it does not figure out automatically what the language is and I know that caused headaches previously for some (I assigned that issue to a colleague to look at but it went nowhere and I've not had time...)

I found the below notes in some documentation:
· Administrator password, POP and HDP cannot be changed simultaneously during one power cycle. If you want to change these settings, you must reboot the system after changing one of them.
·A password cannot be set using this method before there already exists one. Passwords can only be updated or cleared

@superm1 - holler if you need anything else; particularly if you're thinking of looking at doing the certificate based authentication. I can't release the source code for doing the signing for that yet (I've not completed the internal review process) but I can share it on an individual basis for testing needs when needed.

superm1, 2022-11-01:
This is quite a bit of limitations; it's unfortunate they aren't expressed in the kernel class interface in any way.  Will need to give it some thoughts on how to structure this.

Can you at least get some solid numbers how many times you're allowed to try before it's locked out?

Regarding the certificate-based flow, how does it change?  Do you still start a session with the certificate?  Maybe you can walk through how you envision fwupd to work with the certificate.  Who generates it?  Who/how is it enrolled?  Is it tied to the session like the password is?
I want to make sure we get our design right to encompass both certificate and password flows.

The other unfortunate thing that is apparent from the above explanation is that there is no way to lock so "only" fwupd gets to write to the the sysfs files.  So that means if you "start" a password-based session with fwupd, any other userspace with appropriate permissions can also write to the BIOS attributes.  Given this limitation it's probably best that fwupd:
* Only unlocks right before writing attribute(s)
* Locks right after writing attribute(s)

erwinnv, 2023-07-28:
Sorry for reviewing old bugs. But here is some information I found on this topic:

- https://download.lenovo.com/pccbbs/mobiles_pdf/lenovo_bios_setup_linux_wmi_sysfs_1.2.pdf
- https://www.kernel.org/doc/Documentation/ABI/testing/sysfs-class-firmware-attributes

> Can you at least get some solid numbers how many times you're allowed to try before it's locked out?

I didn't find a number, but there is a `PasswordCountExceededError` in BIOS settings. Try setting it to `Disable` to make it less annoying during development/debugging.

mrhpearson, 2023-08-09:
This bug only seems to get updated when I go on holiday....spooky!

As a side note, I recently came across a limitation with ThinkLMI where you can only update 48 settings and then it fails. We're digging into the root cause of that (seems it's an EFI limitation with how we implement WMI and some security settings). Not sure if that's related....

hughsie, 2024-11-11:
@superm1 we probably need this still, right?

superm1, 2024-11-11:
Ideally it should still be done, yes.

cgoesche, 2025-07-25:
> 
>     * Only unlocks right before writing attribute(s)
> 
>     * Locks right after writing attribute(s)

Would that not open a short-lived opportunity for another process to write to an arbitrary attribute ? The sysfs interface itself might need an improvement in that regard. 

For example:
 * if `/sys/class/firmware-attributes/*/authentication/{Admin,System}/is_enabled` = 1, all attributes will require the Admin/System password as second value, delimited by whitespace
 * if a valid password has been provided to `/sys/class/firmware-attributes/*/authentication/Admin/current_password` 
    a lock will be provided to the process (i dont really like this idea though)

superm1, 2025-07-25:
> > 
> >     * Only unlocks right before writing attribute(s)
> > 
> >     * Locks right after writing attribute(s)
> 
> @superm1 would that not open a short-lived opportunity (race condition) for another process to write to an arbitrary attribute ? The sysfs interface itself might need an improvement in that regard. 
> 
> For example:
>  * if `/sys/class/firmware-attributes/*/authentication/{Admin,System}/is_enabled` = 1, all attributes will require the Admin/System password as second value when written to them
>  * if a valid password has been provided to `/sys/class/firmware-attributes/*/authentication/{Admin,System}/current_password` a lock will be provided to the process (i dont really like this idea though)

I don't disagree on TOCTOU risk.

cgoesche, 2025-07-27:
I took the time to convert the firmware-attributes interface to a stateless system on 6.16-rc7 so that it will require the admin password for each write operation if needed. This would eventually make it so that write operations can be performed atomically and mitigates the TOCTOU risk as well.

However, before I dedicate more time to this I would like to know from you if this is actually significant to this issue and whether the ABI break is even worth it. I mean it would make the system more secure all together, given that it removes a non-trivial attack vector which further restricts access to UEFI settings.

Let me know what you think please.

superm1, 2025-07-27:
@mrhpearson can you please provide comments? 

@cgoesche did you post the patches somewhere for what you've done? It might help Mark decide.

cgoesche, 2025-07-27:
I have a commit [here](https://github.com/cgoesche/linux/commit/673afb28cdc1bfd6c8de7ceb0ff13f5e153f3a88#diff-8aa2fc301f4a480ea3e07edf40c7c4ff41b50c1858bff1ac2ced768ca7a957ab), although for now I have only made the change for the Dell platform specific driver, since I don't have a way to test on different platform.

cgoesche, 2025-07-28:
I admit that breaking the ABI will certainly be a pain for applications that rely on them, however given that the interface is part of **testing/** it might be worth considering that in case of security problems a change can technically be justified according to the ABI documentation. 

https://docs.kernel.org/admin-guide/abi.html#introduction
> [...] The interface can be changed to add new features, but the current interface will not break by doing this, unless grave errors or security problems are found in them. Userspace programs can start to rely on these interfaces, but they must be aware of changes that can occur before these interfaces move to be marked stable.

Besides that, I agree that the ideal would be a certificate-based authentication flow, however that bears the questions which were mentioned by @superm1 previously;  Who generates the certificate? Who/how is it enrolled? Is it tied to the session like the password is? 

Could you briefly describe the sequence if possible and/or explain why a certificate-based flow would be impossible through fwupd ? 

Side note: The Dell specific driver for example does not support a certificate-based flow.

mrhpearson, 2025-07-28:
> Could you briefly describe the sequence if possible and/or explain why a certificate-based flow would be impossible through fwupd ?

Any setting change needs to be signed by the user. We have a helper/demo program (https://github.com/lenovo/tlmi-auth/) on how you might do this, if you've enabled certificate authentication. 

If fwupd want to change things they'd need the users private key to be able to generate the signature for each call. I guess 'impossible' is a bit of an over statement - it just seems unlikely.

Is this really a big concern though? A user has to agree to the FW change being proposed by fwupd, which in itself is a rare event. Presumably they put in the password during this event, so I just can't see it being a realistic attack vector - a tiny potential window, for a system someone is actively in front of, for an event that is rare. Is it worth breaking the existing interface for? There are a number of drivers that will need updating from multiple vendors - sounds like a small nightmare to me :)

cgoesche, 2025-07-29:
Yes, I like the idea. So, if I understand correctly the userspace application can opt-in to the safer password authentication flow via an sysfs interface, e.g. _/sys/class/firmware-attributes/*/authentication/Admin/auth_flow_ which accepts two different values, namely **session** and **atomic** or maybe **stateful** and **stateless**.

superm1, 2025-07-29:
> Yes, I like the idea. So, if I understand correctly the userspace application can opt-in to the safer password authentication flow via an sysfs interface, e.g. _/sys/class/firmware-attributes/*/authentication/Admin/auth_flow_ which accepts two different values, namely **session** and **atomic** or maybe **stateful** and **stateless**. Where in both alternatives the former value is the default, so the ABI doesnt break.
> 
> That very same file can also be read to obtain the current **auth_flow** type.
> 
> Example usage:
> 
> ```
> $ cat /sys/class/firmware-attributes/*/authentication/Admin/auth_flow
> stateful
> 
> $ echo "stateless" > /sys/class/firmware-attributes/*/authentication/Admin/auth_flow
> ```

Yup!

cgoesche, 2025-07-30:
I guess we send the RFC to the interface maintainers before putting any significant effort into an actual feature extension in the sysfs ?

superm1, 2025-07-30:
> I guess we send the RFC to the interface maintainers before putting any significant effort into an actual feature extension in the sysfs ? 

I think that's a good idea.

purpleidea, 2025-10-04:
@cgoesche I found what I assume was your article ( https://christiangn.medium.com/breaking-the-linux-kernel-abi-to-make-bios-option-modifications-more-secure-a10957918f4d ) and somehow found my way here too.

I'm building an fwattr resource for https://github.com/purpleidea/mgmt/ and was also wondering about password related things.

One issue I have with your proposed API, is that:

1) Values can have spaces in them, so where would you do the breaking?

2) Passwords can have spaces in them, so where would you do the breaking?

Perhaps I have overlooked something about how this would work. Maybe instead of a space break you can break on a null char?

Here's an example:

```
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# pwd
/sys/class/firmware-attributes/thinklmi/attributes/BootOrder
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# ls
current_value  display_name  type
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# cat type 
string
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# cat current_value 
M.2 Drive 1:M.2 Drive 2:Network 2:USB HDD:USB CDROM:Network 1
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# cat display_name 
BootOrder
root@hostname:/sys/class/firmware-attributes/thinklmi/attributes/BootOrder# 
```

This is on a modern machine bought ~6 months ish ago:

```
BIOS Information
        Vendor: LENOVO
        Version: M55KT1CA
        Release Date: 04/14/2025

```

HTH

cgoesche, 2025-10-05:
@purpleidea this indeed looks like a limiting factor. I have to admit that I haven't fully tested all fw attributes with the modified API and the branch is still in an early development stage, so very open to contributions.

Regarding the break on null character, the string formatted as `<attribute_value> <bios_password>`, echoed to the interfaces in `/sys/class/firmware-attributes/*/attributes/`, is passed as one null-terminated string, so I guess that would lead to the same problem. :/ 

Honestly, the API design is not ideal, and requires a different approach. OTOH, I am wondering if the TOCTOU risk really warrants a new password based authentication flow, since the risk can only be exploited through a privilege escalation, and if that can be achieved a bad actor will probably have even more interesting attack vectors.

purpleidea, 2025-10-07:
PS: This could also split on \n or even *horror but not horror?* be json with fields like password and a list of all the keys and values to apply...