fwupd/fwupd #4959: Bad and lacking documentation for the majority of "fwupdmgr security"

VorpalBlade, 2022-08-28:
**Describe the bug**
fwupdmgr security prints a number of tests and if they are good or bad. It then links to (in my case) these two pages:
* https://github.com/fwupd/fwupd/wiki/Low-host-security-level
* https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues

However, for the most part these pages do not explain the *actual* issues at hand. For example:
* My system fails "CSME manufacturing mode" for HSI-1. What this is is not documented at all.
* My system also fails "SPI BIOS Descriptor". This is not documented either.

Furthermore there is **no** documentation for HSI-2 - HSI-4 from what I can find. Even to an expert user and software developer many of these entries are cryptic. For example: 
* What on earth is "Pre-boot DMA protection" and how can I enable it?
* How do I fix "Intel BootGuard OTP fuse"?
* Why is suspend-to-ram bad and suspend-to-idle good? What can I do about fixing that?

Finally it is also incorrect. I have root and swap on LVM2. The LVM2 PV is inside a dm-crypt partition. Yet fwupdmgr states that the Linux swap is unencrypted.

**Steps to Reproduce**
1. Run `fwupdmgr security`
2. Try to understand the output based on the linked documentation
3. Fail due to bad documentation

**Expected behavior**
Each possible entry that can fail needs to be documented with at least the following:
* What is it/what is it good for?
* Is it possible for the user to perform an any action to rectify the option being in a bad state?
* If so, what options are there:
  * BIOS options to look for
  * Kernel command line flags (e.g. I had to add `intel_iommu=on` on my Thinkpad T480).
  * Are there any caveats/risks associated with turning these on (e.g. system failing to boot on some broken hardware)?

This sort of documentation will be even more important now that Gnome [will apparently expose this in the GUI](https://www.phoronix.com/news/Ubuntu-No-GNOME-Device-Security). More non-expert users will be encountering this feature in the future.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.4-zen1-1-zen
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): `pacman -S fwupd fwupd-efi`

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

**Note**: I have included this, but it is not really relevant to this bug report.

```shell
$ fwupdmgr get-devices --show-all-devices
LENOVO 20L5CT01WW
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         561403e8-143a-5071-ab09-bf5e4c146983 ← CPUID\PRO_0&FAM_06&MOD_8E
│                         54bdc429-0179-5019-9cb6-8a343324c97c ← CPUID\PRO_0&FAM_06&MOD_8E&STP_A
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
[... 209 lines cut ...]
│                         1edaeaa9-f7e2-5a4f-8bf4-7c204fce837e ← PCI\VEN_8086&DEV_5917&SUBSYS_17AA225E&REV_00
│     Device Flags:       • Internal device
│                         • Cryptographic hash verification is available
│   
└─Unifying Receiver:
      Device ID:          048edd6cf41ca5721da20f4916eaf4b6efa7499d
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Updatable
                          • Signed Payload
```

**Additional questions**

- Operating system and version: Arch Linux (rolling release)
- Have you tried rebooting?
  - Sure. Not really relevant.
- Is this a regression?
  - I guess this has probably always been as bad. So, no.

rhysperry111, 2022-08-28:
I'm on a laptop with a Ryzen 5 5500U, and I can't even get a HSI calculation at all. Under HSI-1 it says my system has failed because the CPU isn't supported.

```
 --> sudo fwupdmgr security --force
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Unencrypted

Runtime Suffix -!
✔ Linux swap:                    Disabled
✔ fwupd plugins:                 Untainted
✘ Linux kernel:                  Tainted
✘ Linux kernel lockdown:         Disabled

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues
```

None of the links provided even mention why a CPU may be unsupported, and it is extremely confusing given this is a recent CPU.

It  would certainly be nice for things like this to be documented.

hughsie, 2022-08-28:
> because the CPU isn't supported

That's true, only newer AMD CPUs are supported.

hughsie, 2022-08-28:
> * My system fails "CSME manufacturing mode" for HSI-1. What this is is not documented at all

See https://fwupd.github.io/libfwupdplugin/hsi.html#me-not-in-manufacturing-mode -- your system is very insecure.

hughsie, 2022-08-28:
> Why is suspend-to-ram bad and suspend-to-idle good? What can I do about fixing that?

We're working on that. Try git main branch:

[image]

VorpalBlade, 2022-08-28:
> > because the CPU isn't supported
> 
> That's true, only newer AMD CPUs are supported.

@hughsie What qualifies as a "newer" CPU then? This is Zen 3, which I would consider "newer"...

VorpalBlade, 2022-08-28:
> > * My system fails "CSME manufacturing mode" for HSI-1. What this is is not documented at all
> 
> See https://fwupd.github.io/libfwupdplugin/hsi.html#me-not-in-manufacturing-mode -- your system is very insecure.

This is an improvement! Why is this not linked instead of the useless wiki pages? However it still fails in mentioning what you as an end user can do about it? Is there likely to be some BIOS option to change it?

hughsie, 2022-08-28:
> Why is this not linked instead of the useless wiki pages

It does if you use `--show-all` with `fwupdmgr security`.

> Also: If there is a security advisory about it, why has there not been a fix pushed from Lenovo

If you file it on https://github.com/fwupd/firmware-lenovo then I can assign the correct people. The test result might be wrong, or it could be that the CSME is unlocked. It certainly wouldn't be the first model that was shipped with the wrong setting (hence the test).

> this info should be on LVFS per manufaturer perhaps

We have PSIRT data for some vendors, but this isn't exposed on the public pages.

VorpalBlade, 2022-08-28:
> > Why is this not linked instead of the useless wiki pages
> 
> It does if you use `--show-all` with `fwupdmgr security`.

May I suggest that the normal output (without `--show-all`) is then extended with a line such as:

```
For links to descriptions of each issue rerun "fwupdmgr security" with --show-all
```

> If you file it on https://github.com/fwupd/firmware-lenovo then I can assign the correct people. The test result might be wrong, or it could be that the CSME is unlocked. It certainly wouldn't be [...]

I will do so.
 
> > this info should be on LVFS per manufaturer perhaps
> 
> We have PSIRT data for some vendors, but this isn't exposed on the public pages.

That seems rather bad. Why would you hide such info?

hughsie, 2022-08-28:
> May I suggest that the normal output (without --show-all) is then extended with a line such as:

I think it would be better to expand each test as a multiline option so we can show the translated details too. It certainly needs some design input.

> Why would you hide such info?

Because we've never had a need to show it before. Can you be a little nicer please. Assume we're busy, not incompetent.

VorpalBlade, 2022-08-28:
So I went digging into this AMD "supported CPU" bit. Apparently it has been supported at some points, then stopped being so:

```console
$ fwupdmgr security --force --show-all
HSI-1
✦ Supported CPU:                 Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu (obsoleted)
✔ TPM empty PCRs:                Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Tpm.EmptyPcr
✔ TPM v2.0:                      Found: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Tpm.Version20
✔ UEFI platform key:             Valid: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Uefi.Pk
✘ Supported CPU:                 Failed: https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.SupportedCpu

[...]

Host Security Events
  2022-07-16 16:53:29:  ✔ Supported CPU appeared: Valid
  2022-07-12 18:36:34:  ✔ Supported CPU appeared: Valid
  2022-07-12 18:36:34:  ✔ Supported CPU appeared: Valid
  2022-07-12 18:36:34:  ✘ Pre-boot DMA protection appeared: Disabled
  2022-03-02 18:24:16:  ✘ Encrypted RAM changed: Disabled → Unencrypted
  2022-03-02 18:24:16:  ✘ Encrypted RAM appeared: Disabled
  2022-01-13 08:21:18:  ✘ Encrypted RAM disappeared: Disabled
  2021-11-21 13:21:19:  ✘ Linux kernel lockdown changed: Invalid → Disabled
  2021-11-21 13:21:19:  ✔ All TPM PCRs are valid
$ lscpu | grep '^Model name:'
Model name:                      AMD Ryzen 5 5600X 6-Core Processor
$ dmidecode | grep 'Product Name:'
	Product Name: System Product Name
	Product Name: ROG STRIX B550-F GAMING (WI-FI)
```

This seems quite odd. Why would that happen? Why are there two entries for supported CPU (linking the same page I might add)? Why was this CPU sometimes supported, but sometimes not?

No the system disk has not moved between systems, nor has the CPU been replaced. Last UEFI update was some time back in spring, well before any of those entries.

SkewedZeppelin, 2022-08-28:
> A local attacker could overwrite the S3 resume script to modify system RAM which can lead to privilege escalation.

What script? Can it be further clarified on how s2idle is more secure/safer than s3?

hughsie, 2022-08-28:
> What script?

https://www.google.com/search?q=S3+resume+script

superm1, 2022-08-28:
> So I went digging into this AMD "supported CPU" bit. Apparently it has been supported at some points, then stopped being so:

It was not really supported at any given time.  It's just that we didn't have a lot of checks for AMD systems before the PSP check was added.  Now what fwupd does is will explicitly check that enough things have reported data to declare something supported.  On AMD systems, if you don't have the PSP exported data it's not considered enough to do a calculation.

> @hughsie What qualifies as a "newer" CPU then? This is Zen 3, which I would consider "newer"...

What Richard said is true.  You'll see it on Rembrandt/Yellow Carp (Ryzen 6xxx) as the first consumer hardware with it.  It's going to be a "reasonable" expectation that future mobile APUs and workstation CPUs support it.  The feature was developed and support added to the Linux kernel after Ryzen 5xxx was already on the market and Ryzen 6xxx was under development.

superm1, 2022-08-29:
I've filed the documentation half of my suggestion here https://github.com/fwupd/fwupd/pull/4964

poisonsnak, 2022-08-30:
I have this same issue as VorpalBlade:

>Finally it is also incorrect. I have root and swap on LVM2. The LVM2 PV is inside a dm-crypt partition. Yet fwupdmgr states that the Linux swap is unencrypted.

I guess since the wiki here https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues tells you to follow this guide to encrypt your swap https://blogs.gnome.org/hughsie/2020/05/14/converting-to-encrypted-swap/ , that's what it's looking for.  

Encrypted LVM with root and swap inside is the default behavior for the Debian installer (I'm running Debian 11) and VorpalBlade is running Arch so I guess this setup is fairly common and it would be worth checking for it.

superm1, 2022-08-30:
@poisonsnak I've opened a new issue to track the encrypted swap in this scenario here: https://github.com/fwupd/fwupd/issues/4969
Please add technical details about your configuration to that issue.

@VorpalBlade the PR I submitted #4964 is merged and will be part of tomorrow's 1.8.4 release.  We'll be referencing the public spec as part of messages now explaining the situation and falling back to the wiki page for any suggestions.  This approach is so that users can contribute new suggestions in the wiki pages without having to open PR and wait for a new release.

Also besides my change, 1.8.4 introduces multi-line descriptions on what you can do about problems that are detected.

So as such I'm going to close this issue, but I would ask that you re-check things on fwupd 1.8.4 (or git `main`) and see if you still think there are room for improvements beyond the swap issue (which is split to another issue).  If so, please open up new specific issues, discussions or make changes to the wiki pages yourself.

VorpalBlade, 2022-08-30:
@superm1 I will recheck as soon as Arch gets the new version. This usually only takes a few days, maybe a week or so.

bdaase, 2022-10-14:
> On AMD systems, if you don't have the PSP exported data it's not considered enough to do a calculation.

Sure, but as a result `fwupdmgr security` shows nothing useful at all if you don't specify `force` on these systems. Is this really the best we can do?

superm1, 2022-10-14:
> > On AMD systems, if you don't have the PSP exported data it's not considered enough to do a calculation.
> 
> Sure, but as a result `fwupdmgr security` shows nothing useful at all if you don't specify `force` on these systems. Is this really the best we can do?

Yup

SkewedZeppelin, 2022-10-14:
> You'll see it on Rembrandt/Yellow Carp (Ryzen 6xxx) as the first consumer hardware with it. It's going to be a "reasonable" expectation that future mobile APUs and workstation CPUs support it. 

Is Ryzen 7000 series expected to have the required data? or is it limited to the Pro series?
My 7950X is showing unsupported.
Or is that just due to kernel changes not yet there/pending?

superm1, 2022-10-14:
> Is Ryzen 7000 series expected to have the required data? or is it limited to the Pro series? My 7950X is showing unsupported. Or is that just due to kernel changes not yet there/pending?

It is not limited to Pro series, but I admittedly haven't checked 7000 desktop APUs myself.  What kernel version are you running?  The support lands in kernel 5.19.  If you're already on something newer than 5.19, can you show me a `dmesg`?

SkewedZeppelin, 2022-10-14:
@superm1 
see attached
[7950X-dmesg.txt](https://github.com/fwupd/fwupd/files/9786357/7950X-dmesg.txt)
[7950X-fwupd.txt](https://github.com/fwupd/fwupd/files/9786358/7950X-fwupd.txt)

superm1, 2022-10-14:
> @superm1 see attached [7950X-dmesg.txt](https://github.com/fwupd/fwupd/files/9786357/7950X-dmesg.txt) [7950X-fwupd.txt](https://github.com/fwupd/fwupd/files/9786358/7950X-fwupd.txt)

It _looks_ like the BIOS for your system doesn't allow access to the PSP capabilities register from the OS.

> ccp 0000:11:00.2: psp: unable to access the device: you might be running a broken BIOS.

superm1, 2022-10-14:
> It looks like the BIOS for your system doesn't allow access to the PSP capabilities register from the OS.

FYI - I just double checked using 6.0 kernel on a reference board for a Ryzen 7000 desktop APU and I can see the security attributes populated.  So you'll need to talk to your MB vendor about it.

```
$ dmesg | grep ccp
[    6.011955] ccp 0000:03:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[    6.022649] ccp 0000:03:00.2: tee enabled
[    6.022657] ccp 0000:03:00.2: psp enabled
$ cat  /sys/bus/pci/drivers/ccp/*/fused_part
1
```

jranke, 2022-12-28:
@superm1 I am running a recent kernel

```
Linux ryz 6.1.0-0-amd64 #1 SMP PREEMPT_DYNAMIC Debian 6.1~rc8-1~exp1 (2022-12-09) x86_64 GNU/Linux
```
on a Ryzen 7950x, on a B650M PG Riptide motherboard from ASRock. It seems that also PSP (whatever that is) can't be read:

```
$ sudo dmesg | grep ccp
[   10.765133] ccp 0000:0e:00.2: enabling device (0000 -> 0002)
[   10.765317] ccp 0000:0e:00.2: ccp: unable to access the device: you might be running a broken BIOS.
[   10.765319] ccp 0000:0e:00.2: psp: unable to access the device: you might be running a broken BIOS.
```

What exactly should I be telling my motherboard vendor to improve the situation? Or should I try to update the BIOS or look for a BIOS option?

superm1, 2022-12-28:
The PSP driver (`ccp`) has multiple functions.  Those messages are emitted when the PSP advertises support for a function but actually trying to access it doesn't work.

> What exactly should I be telling my motherboard vendor to improve the situation? Or should I try to update the BIOS or look for a BIOS option?

I don't believe there will be a BIOS option for anything like this.  It's a bug or intentional compile time decision to disable the functionality.

SkewedZeppelin, 2022-12-28:
@jranke 
> Or should I try to update the BIOS 

do this if you haven't, the November update for my board fixed/enabled it.

jranke, 2022-12-29:
@superm1 Thanks. If it is a compile time decision, it is not mine (I haven't compiled my own kernel in the last years), so I will simply trust the Debian kernel maintainers to take a reasonable decision here.

@SkewedZeppelin I updated my BIOS, and after that suspend to RAM did not work any more (the system resumed immediately after completing S3 state). So I updated the BIOS to an even later version (all of these are beta, so I did not feel very comfortable doing that). The `ccp` messages stayed the same, and suspend did not work. Finally, after the last attempt to flash the latest BIOS, the system went black shortly after starting the Instant Flash procedure and did not come back up. After waiting a minute, I did a hard reset (what else should I do), and I think the mobo was reset to factory settings, at least it came up with the original BIOS version and suspend to RAM working.

superm1, 2022-12-29:
> @superm1 Thanks. If it is a compile time decision, it is not mine (I haven't compiled my own kernel in the last years), so I will simply trust the Debian kernel maintainers to take a reasonable decision here.
> 

It's not a Debian decision. It's a compile time decision for the bios.

> @SkewedZeppelin I updated my BIOS, and after that suspend to RAM did not work any more (the system resumed immediately after completing S3 state). So I updated the BIOS to an even later version (all [...]