fwupd/fwupd #4960: Disable demand for Intel CET in `fwupdmgr security`

mhalano, 2022-08-28:
**Describe the bug**
Intel CET isn't implemented in Linux kernel yet, so even you have a compatible processor it will be marked as non-supported and you will never achieve level 3 of security.

**Steps to Reproduce**
Have a processor with Intel CET support (mine is Intel i7-11800H) and execute `fwupdmgr security` command.

**Expected behavior**
A clear and concise description of what you expected to happen.
Not be considered since there is no kernel support yet.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.3-051903-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.gusb          0.3.10

```
Native installation on Ubuntu Kinetic Kudu

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
Dell Inc. Dell G15 5511
│
├─DELL0A70:00 04F3:317E:
│     Device ID:          f0b5af888f84b1518085848a548471c338c31187
│     Summary:            Touchpad
│     Current version:    0x0005
│     Bootloader Version: 0x0001
│     Vendor:             HIDRAW:0x04F3
│     GUIDs:              db972db2-d0dd-557e-9ddd-ea0a77323371 ← HIDRAW\VEN_04F3&DEV_317E
│                         1bdaa505-7fc5-57ba-97b6-80918e9811c7 ← HIDRAW\VEN_04F3&DEV_317E&REV_00
│                         38d75275-10b3-5f2f-8c9f-a8d60111c768 ← HIDRAW\VEN_04F3&DEV_317E&MOD_0089
│                         4296cbee-f355-5bf6-b62d-a6a286b69414 ← ELANTP\ICTYPE_10
│                         4eed77b6-0567-5942-a842-2413a8622e3a ← ELANTP\ICTYPE_10&MOD_0089
│     Device Flags:       • Internal device
│                         • Updatable
│   
├─Unknown Device:
│     Device ID:          bbc98c073ac3282642278424d95dc59fcd055153
│     GUID:               c9ddd069-c7f2-58c5-b382-0487e683463a ← GPIO\ID_INT34C6:00
│   
├─11th Gen Intel Core™ i7-11800H @ 2.30GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x00000040
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         a6bd4ca5-75a6-5796-b564-66b5cab1b11b ← CPUID\PRO_0&FAM_06&MOD_8D
│                         d9dd5e77-df17-5bab-b5ec-22827598bfed ← CPUID\PRO_0&FAM_06&MOD_8D&STP_1
│     Device Flags:       • Internal device
│   
├─GA106M [GeForce RTX 3060 Mobile / Max-Q]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Current version:    a1
│     Vendor:             NVIDIA Corporation (PCI:0x10DE, PCI:0x8086)
│     GUIDs:              76567ea7-c9d9-5b64-9581-755d22386b00 ← PCI\VEN_10DE&DEV_2560
│                         8f2594a6-3673-54fd-bb97-777fa492042a ← PCI\VEN_10DE&DEV_2560&REV_A1
│                         218353e3-7021-5286-8b4f-8016bb66294e ← PCI\VEN_10DE&DEV_2560&SUBSYS_10280A70
│                         1f275bfc-d092-507a-a98e-3298382e8d63 ← PCI\VEN_10DE&DEV_2560&SUBSYS_10280A70&REV_A1
│                         c7dac7da-12ca-57be-ba1d-8617ec295689 ← PCI\VEN_8086&DEV_9A01
│                         7305c01c-2139-594e-ace6-16ab469a5d85 ← PCI\VEN_8086&DEV_9A01&REV_05
[... 115 lines cut ...]
│     Update State:       Success
│     GUID:               096da7a8-2137-4343-8d99-bdc6f640016d
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          b077e814fac8da0ca9bdf4afc7c82267b13599cd
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
    
Idle…                    [***************************************]
Successfully uploaded 1 report
```

**Additional questions**

- Operating system and version: Ubuntu Kinetic Kudu
- Have you tried rebooting? Yes
- Is this a regression? I think no (it probably never worked)

hughsie, 2022-08-28:
Is CET marked as disabled, enabled, or active? We're testing the could flags directly, not asking the kernel. The output of 'fwupdmgr security --json' would be helpful.

mhalano, 2022-08-28:
@hughsie is marked as non-supported. at least the data sheet for the processor says it should be supported. I found in the Linux kernel support for IBT, but not for Shadow Stack. Both are part of CET.
```json
{
  "SecurityAttributes" : [
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Swap",
      "Created" : 1661703827,
      "HsiLevel" : 0,
      "HsiResult" : "not-enabled",
      "Name" : "Linux swap",
      "Plugin" : "linux_swap",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Swap",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Fwupd.Plugins",
      "Created" : 1661703827,
      "HsiLevel" : 0,
      "HsiResult" : "not-tainted",
      "Name" : "fwupd plugins",
      "Plugin" : "core",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Fwupd.Plugins",
      "Flags" : [
        "success",
        "runtime-issue"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Tainted",
      "Created" : 1661703827,
      "HsiLevel" : 0,
      "HsiResult" : "tainted",
      "Name" : "Linux kernel",
      "Plugin" : "linux_tainted",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Tainted",
      "Flags" : [
        "runtime-issue",
        "action-config-os"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Kernel.Lockdown",
      "Created" : 1661703827,
      "HsiLevel" : 0,
      "HsiResult" : "not-enabled",
      "Name" : "Linux kernel lockdown",
      "Plugin" : "linux_lockdown",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Kernel.Lockdown",
      "Flags" : [
        "runtime-issue",
        "action-config-os"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.Uefi.SecureBoot",
      "Created" : 1661703827,
      "HsiLevel" : 0,
      "HsiResult" : "not-enabled",
      "Name" : "UEFI secure boot",
      "Plugin" : "uefi_capsule",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.Uefi.SecureBoot",
      "Flags" : [
        "runtime-issue",
        "action-config-fw"
      ]
    },
    {
[... 754 lines cut ...]
      "Guid" : [
        "b9a2dd81-159e-5537-a7db-e7101d164d3f",
        "30249f37-d140-5d3e-9319-186b1bd5cac3",
        "a6bd4ca5-75a6-5796-b564-66b5cab1b11b",
        "d9dd5e77-df17-5bab-b5ec-22827598bfed"
      ]
    },
    {
      "AppstreamId" : "org.fwupd.hsi.EncryptedRam",
      "Created" : 1661703827,
      "HsiLevel" : 4,
      "HsiResult" : "not-supported",
      "Name" : "Encrypted RAM",
      "Plugin" : "cpu",
      "Uri" : "https://fwupd.github.io/libfwupdplugin/hsi.html#org.fwupd.hsi.EncryptedRam",
      "Guid" : [
        "b9a2dd81-159e-5537-a7db-e7101d164d3f",
        "30249f37-d140-5d3e-9319-186b1bd5cac3",
        "a6bd4ca5-75a6-5796-b564-66b5cab1b11b",
        "d9dd5e77-df17-5bab-b5ec-22827598bfed"
      ]
    }
  ]
}
```

mhalano, 2022-08-28:
[image]
From [here](https://www.intel.com/content/www/us/en/products/sku/213803/intel-core-i711800h-processor-24m-cache-up-to-4-60-ghz/specifications.html)

pinkflames, 2022-08-28:
I'm seeing the same on a CPU that's definitely Intel CET capable. I have not tested any actual ROP gadget to verify it's working but my userspace is built with -fcf-protection=full (unless the particular package has it filtered out at distro level).

hughsie, 2022-08-28:
For the fwupd test to pass we need SHSTK and IBT; I don't see the former from your CPU output. For SHSTK we're checking CPUID 0x7 and then `ecx >> 7` -- is that right? Running `/usr/libexec/fwupd/fwupd-detect-cet` and then checking the return code (e.g. `echo $?`) will tell you if CET is functionally working.

pinkflames, 2022-08-28:
The return code is 1 which I guess in this context means a good result rather than error code 1? I also see that `grep ' ss ' /proc/cpuinfo` does list shadow stack being supported on my system. Just like ibt is supported. However it still prints "✘ Intel CET Enabled:             Not supported".

hughsie, 2022-08-28:
I thought CET was `cet_ss`, not `ss`. If fwupd-detect-cet returns with 1 it means that CET is not actually working.

pinkflames, 2022-08-28:
> I thought CET was `cet_ss`, not `ss`. If fwupd-detect-cet returns with 1 it means that CET is not actually working.

My googling efforts only found ss but, yeah, that flag has probably been around for far to long to be related to CET.

mhalano, 2022-08-28:
`ss` means "self snoop". The CPU flag is `shstk`. I'm wondering if there is a way to check if the flag exists from a place different from /proc/cpuinfo, like a memory or something. For some reason, I can't get Shadow Stack enabled on my machine, even the processor is supported. I was wondering if there is a filter since kernel doesn't seem to support Shadow Stack yet.

hughsie, 2022-08-28:
> like a memory or something

fwupd doesn't read /proc/cpuinfo and just asks the CPU directly. I wonder if it's being turned off by the mainboard or even as an option in the system firmware settings? Maybe we need to ask Intel why shstk isn't being returned?

mhalano, 2022-08-28:
I used the `cpuid` command and got this:
[image]
So, I'm wondering if the verification method used to detect if CET is enabled was changed since the `/usr/libexec/fwupd/fwupd-detect-cet` helper was written.

mhalano, 2022-08-28:
That is the feature enumeration. I don't know how to check this using cpuid in raw mode. Any tips?
[image]

hughsie, 2022-08-28:
This is what we're doing:

	guint32 ebx = 0;
	guint32 ecx = 0;
	if (!fu_cpuid(0x7, NULL, &ebx, &ecx, NULL, error))
		return FALSE;
	if ((ebx >> 20) & 0x1)
		self->flags |= FU_CPU_DEVICE_FLAG_SMAP;
	if ((ecx >> 7) & 0x1)
		self->flags |= FU_CPU_DEVICE_FLAG_SHSTK;
	if ((ecx >> 13) & 0x1)
		self->flags |= FU_CPU_DEVICE_FLAG_TME;
	if ((ecx >> 20) & 0x1)
		self->flags |= FU_CPU_DEVICE_FLAG_IBT;

fwupd-detect-cet is written to actually crash -- it was originally written my an Intel engineer. :/

hughsie, 2022-08-28:
No, but I'd certainly appreciate some debugging 🙏

mhalano, 2022-08-28:
I just can't go further. I don't know about cpuid and C enough to troubleshoot this problem. Can we contact the original author or other Intel engineer?

hughsie, 2022-08-29:
> I just can't go further

Does fwupd-detect-cet work correctly using gcc/clang -- i.e. the other one to what you're using now?

pinkflames, 2022-08-29:
Not sure why without digging into the build system but, when built with Clang 14.0.6, there is no /usr/libexec/fwupd/fwupd-detect-cet binary installed (and presumably neither built).

[raw cpuid.txt](https://github.com/fwupd/fwupd/files/9442575/raw.cpuid.txt)

[decoded cpuid.txt](https://github.com/fwupd/fwupd/files/9442577/decoded.cpuid.txt)

mhalano, 2022-08-29:
[cpuid_raw.txt](https://github.com/fwupd/fwupd/files/9443466/cpuid_raw.txt)
[cpuid_decoded.txt](https://github.com/fwupd/fwupd/files/9443470/cpuid_decoded.txt)

mhalano, 2022-08-29:
> > I just can't go further
> 
> Does fwupd-detect-cet work correctly using gcc/clang -- i.e. the other one to what you're using now?

I built using gcc, but just to get symbols to try to execute using gdb (that's not work well as I expected). I'm using Ubuntu Kinetic with fwupd installed using apt.

hughsie, 2022-08-29:
@mhalano can you compile the wip/hughsie/cpuid-cet branch and then grab the output of `sudo ./src/fwupdtool --verbose security --plugins cpu` please.

mhalano, 2022-08-29:
> `sudo ./src/fwupdtool --verbose security --plugins cpu`

I got this:
```shell
sudo ./src/fwupdtool --verbose security --plugins cpu
11:07:24:0374 FuDebug              Verbose debugging enabled (on console 1)
11:07:24:0375 FuMain               locked /run/lock/fwupdtool
11:07:24:0377 11:07:24:0377 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
11:07:24:0381 FuMain               Failed to stop daemon: failed to find fwupd.service: GDBus.Error:org.freedesktop.systemd1.NoSuchUnit: Unit fwupd.service not loaded.
Loading…                 [-                                      ]11:07:24:0381 FuConfig             trying to load config values from /usr/local/etc/fwupd/daemon.conf
11:07:24:0381 FuConfig             trying to load config values from /var/local/etc/fwupd/daemon.conf
11:07:24:0381 FuConfig             using autodetected max archive size 18,4 EB
11:07:24:0381 FuConfig             failed to read UpdateMotd key: Key file does not have group “fwupd”
11:07:24:0381 FuConfig             failed to read EnumerateAllDevices key: Key file does not have group “fwupd”
11:07:24:0381 FuConfig             failed to read IgnorePower key: Key file does not have group “fwupd”
11:07:24:0381 FuConfig             failed to read OnlyTrusted key: Key file does not have group “fwupd”
11:07:24:0381 FuConfig             failed to read ShowDevicePrivate key: Key file does not have group “fwupd”
11:07:24:0382 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
Loading…                 [ -                                     ]11:07:24:0382 XbSilo               attempting to load /var/local/cache/fwupd/metainfo.xmlb
11:07:24:0382 XbSilo               file: 7599f7fc-9fb2-2ac0-0210-28638eec197e, current:7599f7fc-9fb2-2ac0-0210-28638eec197e, cached: (null)
11:07:24:0382 XbSilo               loading silo with file contents
11:07:24:0382 FuRemoteList         path /usr/local/etc/fwupd/remotes.d does not exist
Loading…                 [                                       ]11:07:24:0383 Jcat                 reading /var/local/lib/fwupd/pki/secret.key with 2455 bytes
11:07:24:0383 Jcat                 reading /var/local/lib/fwupd/pki/client.pem with 1383 bytes
11:07:24:0383 FuEngine             client certificate now exists: nothing to do
Loading…                 [                                       ]11:07:24:0383 FuHistory            trying to open database '/var/local/lib/fwupd/pending.db'
11:07:24:0383 FuHistory            got schema version of 8
Loading…                 [*                                      ]Failed to load plugins: Error opening directory “/usr/local/lib/x86_64-linux-gnu/fwupd-plugins-7”: No such file or directory
```

mhalano, 2022-08-29:
[plugin-cpu.txt](https://github.com/fwupd/fwupd/files/9444033/plugin-cpu.txt)
I got this after compile and install (without install it didn't find the plugin)

hughsie, 2022-08-29:
Aha! We're checking ECX for the IBT check, and intel specified E**D**X. Can you try https://github.com/fwupd/fwupd/pull/4963 please?

mhalano, 2022-08-29:
[plugin-cpu-with-fix.txt](https://github.com/fwupd/fwupd/files/9444910/plugin-cpu-with-fix.txt)
This is the result of command `sudo ./src/fwupdtool --verbose security --plugins cpu` using the branch mentioned on #4963.
I specially worried about Shadow Stack part, because there is no CPU flag for me, even `cpuid` command says that is supported.

hughsie, 2022-08-29:
Did you `ninja install`?

mhalano, 2022-08-29:
I just did. I will reupload the log. My mistake.

mhalano, 2022-08-29:
[image]
Success!

mhalano, 2022-08-29:
@hughsie I will wait the next release (that I think will include the fix) to close the ticket. That is ok for you?

hughsie, 2022-08-29:
Lets close now; I'm tagging a new release tomorrow morning.

closed 2022-08-29 by hughsie

mhalano, 2022-08-29:
Just one final question: do we have any tests about the execution in an older processor, where Intel CET is naturally unavailable?

mhalano, 2022-08-29:
I used KVM to boot a Ubuntu 22.04.1 machine and configured to use an old processor (core2duo). I got what I was expecting. 
[image]
[image]

pinkflames, 2022-08-29:
FWIW, the patch solved this for me, too.