fwupd/fwupd #4961: TPM PCR0 reconstruction: Not found despite being available

Tibladar, 2022-08-28:
**Describe the bug**
`fwupdmgr security --force` shows `TPM PCR0 reconstruction: Not found` but I can query PCR0 just fine via `tpm2_pcrread` and `fwupdtpmevlog`

```
$ sudo fwupdmgr security --force
WARNING: UEFI capsule updates not available or enabled in firmware setup
  See https://github.com/fwupd/fwupd/wiki/PluginFlag:capsules-unsupported for more information.
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Gefunden
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✘ TPM PCR0 reconstruction:       Not found

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Failed

Runtime Suffix -!
✔ Linux kernel lockdown:         Enabled
✔ Linux swap:                    Disabled
✔ fwupd plugins:                 Untainted
✘ Linux kernel:                  Tainted

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues

Host Security Events
  2022-08-28 16:06:14:  ✔ Supported CPU appeared: Valid
  2022-08-28 16:06:14:  ✘ Pre-boot DMA protection appeared: Disabled
  2022-08-28 13:44:57:  ✔ All TPM PCRs are valid
  2022-08-28 13:44:57:  ✘ TPM PCR0 reconstruction appeared: Not found
```

<details><summary>tpm2_pcrread</summary>

```
  sha1:
  sha256:
    0 : 0x9E9FBB7718458953D273CBBFD56B3A24C786726FDD29F1026CAD7D1CC3E138D1
    1 : 0xB7A16486DB864817BFBF8BA9A260647ABA9C716489D36D7266303763CB845A1D
    2 : 0x74EAC3CA90F0D0FAB4CAADB4B35264A41411B7EF8A2ACF23A44744F0462133C6
    3 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    4 : 0xFAD3643BD7DAF0226A1A2B334BE24D0B6A4CD6598D9D0552EEEAE4130F8DB636
    5 : 0x4EF2DFFA37394D60F1DE68617CAC73A2FBCFF7E769322D846769D0F54664053E
    6 : 0x3D458CFE55CC03EA1F443F1562BEEC8DF51C75E14A9FCF9A7234A13F198E7969
    7 : 0x49A41AC0BCB309FF4A6A6676744C26FC4977AE5F1F0F7A6A26B52DE58C2A3248
    8 : 0x52A0B7B83D6337037DDF67D18AAF9DEFCF55D0257DE7D287D2BC4136531C6AD1
    9 : 0x5147B713844D00DADDA2EDDCE9C3ABDC59C1CA476A19DC283AAC46A3DA51C3E9
    10: 0xEE20BC32679F01EF9BC7BF1DC980839F41E8BDB4A6AE6526020514F0AE0D8368
    11: 0x0000000000000000000000000000000000000000000000000000000000000000
    12: 0x0000000000000000000000000000000000000000000000000000000000000000
    13: 0x0000000000000000000000000000000000000000000000000000000000000000
    14: 0x60000A3F87D3D724F4DFB2CF978C554FCCD2FE09B7A00C9C5358E9E117D09D7D
    15: 0x0000000000000000000000000000000000000000000000000000000000000000
    16: 0x0000000000000000000000000000000000000000000000000000000000000000
    17: 0xFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
[... 659 lines cut ...]
│   
└─UEFI dbx:
      Device ID:          362301da643102b9f38477387e2193e57abaa590
      Summary:            UEFI revocation database
      Current version:    241
      Minimum Version:    241
      Vendor:             UEFI:Linux Foundation
      Install Duration:   1 second
      GUID:               c6682ade-b5ec-57c4-b687-676351208742 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503
                          f8ba2887-9411-5c36-9cee-88995bb39731 ← UEFI\CRT_A1117F516A32CEFCBA3F2D1ACE10A87972FD6BBE8FE0D0B996E09E65D802A503&ARCH_X64
                          6d411c5b-df6a-5910-ab91-fd9ae4c19242 ← UEFI\CRT_2489B9C54D86350473D6DC6FEEBE60D1B7D0FFF7B3E5978658EDC53E1F41A767
                          88d49691-fc31-5b47-94a1-a4b045be570e ← UEFI\CRT_2489B9C54D86350473D6DC6FEEBE60D1B7D0FFF7B3E5978658EDC53E1F41A767&ARCH_X64
      Device Flags:       • Internal device
                          • Updatable
                          • Needs a reboot after installation
                          • Only version upgrades are allowed
                          • Signed Payload
```
</details>

**Additional questions**

- Operating system and version: Linux Mint 21
- Have you tried rebooting? Yes
- Is this a regression? Don't know

hughsie, 2022-08-28:
We only get `not-found` when this is true:

	/* check reconstructed to PCR0 */
	if (priv->ev_items == NULL || priv->bios_device == NULL) {
		fwupd_security_attr_set_result(attr, FWUPD_SECURITY_ATTR_RESULT_NOT_FOUND);
		return;
	}

Does `/sys/kernel/security/tpm0/binary_bios_measurements` exist for you?

superm1, 2022-08-28:
It looks to me the problem is probably that you didn't get a BIOS device set that exports `main-system-firmware`.  I guess your system doesn't have UEFI capsule updates available in the BIOS setup?

I tend to think we should loosen this check.  We can construct the PCR and compare it against the TPM without the UEFI capsule device can't we?

Tibladar, 2022-08-29:
> Does `/sys/kernel/security/tpm0/binary_bios_measurements` exist for you?

Yes it exists

> I guess your system doesn't have UEFI capsule updates available in the BIOS setup?

I couldn't find it in my MSI bios (yet?), so yes

PR #4962 (2022-08-29): tpm: Don't require uefi capsule updates for checking TPM PCR0
There isn't any reason that we need to have a system advertising a
GUID with `main-system-firmware` in order to check that PCR0 reconstruction
matches what the TPM actually has.

Fixes: #4961

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [ ] Code fix
- [ ] Feature
- [ ] Documentation
commit 1f7526cc (2022-08-29): tpm: Don't require uefi capsule updates for checking TPM PCR0

There isn't any reason that we need to have a system advertising a
GUID with `main-system-firmware` in order to check that PCR0 reconstruction
matches what the TPM actually has.

Fixes: #4961
closed 2022-08-29 by hughsie