fwupd/fwupd #4969: Detect encrypted swap in LVM2

superm1, 2022-08-30:
It was mentioned in another issue that encrypted swap detection is failing in a common configuration on Debian 11.  We should modify our detection mechanism to pick this up:
 
>Finally it is also incorrect. I have root and swap on LVM2. The LVM2 PV is inside a dm-crypt partition. Yet fwupdmgr states that the Linux swap is unencrypted.

I guess since the wiki here https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues tells you to follow this guide to encrypt your swap https://blogs.gnome.org/hughsie/2020/05/14/converting-to-encrypted-swap/ , that's what it's looking for.  

Encrypted LVM with root and swap inside is the default behavior for the Debian installer if you ask it to encrypt your drive (I'm running Debian 11) and VorpalBlade is running Arch so I guess this setup is fairly common and it would be worth checking for it.

_Originally posted by @poisonsnak in https://github.com/fwupd/fwupd/issues/4959#issuecomment-1231016110_

poisonsnak, 2022-08-30:
Thanks for the quick reply! While this happens on Debian 11, it only has fwupd version 1.5.7 so I used my Debian Testing machine which has fwupd version 1.8.3.  I figured you'd rather see results from a newer version. If you want to see results from Debian 11 with fwupd 1.5.7 let me know and I can post them too.  They both have an encrypted LVM with root and swap inside and fwupdmgr security says swap is not encrypted for both of them.

Here is the output I get:
```
$ fwupdmgr security --force
Host Security ID: HSI:INVALID:missing-data

HSI-1
✔ TPM empty PCRs:                Valid
✔ TPM v2.0:                      Found
✔ UEFI platform key:             Valid
✔ UEFI secure boot:              Enabled
✘ Supported CPU:                 Failed

HSI-2
✔ IOMMU:                         Enabled
✔ TPM PCR0 reconstruction:       Valid

HSI-3
✘ Pre-boot DMA protection:       Disabled
✘ Suspend-to-idle:               Disabled
✘ Suspend-to-ram:                Enabled

HSI-4
✘ Encrypted RAM:                 Not supported

Runtime Suffix -!
✔ Linux kernel:                  Untainted
✔ Linux kernel lockdown:         Enabled
✔ fwupd plugins:                 Untainted
✘ Linux swap:                    Unencrypted

This system has a low HSI security level.
 » https://github.com/fwupd/fwupd/wiki/Low-host-security-level

This system has HSI runtime issues.
 » https://github.com/fwupd/fwupd/wiki/Host-security-ID-runtime-issues

Host Security Events
  2022-08-30 03:31:35:  ✔ Supported CPU appeared: Valid
  2022-08-30 03:31:35:  ✘ Pre-boot DMA protection appeared: Disabled
  2022-04-27 11:03:04:  ✘ Encrypted RAM changed: Disabled → Not supported
  2022-04-27 10:48:31:  ✔ Kernel lockdown enabled
  2022-04-27 10:48:31:  ✔ Secure Boot enabled
  2022-04-02 16:49:56:  ✘ Kernel lockdown disabled
  2022-04-02 16:49:56:  ✘ UEFI secure boot changed: Not found → Disabled
```
My disk setup:
```

$ lsblk
NAME                        MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
nvme0n1                     259:0    0 465.8G  0 disk
├─nvme0n1p1                 259:1    0   499M  0 part
├─nvme0n1p2                 259:2    0   100M  0 part  /boot/efi
├─nvme0n1p3                 259:3    0    16M  0 part
├─nvme0n1p4                 259:4    0 367.5G  0 part
├─nvme0n1p5                 259:5    0   954M  0 part  /boot
└─nvme0n1p6                 259:6    0  96.7G  0 part
  └─nvme0n1p6_crypt         254:0    0  96.7G  0 crypt
    ├─tygra--vg-tygra--swap 254:1    0   1.9G  0 lvm   [SWAP]
    └─tygra--vg-tygra--root 254:2    0  94.8G  0 lvm   /
```
fwupd version:
```

$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.3
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
[... 111 lines cut ...]
│     Device ID:          349bb341230b1a86e5effe7dfe4337e1590227bd
│     Summary:            UEFI ESRT device
│     Current version:    335675411
│     Vendor:             DMI:American Megatrends Inc.
│     Update State:       Success
│     GUID:               f766f6e6-b43d-4acd-a4bd-80ff2f0af5cc
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─UEFI Device Firmware:
      Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
      Summary:            UEFI ESRT device
      Current version:    196608
      Vendor:             DMI:American Megatrends Inc.
      Update State:       Success
      GUID:               f5536e63-e4c0-4e0d-84d4-e8e152b1ba65
      Device Flags:       • Internal device
                          • Updatable
                          • System requires external power source
                          • Needs a reboot after installation
                          • Device is usable for the duration of the update
```

PhilippWendler, 2022-08-30:
Note that this is the same as #3176, which was closed as "won't fix". There it was also mentioned that the same is the default for encrypted Ubuntu installations. Nice to see that it is planned that this gets implemented. :+1:

VorpalBlade, 2022-08-30:
It seems that this setup is fairly common on Debian and Debian derived distros, which I understand is one of the more common Linux distros for end users (i.e. not corporate deployments of RH/CentOS/whatever-the-centos-replacement-was-called).

I'm using Arch and also using this setup, though as this is Arch you are basically rolling your own install. My setup is as follows:

* Unencrypted EFI partition
* Unencrypted Windows NTFS partition
* Unencrypted boot partition (ext4, for grub)
* cryptsetup LUKS volume
  * LVM2 VG consisting of a single LVM2 PV which is inside the LUKS volume
    * root LV: btrfs
    * swap LV: swap

swap (as well as all other partitions) are specified in fstab via UUID.

Is there any other information that would be of use to you?

cesarb, 2022-08-30:
This is also the default configuration for encrypted Fedora installations, or at least used to be back when it used swap on disk instead of zram. Upgrading an encrypted Fedora install from back then will keep that swap-on-LVM-on-LUKS configuration.

hughsie, 2023-05-07:
@julianfairfax want to have a go at fixing it? I think it's fair people using other distros contribute the needed patches.

MateusRodCosta, 2023-06-28:
Please also support [BTRFS swapfiles inside a subvolume](https://wiki.archlinux.org/title/Btrfs#Swap_file)!

hughsie, 2023-06-28:
I don't run any btrfs systems I'm afraid. If you want it to be detected someone else other than me has to write the code.

superm1, 2023-06-28:
Same with me, I don't use btrfs.

superm1, 2024-02-15:
Those are not related.

openmindead, 2024-02-15:
>Does hibernate not work _because_ fwupdmgr fails to detect encrypted swap? What is it on an Ubuntu system that is calling fwupdmgr and disabling hibernate as a result?

You can't hibernate (probably) because of lockdown mode triggered by Secure Boot. Fwupdmgr message is just wrong. Both things are not related with each other.

heeen, 2024-09-16:
is there any way to manually convince my system that the swap partition in my crypto_LUKS volume is actually encrypted?

hughsie, 2024-09-16:
Someone needs to write the code to detect that I'm afraid.

heeen, 2024-09-17:
same question I asked here I guess https://github.com/fwupd/fwupd/issues/4969#issuecomment-2354998005

hughsie, 2024-09-17:
> who is denying my request to hibernate

Not fwupd.

superm1, 2024-09-17:
> > who is denying my request to hibernate
> 
> Not fwupd.

Specifically the kernel does this when lockdown is turned on. 

Some Linux distributions tie lockdown with secure boot, but the upstream Linux kernel does not.

heeen, 2024-09-17:
So who tells the kernel the verdict on the swap encryption situation?

superm1, 2024-09-17:
> So who tells the kernel the verdict on the swap encryption situation?

No one. The kernel turns off hibernate because it can't know for sure.

heeen, 2024-09-17:
But clearly some configurations allow hibernate. Can you outline how this comes to pass?

superm1, 2024-09-17:
> But clearly some configurations allow hibernate. Can you outline how this comes to pass?

If using a distro that ties secure boot to lockdown then turn off secure boot.

If using lockdown for any other reason turn off lockdown.

Lawstorant, 2025-07-17:
I'd like to add to this thread. Most people here are using the installer defaults, but another configuration to consider is just running a swapfile in an encrypted root partition in LVM-on-LUKS. This, again, provides full swap encryption but shows as unencrypted swap in fwupd.

superm1, 2025-07-17:
To me it sure would be nice if we had a hint from udisks whether the swap has an encrypted backing.