fwupd/fwupd #4976: 1.8.4: meson fails with `Feature docs cannot be enabled`

kloczek, 2022-08-30:
**Describe the bug**
meson fails with `Feature docs cannot be enabled` error message

**Steps to Reproduce**
Run meson with `-D docs=enabled`

**Expected behavior**
Meson should not fail or it shoud show clear message what is wrong.

**fwupd version information**
1.8.4

```shell
N/A

**fwupd device information**
N/A

**Additional questions**
- Operating system and version: Linux x86/64

Meson fails witth message
```console
Run-time dependency flashrom found: YES unknown
Run-time dependency systemd found: YES 251
Dependency systemd found: YES 251 (cached)
Configuring config.h using configuration
Run-time dependency libprotobuf-c found: YES 1.4.1
Program protoc found: YES (/usr/bin/protoc)

meson.build:463:0: ERROR: Feature docs cannot be enabled

A full log can be found at /home/tkloczko/rpmbuild/BUILD/fwupd-1.8.4/x86_64-redhat-linux-gnu/meson-logs/meson-log.txt
```
In meson-log.txt I was able to find
```console
Running command: /usr/bin/python3 docs/test-deps.py
--- stdout ---

--- stderr ---
Traceback (most recent call last):
  File "docs/test-deps.py", line 7, in <module>
    from packaging.version import Version
ModuleNotFoundError: No module named 'packaging'
```
Meson now provides routines allowing check presence of exact python modules so using custom scrips is no loger needed
https://mesonbuild.com/Python-module.html

hughsie, 2022-08-30:
@kloczek can you specify the modules as an auto-feature like we're doing here?

kloczek, 2022-08-31:
> @kloczek can you specify the modules as an auto-feature like we're doing here?

In tbis case is executed pythons script which requires `packaging` python module installes.
IMO it should be added checing is that module installed before executing that script. Checking can be done using meson build in python module checker.

eli-schwartz, 2022-08-31:
The actual error comes from a completely different line than the one shown in your error paste:

https://github.com/fwupd/fwupd/blob/dbd6515e7b0d178057c7f65dd4d39ed8e43c1eca/meson.build#L544-L545

`r.stdout()` can be empty, because:
https://github.com/fwupd/fwupd/blob/dbd6515e7b0d178057c7f65dd4d39ed8e43c1eca/docs/test-deps.py#L7-L12

This should check if `packaging` is installed too, with a try/except, and printing a suitable message to ***stdout***.

It should also probably not assume `markdown` is installed, of whatever version.

kloczek, 2022-08-31:
> This should check if `packaging` is installed too, with a try/except, and printing a suitable message to _**stdout**_.
> 
> It should also probably not assume `markdown` is installed, of whatever version.

So instead introducing custom sripts it shluld be useed meson native python modules checker.

eli-schwartz, 2022-08-31:
> it shluld be useed meson native python modules checker

There is no native python modules checker.

There is an imports checker, but that doesn't handle versions anyway.

kloczek, 2022-08-31:
> There is no native python modules checker.
> 
> There is an imports checker, but that doesn't handle versions anyway.

And that imprt checker faiils with error message if module is not installed
Example
https://gitlab.gnome.org/GNOME/gi-docgen/-/blob/main/meson.build#L10-20

eli-schwartz, 2022-08-31:
I fail to see how that has anything to do with what I said.

Once again:

The import checker in Meson doesn't do what fwupd needs. Therefore, it:
- means nothing
- doesn't help
- runs slower
- bloats the code
- adds additional points of failure since it cannot single-source the list of checks

My official recommendation as a core developer of Meson is to ***not*** do that, and instead to fix the existing python script.

Sometime next year we will add new API for a python distributions checker with version support.

closed 2022-08-31 by superm1

kloczek, 2022-09-01:
Just checked https://github.com/fwupd/fwupd/blob/main/docs/test-deps.py
and 
https://github.com/fwupd/fwupd/blob/ff06f98a38b76f71ad61776621eb4c4d783494e8/meson.build#L20-L31

In meson already is checking `dependency('gi-docgen')`. Than it is done checking `find_program('gi-docgen')` (which is part of that package). And at the end using custom script is done checking python module which should be installed because already checked `dependency('gi-cocgen`)`.

Don't get me wrong but this looks like old joke with drunk fisherman who cached gold fish 😋 

Gold fish: release me and you will have 3 wishes in exchange
Fisherman: OK firs whish: I want to have two railroad tanks of vodka 
G: Done. What is your 2nd wish?
F: OK. Please change to the vodka all water in that lake
G: Done. And what is 3rd wish? 🤔 
F: OK .. give me bottle of vodka and we will be done.

Please check any other project which is using `gi-docgen` Is anyone is doing combination like here
```console
[tkloczko@pers-jacek SPECS]$ grep 'pkgconfig(gi-docgen)' *spec -l |xargs grep ^VCS
aravis.spec:VCS:                https://github.com/AravisProject/aravis/
calls.spec:VCS:         https://source.puri.sm/Librem5/calls/
devhelp.spec:VCS:               https://gitlab.gnome.org/GNOME/devhelp/
eog.spec:VCS:           https://gitlab.gnome.org/GNOME/eog/
evince.spec:VCS:                https://gitlab.gnome.org/GNOME/evince/
fwupd.spec:VCS:         https://github.com/fwupd/fwupd/
gcr.spec:VCS:           https://gitlab.gnome.org/GNOME/gcr/
ghex.spec:VCS:          https://gitlab.gnome.org/GNOME/ghex/
gssdp.spec:VCS:         https://gitlab.gnome.org/GNOME/gssdp/
gtk.spec:VCS:           https://gitlab.gnome.org/GNOME/gtk/
gupnp.spec:VCS:         https://gitlab.gnome.org/GNOME/gupnp/
libadwaita.spec:VCS:            https://gitlab.gnome.org/GNOME/libadwaita/
libhandy.spec:VCS:              https://gitlab.gnome.org/GNOME/libhandy/
libpeas.spec:VCS:               https://gitlab.gnome.org/GNOME/libpeas/
libportal.spec:VCS:             https://github.com/flatpak/libportal/
libsecret.spec:VCS:             https://gitlab.gnome.org/GNOME/libsecret/
libsoup.spec:VCS:               https://gitlab.gnome.org/GNOME/libsoup/
mutter.spec:VCS:                https://gitlab.gnome.org/GNOME/mutter/
rest.spec:VCS:          https://gitlab.gnome.org/GNOME/librest/
vte.spec:VCS:           https://gitlab.gnome.org/GNOME/vte/
```

Generally issue is that gi-docgen is not using `packaging` module. Just checked my rpm package autogenerated Requires dependencies
```
/usr/bin/python3
python(abi) = 3.8
python3dist(jinja2)
python3dist(markdown)
python3dist(pygments)
python3dist(toml)
python3dist(typogrify)
```
```console
[tkloczko@pers-jacek gi-docgen-2022.1]$ grep -r packaging
.gitignore:# Distribution / packaging
.reuse/dep5:Format: https://www.debian.org/doc/packaging-manuals/copyright-format/1.0/
LICENSES/GPL-3.0-or-later.txt:packaging a Major Component, but which is not part of that Major
meson_options.txt:       description : 'run tests that are appropriate during gi-docgen development but not during packaging')
```
As you see there is here import lines with `packaging`

All what really needed here is single line with `dependency('gi-cocgen', ...)`.

kloczek, 2022-09-01:
Please consider use patch like below.
```patch
--- a/meson.build
+++ b/meson.build
@@ -459,15 +459,9 @@

 fwupd_gir = []
 introspection = get_option('introspection').disable_auto_if(host_machine.system() != 'linux')
-r = run_command([python3, 'docs/test-deps.py'])
-docs_python_deps = get_option('docs').require(r.returncode() == 0, error_message: r.stdout())
-gidocgen_dep = dependency('gi-docgen',
-  version: '>= 2021.1',
-  native: true,
-  fallback: ['gi-docgen', 'dummy_dep'],
-  required: docs_python_deps,
-)
-gidocgen_app = find_program('gi-docgen', required: get_option('docs'))
+if dependency('gi-docgen', version: '>= 2021.1', required: get_option('docs')).found()
+  subdir('docs')
+endif

 # using "meson configure -Db_sanitize=address,undefined" is super useful in finding corruption,
 # but it does not work with our GMainContext-abuse tests...
@@ -496,7 +495,6 @@
   subdir('src')

 endif
-subdir('docs')

 if build_daemon and libsystemd.found()
   install_symlink('fwupd-offline-update.service',
```
This is only proposal showing in which direction actual fix should go. Please remember docs/meson.build needs to be adjusted (it can be by this a bit simpler) as well.

hughsie, 2022-09-01:
I think the problem is that sometimes gi-docgen is provided by a system package, and sometimes it's being provided by a subproject.

superm1, 2022-09-01:
Shouldn't the minimum markdown version apply even as a system package? I guess so but instead it's using distro package manager to evaluate.

kloczek, 2022-09-01:
> I think the problem is that sometimes gi-docgen is provided by a system package, and sometimes it's being provided by a subproject.

Generally speaking such fusses of the build evn should not be handled by any meson.build definition.
If for example some project depends on 'pkgconfig(cairo)` and at the same time someone deleted in that env `glib-2.0.pc` or will have some gaps in installed glib header files it is not cairo maintainer responsibilities to check that in cairo meson.build build procedure definition. That is only what I want to say ..

`gi-docgen` and its own install time dependencies may change from version to version however if exact project which uses/needs 'gi-doocgen` will be checking only for `dependency(`gi-docgen', ..)` taht still should be OK.

PS. Hopefully no one took to seriously or personally that my joke 😋

kloczek, 2022-09-01:
> Shouldn't the minimum markdown version apply even as a system package? I guess so but instead it's using distro package manager to evaluate.

Hmm .. I've just realised that it is kind of gap in pkgconfig definition. .. 🤔 
In current definition of what is possible to use the  .pc file `Requires:`  field it is not possible to put python modules dependencies. I think that it would/could be good to flag that as RFE (will try do that).
Hmm .. this could be even stretched/extended to rust, perl, R, ocaml, erlang, nodejs, java WhateverElseWillComeToYourMind™️  🤔 

That direction could be explored as long term solution. For now however it is IMO a bit better to assume that if project A needs `pkgconfig(X)` somehow all python dependencies of the `pkgconfig(X)`  have been installed correctly by someone who crafted build env.
Fullfill that requirement is quite easy to reach even in currently state if to assembly build env where used packages like rpm or deb used (only).
Nevertheless if on typical FreeStyle™️  development package management is not used more or less that is already fading away ..

superm1, 2022-09-01:
> That direction could be explored as long term solution.

To be honest I don't see what's wrong with the state of things after my PR yesterday. We won't allow you to turn on docs unless we can verify both the markdown version and gi-docgen version.

kloczek, 2022-09-01:
> To be honest I don't see what's wrong with the state of things after my PR yesterday. We won't allow you to turn on docs unless we can verify both the markdown version and gi-docgen version.

As I wrote in https://github.com/fwupd/fwupd/issues/4976#issuecomment-1233776230 latest 'gi-docgen` does not uses `packaging` python module.

superm1, 2022-09-01:
But *we* require it for fwupd. If *we* can't verify the markdown version somehow we won't let you enable docs.

kloczek, 2022-09-01:
> But _we_ require it for fwupd. If _we_ can't verify the markdown version somehow we won't let you enable docs.

Are you sure?
```console
[tkloczko@devel-g2v fwupd-1.8.4]$ grep -r packaging | grep import
contrib/qubes/src/qubes_fwupd_heads.py:from packaging.version import Version
contrib/qubes/src/qubes_fwupdmgr.py:from packaging.version import Version
contrib/qubes/test/test_qubes_fwupdmgr.py:from packaging.version import Version
docs/test-deps.py:from packaging.version import Version
```
In the script
```python
#!/usr/bin/python3
# SPDX-License-Identifier: LGPL-2.1+

import sys
import markdown

from packaging.version import Version

# https://github.com/fwupd/fwupd/pull/3337#issuecomment-858947695
if Version(markdown.__version__) < Version("3.3.3"):
    print("python3-markdown version 3.3.3 required for gi-docgen")
    sys.exit(1)

# success
sys.exit(0)
```
`packaging` is used to verify `markdown` presence.
As I wrote that could be without any risk using meson python module importer/checker.
If it will be done that way `packaging` will be not requires.
As well as I already wrote checking `gi-docgen` python dependencies is like additionally checking glib if cairo is needed.

And completely on the side comment not directly connected to this issue. Instead `packaging` and `pkg_resources` better is to use `importlib.metadata` because AFAIK both modules reads all python modules metadata when `modules metadata` reads only requested module metadata.
https://github.com/pypa/setuptools/issues/3279
Other thing is that to check `markdown` version can be done with`markdown.__version_info__`. All without use `packaging`.

As I wrote IMO it should be only used something like ..
```python
if dependency('gi-docgen', version: '>= 2021.1', required: get_option('docs')).found()
  subdir('docs')
endif`
```

kloczek, 2022-09-01:
> Other thing is that to check `markdown` version checking can be done with`markdown.__version_info__`. All without use `packaging`.

That is not true because it is not only about string compare .. my fault.

kloczek, 2022-09-01:
Checked few other projects hew they are checking `gi-docgen` (I've checking from those which have been recently released)

mutter
https://gitlab.gnome.org/GNOME/mutter/-/blob/main/meson.build#L299-303
```meson
have_documentation = get_option('docs')
if have_documentation
  gidocgen_dep = dependency('gi-docgen', version: '>= 2021.1',
                            fallback: ['gi-docgen', 'dummy_dep'])
endif
```
libsoup
https://gitlab.gnome.org/GNOME/libsoup/-/blob/master/docs/reference/meson.build#L14-18
```meson
gidocgen_dep = dependency('gi-docgen', version: '>= 2021.1',
  fallback: ['gi-docgen', 'dummy_dep'],
  native: true,
  required: get_option('docs')
)
```
libadwaita
https://gitlab.gnome.org/GNOME/libadwaita/-/blob/main/doc/meson.build#L27-31
```meson
dependency('gi-docgen', version: '>= 2021.1',
           fallback: ['gi-docgen', 'dummy_dep'],
           native: true,
           required: get_option('gtk_doc'))
```
In few other packages is checked only exact program provided by `gi-docgen` without using `dependency()`

superm1, 2022-09-01:
> In few other packages is checked only exact program provided by gi-docgen without using dependency()

None of those are checking markdown version.  We have a bug that we found in a certain version of markdown so we need to protect ourselves from it.

> packaging is used to verify markdown presence in exact minimum version.
As I wrote that could be without any risk using meson python module importer/checker.

No, you can't get and compare the version this way.

I wrote a PR that I tagged you in, but I'm stil not convinced it's the right direction.

kloczek, 2022-09-01:
> None of those are checking markdown version. We have a bug that we found in a certain version of markdown so we need to protect ourselves from it.

As I wrote this should be handled in `gi-docgen` dependencies because thi tool will be not working corectly probabnly with any other propject which is using it. 

Q: Do you really expect that all projects which are using `gi-docgen` should take care of checking `markdown` version before use it? 🤔

kloczek, 2022-09-01:
And not to mention that `gi-docgen` effectively uses four python modules ..

superm1, 2022-09-01:
> As I wrote this should be handled in gi-docgen dependencies because thi tool will be not working corectly probabnly with any other propject which is using it.

Sure.  If you want to file a bug with gi-docgen to get them to add the check then maybe we can drop this some day when we require a gi-docgen version that explicitly validates it.

> Q: Do you really expect that all projects which are using gi-docgen should take care of checking markdown version before use it?

TBH I don't know if the markdown issue is exclusive to our use of `gi-docgen`, a bug in `gi-docgen` or a bug in markdown.  All I know is that it fails with an older version and I need to protect fwupd from triggering it.  If you can tell me that `gi-docgen` X.Y.Z works with the older markdown and I can reproduce it, I'm fine to drop the check and instead require that version of gi-docgen instead.

kloczek, 2022-09-01:
Just FTR: I've opened RFE https://gitlab.freedesktop.org/pkg-config/pkg-config/-/issues/73
If you have anything to add feel free to add comment.

eli-schwartz, 2022-09-01:
> TBH I don't know if the markdown issue is exclusive to our use of `gi-docgen`, a bug in `gi-docgen` or a bug in markdown. All I know is that it fails with an older version and I need to protect [...]

It's not a bug in anything at all, gi-docgen just happens to use a newer feature of markdown and needs a specific minimum version regardless.

Also, that version is not the one you have listed.

Anyway, I implemented that in https://gitlab.gnome.org/GNOME/gi-docgen/-/merge_requests/166

kloczek, 2022-09-01:
> Anyway, I implemented that in https://gitlab.gnome.org/GNOME/gi-docgen/-/merge_requests/166

LGTM as at least kind of partial mitigation applied in right point 👍 
However as long term solution IMO it would be good to have possiibility to define python depenencies in .pc files because it will solve entire classs of similar issues (not a single most annoing case).

eli-schwartz, 2022-09-01:
It's not a partial mitigation, it's exactly the correct way to do it. That is also the metadata that guarantees that e.g. `pip install` pulls in an updated version of Markdown, and rpm will collect that into a versioned dependency on markdown as well: https://docs.fedoraproject.org/en-US/packaging-guidelines/Python/#Automatically-generated-dependencies

PR #4983 (2022-08-31): trivial: add an explicit check for python's packaging module
Fixes: #4976
Suggested-by: eschwartz93@gmail.com

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 473459c9 (2022-08-31): trivial: add an explicit check for python's packaging module

Fixes: #4976
Suggested-by: eschwartz93@gmail.com
commit d8be7e50 (2022-09-01): trivial: Allow satisfying markdown check with packaging or distutils

Some distributions are opposed to checking the version of something
using modern tools.

Fixes #4976