fwupd/fwupd #4977: Segfault when probing complete in plugin

DylanVanAssche, 2022-08-30:
**Describe the question**
I'm working on a new plugin and I get a segfault after my probing method is finished.
The segfault happens in the `fu-engine` according to gdb, but I can't see what I'm missing

**fwupd version information**

1.8.3

**gdb backtrace**

```
shift-axolotl:/home/dylan# export FWUPD_PROBE_VERBOSE=1
shift-axolotl:/home/dylan# gdb --args /usr/libexec/fwupd/fwupd -v
GNU gdb (GDB) 12.1
Copyright (C) 2022 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-alpine-linux-musl".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/libexec/fwupd/fwupd...
Reading symbols from /usr/lib/debug//usr/libexec/fwupd/fwupd.debug...
(gdb) r
Starting program: /usr/libexec/fwupd/fwupd -v

Program received signal SIGILL, Illegal instruction.
0x0000fffff73a5dd0 in ?? () from /lib/libcrypto.so.3
(gdb) c
Continuing.
14:08:10:0287 FuDebug              Verbose debugging enabled (on console 1)
14:08:10:0289 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
14:08:10:0289 FuCommon             reading /etc/fwupd/daemon.conf with 2080 bytes
14:08:10:0289 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
14:08:10:0290 FuConfig             using autodetected max archive size 18.4 EB
[New LWP 4265]
14:08:10:0292 FuRemoteList         loading /usr/share/fwupd/metainfo
14:08:10:0293 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
14:08:10:0293 XbSilo               file: d349ccf2-2103-f32b-87ac-76b689d78f37, current:1437c7d0-f45c-2fd2-46e2-2c1b87168623, cached: (null)
14:08:10:0307 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
14:08:10:0308 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
14:08:10:0308 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
14:08:10:0309 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
14:08:10:0309 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
14:08:10:0310 FuRemoteList         ignoring unfound remote fwupd
14:08:10:0310 FuRemoteList         ordering lvfs-testing=lvfs+1
14:08:10:0310 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:08:10:0310 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
14:08:10:0311 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
14:08:10:0311 Jcat                 loaded 1 certificates
14:08:10:0312 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
14:08:10:0312 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
14:08:10:0312 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
14:08:10:0313 Jcat                 loaded 1 certificates
14:08:10:0313 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
14:08:10:0313 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
14:08:10:0313 FuEngine             client certificate now exists: nothing to do
14:08:10:0313 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
14:08:10:0315 FuHistory            got schema version of 8
warning: the debug information found in "/usr/lib/debug//usr/lib/fwupd-plugins-7/libfu_plugin_dd.so.debug" does not match "/usr/lib/fwupd-plugins-7/libfu_plugin_dd.so" (CRC mismatch).

14:08:10:0331 FuIdle               setting timeout to 7200s
14:08:10:0419 FuCommon             reading /proc/self/exe with 330400 bytes
[... 71 lines cut ...]
14:08:10:0500 FuEngine             failed to add device /sys/devices/platform/soc@0/1d84000.ufshc/host0/target0:0:0/0:0:0:0/block/sda: no plugin ata found
14:08:10:0500 FuEngine             failed to add device /sys/devices/platform/soc@0/1d84000.ufshc/host0/target0:0:0/0:0:0:0/block/sda: no plugin scsi found
14:08:10:0500 FuEngine             failed to add device /sys/devices/platform/soc@0/1d84000.ufshc/host0/target0:0:0/0:0:0:0/block/sda: no plugin uf2 found
14:08:10:0500 FuEngine             failed to add device /sys/devices/platform/soc@0/1d84000.ufshc/host0/target0:0:0/0:0:0:0/block/sda: no plugin emmc found
14:08:10:0500 FuPluginDd           ----------------- fu_dd_device_class_init -----------
14:08:10:0500 FuPluginDd           ----------------- fu_dd_device_init -----------
14:08:10:0500 FuPluginDd           Found A/B partitioning scheme, active bootslot: '_a'
14:08:10:0500 FuPluginDd           ----------------- fu_dd_device_probe -----------
14:08:10:0502 FuPluginDd           Parent class probe done
14:08:10:0502 FuPluginDd           probe OK

Thread 1 "fwupd" received signal SIGSEGV, Segmentation fault.
0x0000aaaaaaac0364 in fu_engine_backend_device_added_run_plugins (progress=0xfffff6513490, device=0xfffff66d22e0, self=0xfffff67fe840) at ../src/fu-engine.c:6752
6752	../src/fu-engine.c: No such file or directory.
(gdb) bt
#0  0x0000aaaaaaac0364 in fu_engine_backend_device_added_run_plugins (progress=0xfffff6513490, device=0xfffff66d22e0, self=0xfffff67fe840) at ../src/fu-engine.c:6752
#1  fu_engine_backend_device_added (self=self@entry=0xfffff67fe840, device=device@entry=0xfffff66d22e0, progress=0xfffff650b710) at ../src/fu-engine.c:6806
#2  0x0000aaaaaaacc3dc in fu_engine_backends_coldplug_backend_add_devices (error=0xfffffffffa68, progress=0xfffff64ff180, backend=0xfffff6e7c790, self=0xfffff67fe840) at ../src/fu-engine.c:7166
#3  fu_engine_backends_coldplug_backend (error=0xfffffffffa68, progress=0xfffff67f4f90, backend=0xfffff6e7c790, self=0xfffff67fe840) at ../src/fu-engine.c:7193
#4  fu_engine_backends_coldplug (progress=0xfffff67fae10, self=0xfffff67fe840) at ../src/fu-engine.c:7228
#5  fu_engine_load (self=self@entry=0xfffff67fe840, flags=<optimized out>, flags@entry=14, progress=0xfffff6800910, error=error@entry=0xfffffffffb50) at ../src/fu-engine.c:7547
#6  0x0000aaaaaaabaa0c in fu_daemon_setup (self=self@entry=0xfffff6fc6410, socket_address=socket_address@entry=0x0, error=error@entry=0xfffffffffb50) at ../src/fu-daemon.c:2189
#7  0x0000aaaaaaab67b0 in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-main.c:172

```

hughsie, 2022-08-30:
Are you returning FALSE without setting the GError? Also try running until valgrind which will tell you straight away if you're doing something like reading from free'd memory.

DylanVanAssche, 2022-08-30:
Oh that was it! I didn't set the GError... I didn't know I had to do that :P I must have missed it in the other plugins!

closed 2022-08-30 by DylanVanAssche