fwupd/fwupd #4979: lenovo-thinklmi plugin doesn't work anymore

superm1, 2022-08-30:
I noticed that the `lenovo-thinklmi` plugin stopped working with 1.8.4.  It's because `firmware-attributes` doesn't load until much later in the daemon startup.  So there are a few possible solutions:

1. Subsume all the content from lenovo-thinklmi directly into uefi-capsule.  I'm not convinced this would work either though, uefi-capsule loads too soon.
2. Manually probe firmware-attributes at daemon startup rather than waiting for udev.  Not sure how we would avoid the re-probing when the udev event comes through though.  We want to have this used because if the kernel loads or unloads the underlying module we want the daemon to react.
3. Move functionality of thinklmi directly into the libfwupdplugin.

hughsie, 2022-08-30:
>  It's because firmware-attributes doesn't load until much later in the daemon startup

Can you elaborate a bit? What's needed at what stage?

hughsie, 2022-08-30:
Maybe we want to do g_udev_client_new() manually rather than fu_context_add_udev_subsystem() for the `firmware-attributes` subsystem?

superm1, 2022-08-30:
> Can you elaborate a bit? What's needed at what stage?

We don't setup the udev-subsystem for firmware-attributes until `fu_context_load_hwinfo`, and then we don't get the notification for it until `fu_engine_backend_device_added`, which comes after various other devices and plugins already ran.

I think to keep plugin for support for BIOS settings we would probably need to probe separately much earlier on before this.

superm1, 2022-08-30:
> 
Maybe we want to do g_udev_client_new() manually rather than fu_context_add_udev_subsystem() for the firmware-attributes subsystem?

I'm hoping we don't lose the ability to react to the kernel's uevents though, that's the part that is tough to decide when to do it.

commit 3a1c65e2 (2022-08-31): bios-settings: Fix plugin functionality

By moving the attribute loading into udev it was loading too late
and plugins couldn't use it.  Move the loading to earlier in the
daemon and ignore add events if we already have attributes loaded.

Fixes: #4979
commit be307ae8 (2022-08-31): bios-settings: Fix plugin functionality

By moving the attribute loading into udev it was loading too late
and plugins couldn't use it.  Move the loading to earlier in the
daemon and ignore add events if we already have attributes loaded.

Fixes: #4979
commit ad461d50 (2022-08-31): bios-settings: Fix plugin functionality

By moving the attribute loading into udev it was loading too late
and plugins couldn't use it.  Move the loading to earlier in the
daemon and ignore add events if we already have attributes loaded.

Fixes: #4979
closed 2022-08-31 by superm1