fwupd/fwupd #4982: Linker failure for `fu_common_string_append_kv`

DylanVanAssche, 2022-08-31:
**Describe the question**
I'm trying to use `fu_common_string_append_kv` in a `to_string` method for a new plugin, but the linker fails to find it. Other plugins use it and I have the same stuff in my `meson.build` so I'm not sure why the linker can't find it? It is part of `fu-common.h` which is in `libfwupdplugin.h`.

```
ninja: job failed: gcc  -o plugins/dd/libfu_plugin_dd.so plugins/dd/libfu_plugin_dd.so.p/fu-plugin-dd.c.o plugins/dd/libfu_plugin_dd.so.p/fu-dd-device.c.o -Wl,--as-needed -Wl,--allow-shlib-undefined -shared -fPIC -Wl,-z,relro -Wl,-z,defs -Wl,-z,now -Wl,--as-needed,-O1,--sort-common -Os -fomit-frame-pointer -g -Os -fomit-frame-pointer '-Wl,-rpath,$ORIGIN/../../libfwupd:$ORIGIN/../../libfwupdplugin' -Wl,-rpath-link,/home/pmos/build/src/fwupd-1.8.3/output/libfwupd -Wl,-rpath-link,/home/pmos/build/src/fwupd-1.8.3/output/libfwupdplugin -Wl,--start-group libfwupd/libfwupd.so.2.0.0 libfwupdplugin/libfwupdplugin.so.7.0.0 /usr/lib/libgio-2.0.so /usr/lib/libglib-2.0.so /usr/lib/libintl.so /usr/lib/libgobject-2.0.so /usr/lib/libgnutls.so -Wl,--export-dynamic /usr/lib/libgmodule-2.0.so -pthread /usr/lib/libgudev-1.0.so /usr/lib/libgusb.so /usr/lib/libusb-1.0.so /usr/lib/libarchive.so /usr/lib/libjson-glib-1.0.so /usr/lib/libxmlb.so /usr/lib/libprotobuf-c.so -Wl,--end-group
/usr/lib/gcc/aarch64-alpine-linux-musl/12.1.1/../../../../aarch64-alpine-linux-musl/bin/ld: plugins/dd/libfu_plugin_dd.so.p/fu-dd-device.c.o: in function `fu_dd_device_to_string':
/home/pmos/build/src/fwupd-1.8.3/output/../plugins/dd/fu-dd-device.c:42: undefined reference to `fu_common_string_append_kv'
/usr/lib/gcc/aarch64-alpine-linux-musl/12.1.1/../../../../aarch64-alpine-linux-musl/bin/ld: /home/pmos/build/src/fwupd-1.8.3/output/../plugins/dd/fu-dd-device.c:45: undefined reference to `fu_common_string_append_kv'
/usr/lib/gcc/aarch64-alpine-linux-musl/12.1.1/../../../../aarch64-alpine-linux-musl/bin/ld: /home/pmos/build/src/fwupd-1.8.3/output/../plugins/dd/fu-dd-device.c:48: undefined reference to `fu_common_string_append_kv'
collect2: error: ld returned 1 exit status
Fontconfig warning: ignoring UTF-8: not a valid region tag
ninja: subcommand failed
```

My patches: https://github.com/DylanVanAssche/fwupd/tree/dd-plugin

**fwupd version information**
Please provide the version of the daemon and client if applicable to your current installation of `fwupd`.

1.8.3

hughsie, 2022-08-31:
That's an old symbol name, are you working with git main? If so `contrib/migrate.py` might help.

DylanVanAssche, 2022-08-31:
Ah that was it! It was outdated :/ Thanks for the quick review already :)

closed 2022-08-31 by DylanVanAssche