fwupd/fwupd #4999: CSME Invalid

ganchair, 2022-09-04:
**Describe the bug**
fwupdmgr security output shows CSME Invalid it used to be Valid but after update (I assume) it shows invalid now.

[image]
[image]

**Steps to Reproduce**
1) fwupdmgr security
2) See CSME as Invalid

**Expected behavior**
Expect it to be valid as it used to be since months. 

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.6-300.fc37.x86_64
compile   com.hughsie.libjcat           0.1.10
compile   org.freedesktop.fwupd         1.8.4
runtime   org.freedesktop.gusb          0.3.10

```

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
LENOVO 20VJS0HB00
│
├─Unknown Device:
│     Device ID:          63223cc2d1b0b8d4c45ef8cedcb1f45315d3ff20
│     GUID:               ca5c154c-a46a-59d8-bf77-0ca40522cc85 ← GPIO\ID_INT34C5:00
│   
├─11th Gen Intel Core™ i7-1165G7 @ 2.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000a4
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f ← cpu
│                         30249f37-d140-5d3e-9319-186b1bd5cac3 ← CPUID\PRO_0&FAM_06
│                         2347ad97-b545-5bae-b6b3-b70a9553cbba ← CPUID\PRO_0&FAM_06&MOD_8C
│                         1df99df5-22f4-5505-98cc-22dcc53b132a ← CPUID\PRO_0&FAM_06&MOD_8C&STP_1
│     Device Flags:       • Internal device
│   
├─Intel Management Engine:
│     Device ID:          a45df35ac0e948ee180fe216a5f703f32dda163f
│     Summary:            UEFI ESRT device
│     Current version:    0.35.1951
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               19aef08f-2701-4db3-b90b-d4bc1412ade7
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─KXG6AZNV512G TOSHIBA:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
│     Current version:    5108AGLA
│     Vendor:             Toshiba Corporation (NVME:0x1179)
│     Serial Number:      31DF33TRFDM3
│     GUIDs:              94eae9cd-5ee6-5b80-9a3e-ffa2db88cd87 ← NVME\VEN_1179&DEV_011A
│                         de51500a-8b48-5f13-b0c1-0bbded7c54c1 ← NVME\VEN_1179&DEV_011A&REV_00
│                         141102bc-3608-5bc8-a700-983f47d476f4 ← NVME\VEN_1179&DEV_011A&SUBSYS_11790001
│                         9da14130-275e-52fc-8119-63be32bd58dd ← NVME\VEN_1179&DEV_011A&SUBSYS_11790001&REV_00
[... 197 lines cut ...]
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─USB4 host controller:
      Device ID:          b4e925a2487995d54faa9fafcdd767cf954bc03e
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
    

```

**Additional questions**

- Operating system and version:
Fedora Silverblue 37

- Have you tried rebooting? 
Yes (Also tried to reset UEFI settings to default, enable/disable secure boot etc.)

- Is this a regression? 
I assume yes since I didn't had any changes on firmware or BIOS update since months and I am sure everything was OK since months but recently (probably after last update things changed for example I see Intel CET as enabled now)

ganchair, 2022-09-04:
I downgraded to 1.8.3 (https://koji.fedoraproject.org/koji/buildinfo?buildID=2019497) and here is the result of fwupdmgr security. 
[image]

So it seems like a regression.

hughsie, 2022-09-04:
Totally expected I'm afraid, Intel released a new dataset with the new vulnerable versions, and so we updated to that: https://github.com/fwupd/fwupd/commit/4e5a928d919a1eb91f21e0aef1c07a1adf836ada

ganchair, 2022-09-04:
thanks for the information. I assume I have to wait for a new Intel ME update to be released from my vendor right?

hughsie, 2022-09-04:
> I assume I have to wait for a new Intel ME update to be released from my vendor right?

I'm afraid so.

ganchair, 2022-09-04:
thanks again. closing the issue since its not a bug.

closed 2022-09-04 by ganchair