fwupd/fwupd #5035: Cannot update on Fedora 36 wirh Secure Boot dbx

thorrrr, 2022-09-16:
Cannot update today on Fedora 36 with Secure Boot dbx
❯ sudo dsync --refresh
[sudo] password for dale:
sudo: dsync: command not found
❯ sudo dsync --refresh
sudo: dsync: command not found
❯
❯ sudo dnf dsync --refresh
Copr repo for fedy owned by kwizart                                                           7.0 kB/s | 3.0 kB     00:00
Copr repo for PyCharm owned by phracek                                                         11 kB/s | 3.6 kB     00:00
Fedora 36 - x86_64                                                                            8.5 kB/s |  21 kB     00:02
Fedora 36 openh264 (From Cisco) - x86_64                                                      2.8 kB/s | 989  B     00:00
Fedora Modular 36 - x86_64                                                                     29 kB/s |  20 kB     00:00
Fedora 36 - x86_64 - Updates                                                                   23 kB/s |  12 kB     00:00
Fedora Modular 36 - x86_64 - Updates                                                           32 kB/s |  13 kB     00:00
google-chrome                                                                                 9.6 kB/s | 1.3 kB     00:00
microsoft-edge                                                                                 17 kB/s | 3.0 kB     00:00
created by dnf config-manager from https://packages.microsoft.com/yumrepos/edge                19 kB/s | 3.0 kB     00:00
RPM Fusion for Fedora 36 - Free                                                                34 kB/s | 7.7 kB     00:00
RPM Fusion for Fedora 36 - Free - Updates                                                      41 kB/s | 7.0 kB     00:00
RPM Fusion for Fedora 36 - Nonfree                                                             23 kB/s | 7.8 kB     00:00
RPM Fusion for Fedora 36 - Nonfree - NVIDIA Driver                                             44 kB/s | 7.3 kB     00:00
RPM Fusion for Fedora 36 - Nonfree - Steam                                                     20 kB/s | 7.0 kB     00:00
RPM Fusion for Fedora 36 - Nonfree - Updates                                                   40 kB/s | 7.1 kB     00:00
Dependencies resolved.
Nothing to do.
Complete!
❯
❯ sudo fwupdmgr update
Devices with no available firmware updates:
 • THNSN5512GPUK NVMe TOSHIBA 512GB
Devices with the latest available firmware version:
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 83 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]:
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Decompressing…           [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
❯ sudo fwupdtool esp-list --verbose:
Failed to parse arguments: Unknown option --verbose:

hughsie, 2022-09-16:
> sudo fwupdtool esp-list --verbose:
> Failed to parse arguments: Unknown option --verbose:

That's the one we need -- but without the colon at the end.

thorrrr, 2022-09-16:
> 

I realised it and added it in main post

hughsie, 2022-09-16:
The issue is that fu_uefi_dbx_signature_list_validate() is only checking for FU_VOLUME_KIND_ESP and needs to fall back to FU_VOLUME_KIND_BDP too. I can fix on Monday if nobody beats me to it.

thorrrr, 2022-09-16:
You might as well have written a Czechoslovakian poem backwards  :) 
But i understood you will fix it Monday  have a great weekend

sbadux, 2022-09-17:
Same error on KDE Neon based on Ubuntu LTS 20.04. I hope it will be available for us too (I know that it doesn't depends on you)

BiNh0X, 2022-09-17:
Same error on a Dell laptop with Ubuntu 20.04 LTS. My version of fwupd is 1.7.5, will the fix come through get-update or in the latest version of fwupd? Thank you very much for your work.

hughsie, 2022-09-17:
> will the fix come through get-update or in the latest version of fwupd

You need to use the newest fwupd version unfortunately.

closed 2022-09-17 by hughsie

thorrrr, 2022-09-17:
How can i get that ?
Via terminal or will it come through to Fedora ?

hughsie, 2022-09-17:
It'll be in Fedora about 48 hours after it's been released upstream (usually the end of the month, every month) but until they you can use https://copr.fedorainfracloud.org/coprs/rhughes/fwupd/

pjwelsh, 2022-09-19:
What's the chance that this fix will come down to CentOS 8-stream? It's at fwupd-1.7.4-2.el8.x86_64 and has this issue. I am going to attempt to rebuild your coprs src.rpm.

Is there a concern that FC35 has a failed build state? "Status: failed - Build failed. See logs for more details."
I generally need to use the older of the FC versions to rebuild for my C8 systems.

w5pny, 2022-09-19:
This problem exists on Ubuntu 20.04 -- what's the chance it will get fixed???  Current version here is 1.7.5-3~20.04.1

pjwelsh, 2022-09-19:
Attempt to rebuild the current copr src.rpm is not working out... dependency hell so far. Getting way into the weeds replacing el8 system packages with either newer FC3[5,6] or 9-Stream package version and still running into more build issues.

BiNh0X, 2022-09-19:
@w5pny 

> > will the fix come through get-update or in the latest version of fwupd
> 
> You need to use the newest fwupd version unfortunately.

As soon as version 1.8.5 comes out in the snap-store, install it temporarily and do the UEFI dbx update.

maymaypp, 2022-09-20:
I think my laptop has a similar problem. The laptop runs Ubuntu 20.04 and the versions are 

```
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.14.0-1051-oem
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.4
```
I didn't get the `Blocked executable in the ESP, ensure grub and shim are up to date: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b` error. Instead, I get the following 

```
UEFI dbx:
  Device ID:            362301da643102b9f38477387e2193e57abaa590
  Previous version:     77
  Update State:         Failed
  Update Error:         failed to run update on reboot
  Last modified:        2022-09-19 08:20
  GUID:                 00fe3755-a4d8-5ef7-ba5f-47979fbb3423
  Device Flags:         • Internal device
                        • Updatable
                        • Needs a reboot after installation
                        • Reported to remote server
```
I updated dbx related firmware via Ubuntu Software Center about two days ago.

dankhrapiyush, 2022-09-20:
I'm getting the issue while trying to update the UEFI dbx on Ubuntu 22.04.
 
`error in fwupd client: calling Install method failed: Blocked executable in the ESP, ensure grub and shim are up to date: /media/root/9E33-C958/EFI/Boot/bootx64.efi Authenticode checksum [2ea4cb6a1f1eb1d3dce82d54fde26ded243ba3e18de7c6d211902a594fe56788] is present in dbx`

hughsie, 2022-09-20:
> What's the chance that this fix will come down to CentOS 8-stream

I guess fairly likely; I'm doing a less invasive fix for the 1_7_X branch now and when 1.7.<next> is released I'll probably rebase both c9s and c8s packages. No idea on actual RHEL tho, that's not my decision.

> This problem exists on Ubuntu 20.04 -- what's the chance it will get fixed???

You need to ask Canonical in Launchpad.

pjwelsh, 2022-09-20:
@hughsie Thank you very much for the update on the 1_7_X branch. Where would your c8s package be release at? Not sure if that indicates an official (CentOS) repository or not.
PJ

hughsie, 2022-09-20:
> Where would your c8s package be release at

I'd need an upstream release first, but I'm planning both branches to release at the end of the week. I'm also a Red Hat employee, so I'd be the official stream repo -- although I need to get approval for that first.

pjwelsh, 2022-09-20:
Good thing this is a security related issue then; It should hopefully move along more quickly.
So many systems not update-able at this time :(

hughsie, 2022-09-20:
@pjwelsh if you could file a fwupd c9s and c8s bugzilla entry that would speed things up a bit.

pjwelsh, 2022-09-20:
Certainly.
Should this be considered a "security sensitive" issue based on the bugzilla description? I tend to think yes based on this.

hughsie, 2022-09-20:
I don't think so; this isn't the tool with the security issue, it's the tool to apply the mitigation. Up to you tho.

pjwelsh, 2022-09-20:
"Bug 2128349 has been successfully created" (https://bugzilla.redhat.com/show_bug.cgi?id=2128349)
Thank you!

pjwelsh, 2022-09-20:
(C9) "Bug 2128384 has been successfully created" (https://bugzilla.redhat.com/show_bug.cgi?id=2128384)
Forgot you wanted both 'til now.

fkru, 2022-09-20:
> I think my laptop has a similar problem. The laptop runs Ubuntu 20.04 and the versions are
>
> ```
> runtime   org.freedesktop.fwupd         1.7.5
> runtime   com.dell.libsmbios            2.4
> compile [...]

Seems to be a different issue, I am experiencing the same, even with secure boot disabled. Update of (other) firmware works fine. @hugsie: Any idea? Thx.

SunJun8, 2022-09-21:
```
Devices with no available firmware updates:
 • MSFT0002:00 04F3:3140
 • System Firmware
 • WDC PC SN530 SDBPMPZ-512G-1101
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 83 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Failed to download, server response was 404
```
😂

hughsie, 2022-09-21:
> Failed to download, server response was 404

Weird, a LVFS bug. Try `fwupdmgr refresh --force` and try again please.

hughsie, 2022-09-21:
> Weird, a LVFS bug

Fixed in https://gitlab.com/fwupd/lvfs-website/-/merge_requests/1363 for future firmwares.

SunJun8, 2022-09-21:
> > Failed to download, server response was 404
> 
> Weird, a LVFS bug. Try `fwupdmgr refresh --force` and try again please.

okay

thorrrr, 2022-09-21:
[image]

BiNh0X, 2022-09-22:
I solved the problem on Ubuntu 20.04 LTS by installing the flatpak Firmware package: [https://flathub.org/apps/details/org.gnome.Firmware](url)
In UEFI dbx, list available updates but if you choose latest version 217, the executable blocked error in ESP continues. The tip is to choose the first version above 77 and have it updated, which will not give an error and once installed, it will ask to reboot the system. After that, it was updated from version 77 to version 238 and no longer has updates.
[image]
Note: fwupd 1.8.5 snap version is giving error to install, shows error 'GLIBC_2.32 e 2.34' not found, and Ubuntu 20.04 uses version 2.31.

jwakely, 2022-09-22:
> I'm getting the issue while trying to update the UEFI dbx on Ubuntu 22.04.
>
> `error in fwupd client: calling Install method failed: Blocked executable in the ESP, ensure grub and shim are up to date: [...]

Is this actually a different issue? It's a different error message at least. Will that be fixed by the new fwupd?

I'm getting this error (with a different checksum). I think it's because I have an old boot partition on one of my disks, which I no longer use. fwupdmgr seems to be finding that partition and complaining that it has an old BOOTX64.EFI present. Note that the error is not about `/boot/efi/...` but some other unmounted disk.

I fixed the problem by using `gparted` to remove the boot flag from the `/dev/sdc1` partition (which is the one where the old blocked BOOTX64.EFI was found).

superm1, 2022-09-22:
> Is this actually a different issue? It's a different error message at least. Will that be fixed by the new fwupd?

This won't be changed by a new fwupd release.  Fwupd has no idea about user intent.  If you have a blocked executable on your bootable partitions, it doesn't want to be the reason that you can't boot something anymore.  So you should either remove the partition, remove the executable or update the executable.

dkelson, 2022-09-26:
> [image]

What's your `PROMPT_COMMAND` / `PS1` etc?

mskinnerRH, 2022-09-26:
i had an old UEFI boot partition that was missing the "esp" flag. make sure to check your boot partition it should have both boot,esp flags.

Steps to fix or validate:

type "parted /dev/sda"
type "p"
look for first partition, it should have boot and esp flags. if not.
type "t"
type "1"
type "esp"
type "p" to validate
type "q" to quit

laur89, 2022-09-27:
Running debian testing, fwupd `1.8.5-1`:
```╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures. If the installation fails, you will    ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly. You may have to       ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Decompressing…           [                                       ]
Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/EFI/BOOT/BOOTX64.EFI Authenticode checksum [af79b14064601bc0987d4747af1e914a228c05d622ceda03b7a4f67014fee767] is present in dbx
```

System is using encrypted `/dev/nvme0n1p4` partition; `/boot` partition has no flags, but `/boot/efi` has both "boot" & "esp" flags:
[image]

superm1, 2022-09-27:
> Running debian testing, fwupd `1.8.5-1`, grub `2.06-4`:
> ```╔══════════════════════════════════════════════════════════════════════════════╗
> ║ Upgrade UEFI dbx from 77 to 217?                                             ║
> ╠══════════════════════════════════════════════════════════════════════════════╣
> ║ This updates the dbx to the latest release from Microsoft which adds         ║
> ║ insecure versions of grub and shim to the list of forbidden signatures due   ║
> ║ to multiple discovered security updates.                                     ║
> ║                                                                              ║
> ║ Before installing the update, fwupd will check for any affected executables  ║
> ║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
> ║ with any of the forbidden signatures. If the installation fails, you will    ║
> ║ need to update shim and grub packages before the update can be deployed.     ║
> ║                                                                              ║
> ║ Once you have installed this dbx update, any DVD or USB installer images     ║
> ║ signed with the old signatures may not work correctly. You may have to       ║
> ║ temporarily turn off secure boot when using recovery or installation media,  ║
> ║ if new images have not been made available by your distribution.             ║
> ║                                                                              ║
> ╚══════════════════════════════════════════════════════════════════════════════╝
> 
> Perform operation? [Y|n]: 
> Downloading…             [***************************************]
> Downloading…             [***************************************]
> Decompressing…           [***************************************]
> Authenticating…          [***************************************]
> Decompressing…           [                                       ]
> Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/EFI/BOOT/BOOTX64.EFI Authenticode checksum [af79b14064601bc0987d4747af1e914a228c05d622ceda03b7a4f67014fee767] is present in dbx
> ```
> 
> `/boot` partition has no flags, but `/boot/efi` has both "boot" & "esp" flags set:
> [image]
> 

Whatever provided your bootx64.efi needs to update it.

pjwelsh, 2022-09-28:
@hughsie  "I'd need an upstream release first, but I'm planning both branches to release at the end of the week. I'm also a Red Hat employee, so I'd be the official stream repo -- although I need to get approval for that first."

Is the updated release still on track for the end of this week?

hughsie, 2022-09-28:
I've done 1.7.10 and 1.8.5 -- I just need to do the stream bits. WIP.

nsslh, 2022-10-03:
My problem was that I had a very old /boot/efi/EFI/ubuntu from a previous install. I am running Fedora. So the very old ubuntu components were not being updated, blocking the UEFI dbx update.

So folks who continue to experience this issue after upgrading fwupd should be advised to look out for that.

HTH HAND YMMV IANAL

gmanic, 2022-10-03:
> My problem was that I had a very old /boot/efi/EFI/ubuntu from a previous install. I am running Fedora. So the very old ubuntu components were not being updated, blocking the UEFI dbx update.
>
> So [...]

What's your fix for that situation?

nsslh, 2022-10-03:
> What's your fix for that situation?

Very, very carefully delete the old, unmaintained components. In my case, since I know I am not using Ubuntu, it was safe for me to `rm -r /boot/efi/EFI/ubuntu`.

superm1, 2022-10-03:
Obviously this is going to keep coming up.  I'm personally happy that we have the safety check in place.  If nothing else it prevents fwupd from being the cause of non-obvious breakage to users.

I've added a wiki page to document this further:
https://github.com/fwupd/fwupd/wiki/Blocked-executable-in-the-ESP,-ensure-grub-and-shim-are-up-to-date