fwupd/fwupd #5042: unable to install update with ESP on raid1

ausil, 2022-09-20:
```
[root@kasuf ~]# fwupdmgr update
Devices with no available firmware updates: 
 • BCM20702A0
 • SSD 850 EVO M.2 500GB
 • SSD 970 EVO Plus 1TB
 • SSD 970 EVO Plus 1TB
 • System Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Decompressing…           [                                       ]Blocked executable in the ESP, ensure grub and shim are up to date: no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
[root@kasuf ~]# lsblk
NAME                                            MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
sda                                               8:0    0 465.8G  0 disk  
└─sda1                                            8:1    0 465.8G  0 part  /run/media/dennis/252690c6-5dde-471a-abe2-2b59dd3ca718
zram0                                           252:0    0     8G  0 disk  [SWAP]
nvme1n1                                         259:0    0 931.5G  0 disk  
├─nvme1n1p1                                     259:5    0     1G  0 part  
│ └─md127                                         9:127  0  1023M  0 raid1 /boot
├─nvme1n1p2                                     259:6    0   601M  0 part  
│ └─md125                                         9:125  0 600.9M  0 raid1 /boot/efi
└─nvme1n1p3                                     259:7    0 929.7G  0 part  
  └─md126                                         9:126  0 929.6G  0 raid1 
    └─luks-a4d3f8ab-8545-40d9-be42-c960666ec138 253:0    0 929.6G  0 crypt 
      └─fedora_kasuf-root                       253:1    0   900G  0 lvm   /
nvme0n1                                         259:1    0 931.5G  0 disk  
├─nvme0n1p1                                     259:2    0     1G  0 part  
│ └─md127                                         9:127  0  1023M  0 raid1 /boot
├─nvme0n1p2                                     259:3    0   601M  0 part  
│ └─md125                                         9:125  0 600.9M  0 raid1 /boot/efi
└─nvme0n1p3                                     259:4    0 929.7G  0 part  
  └─md126                                         9:126  0 929.6G  0 raid1 
    └─luks-a4d3f8ab-8545-40d9-be42-c960666ec138 253:0    0 929.6G  0 crypt 
      └─fedora_kasuf-root                       253:1    0   900G  0 lvm   /

```

I have my esp partition as a raid1, and I have set in /etc/fwupd/uefi_capsule.conf OverrideESPMountPoint=/boot/efi this allows updates to be checked but apparently not applied

ausil, 2022-09-20:
```
fwupdtool esp-list --verbose
02:57:02:0107 FuDebug              Verbose debugging enabled (on console 1)
02:57:02:0119 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0120 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0121 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0123 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0125 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0126 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: xfs
02:57:02:0128 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0129 FuVolume             no volumes of type c12a7328-f81f-11d2-ba4b-00a0c93ec93b
02:57:02:0141 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0143 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0145 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0147 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0148 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0149 FuVolume             device /org/freedesktop/UDisks2/block_devices/sda1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 0, fs: xfs
02:57:02:0151 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: a19d880f-05fc-4d3b-a006-743f0f84911e, internal: 1, fs: linux_raid_member
02:57:02:0152 FuVolume             no volumes of type ebd0a0a2-b9e5-4433-87c0-68b6b72699c7
No ESP or BDP found

```

hughsie, 2022-09-20:
I didn't know you could have a ESP on RAID tbh. @vathpela do you think firmware updates are going to work in reality? I suppose the dbx checker should respect `OverrideESPMountPoint=/boot/efi` regardless....

hughsie, 2022-09-20:
@superm1 I guess there are a few ways we could solve this:

 * Add a `fu_plugin_get_config_value_full()` that allows us to specify the plugin name, e.g. `fu_plugin_get_config_value_full(plugin, "uefi-capsule", "OverrideESPMountPoint")` which is a bit eww from a layers point of view, although preserves forward and backwards compat
 * Add a `OverrideESPMountPoint` option to daemon.conf and migrate the value in `uefi_capsule.conf` at startup and then have something like `fu_context_get_esp_volume**s**()`
 * Read the `OverrideESPMountPoint` in `uefi_capsule` and set the environment value `FWUPD_ESP_DIRECTORY` so we can use something like `fu_path_from_kind(FU_PATH_KIND_CUSTOM_ESP_MOUNTPOINT)` -- this is a bit eww as setting env variables at runtime leads to all kinds of funky as getenv returns a pointer, not a dup'd string.

I'm erring on the 2nd option, as this would also mean we could remove `fu_volume_new_by_esp()` and deprecate `fu_volume_new_esp_default()` and `fu_volume_new_esp_for_path()`. Comments welcome!

superm1, 2022-09-20:
#2 is my gut too, thanks!

vathpela, 2022-09-20:
> I didn't know you could have a ESP on RAID tbh. @vathpela do you think firmware updates are going to work in reality? I suppose the dbx checker should respect `OverrideESPMountPoint=/boot/efi` regardless....

In general I think it should work fine; while ESP on RAID 1 is a gigantic hack, it basically works so long as the bootloader doesn't do any writing to the drive. So assuming it's not degraded, updates just need to boot one or the other.

vathpela, 2022-09-20:
> A lot of BIOS vendors have an assumption that they can write to the ESP. I've seen recovery BIOS image end up there and log dumps end up there. If the ESP is actually supposed to be RAIDed between two disks this sounds like a recipe for disaster to me. Wouldn't the BIOS essentially degrade the array?

Yeah - for sure if you need this functionality the right thing to do is BIOS RAID, but IBVs continue to make that a value add that isn't in most builds, so people do OS-level RAID instead.

That said, the failure mode for e.g. writing logs out to the ESP is that you can't find the logs, but redundant booting still works.

closed 2022-09-22 by hughsie