fwupd/fwupd #5069: 1.8.5-1-g713bf0467 snap fails to refresh due to "no libcurl support"

estan, 2022-09-22:
**Describe the bug**
1.8.5-1-g713bf0467 was published in the latest/stable channel today, when running fwupdmgr refresh it gives the error "no libcurl support".

Possibly same issue as in https://bugs.gentoo.org/841836#c3 ?

**Steps to Reproduce**
Install 1.8.5-1-g713bf0467 from snap and run fwupdmgr refresh.

**Expected behavior**
Refresh should happen

**fwupd version information**

```shell
estan@edison:~$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-48-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.3.10
estan@edison:~$
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): snap

superm1, 2022-09-22:
Oh, I think I know what's going on here, when we re-enabled the snap and cleaned it up for core22, this was lost.

closed 2022-09-22 by superm1

estan, 2022-09-22:
Can confirm 1.8.5-10-g34dfa8689 is working :+1: Thanks!

commit 34dfa868 (2022-09-22): trivial: restore libcurl support in snap (Fixes: #5069)