fwupd/fwupd #5073: 1.8.5: test suite is failing in `fwupd-self-test` unit

kloczek, 2022-09-23:
**Describe the bug**
test suite is failing in `fwupd-self-test` unit.

**Steps to Reproduce**
Meson settiongs:
```spec
%meson \
        -D bluez=enabled \
        -D build=all \
        -D consolekit=enabled \
        -D curl=enabled \
        -D docs=enabled \
        -D firmware-packager=true \
        -D gudev=enabled \
        -D gusb=enabled \
        -D introspection=enabled \
        -D libarchive=enabled \
        -D lvfs=true \
        -D lzma=enabled \
        -D man=true \
        -D plugin_acpi_phat=enabled \
        -D plugin_amt=enabled \
        -D plugin_bcm57xx=enabled \
        -D plugin_cfu=enabled \
        -D plugin_cpu=enabled \
        -D plugin_dell=enabled \
        -D plugin_dummy=enabled \
        -D plugin_dummy=true \
        -D plugin_emmc=enabled \
        -D plugin_ep963x=enabled \
        -D plugin_fastboot=enabled \
        -D plugin_flashrom=enabled \
        -D plugin_gpio=enabled \
        -D plugin_intel_spi=enabled \
        -D plugin_intel_spi=true \
        -D plugin_logitech_bulkcontroller=enabled \
        -D plugin_modem_manager=enabled \
        -D plugin_msr=enabled \
        -D plugin_mtd=enabled \
        -D plugin_nitrokey=enabled \
        -D plugin_nvme=enabled \
        -D plugin_parade_lspcon=enabled \
        -D plugin_pixart_rf=enabled \
        -D plugin_platform_integrity=enabled \
        -D plugin_platform_integrity=true \
        -D plugin_powerd=enabled \
        -D plugin_realtek_mst=enabled \
        -D plugin_redfish=enabled \
        -D plugin_scsi=enabled \
        -D plugin_synaptics_mst=enabled \
        -D plugin_synaptics_rmi=enabled \
        -D plugin_tpm=enabled \
        -D plugin_uefi_capsule=enabled \
        -D plugin_uefi_capsule_splash=enabled \
        -D plugin_uefi_capsule_splash=true \
        -D plugin_uefi_pk=enabled \
        -D plugin_uf2=enabled \
        -D plugin_upower=enabled \
        -D polkit=enabled \
        -D soup_session_compat=true \
        -D supported_build=enabled \
        -D systemd=enabled \
        -D tests=true \
        %{nil}
```

**Expected behavior**
Test suite shoul not fail.

**fwupd version information**
[... 142 lines cut ...]
24/35 redfish-self-test                      OK               0.14s
25/35 synaptics-mst-self-test                OK               0.14s
26/35 synaptics-prometheus-self-test         OK               0.13s
27/35 synaptics-rmi-self-test                OK               0.12s
28/35 tpm-self-test                          OK               0.10s
29/35 uefi-self-test                         OK               0.09s
30/35 uefi-dbx-self-test                     OK               0.09s
31/35 uf2-self-test                          OK               0.08s
32/35 vli-self-test                          OK               0.07s
33/35 wacom-usb-self-test                    OK               0.06s
34/35 thunderbolt-self-test                  OK               3.42s
35/35 fu-self-test                           OK               6.39s

Summary of Failures:

 5/35 fwupd-self-test                FAIL             0.34s   killed by signal 6 SIGABRT

Ok:                 34
Expected Fail:      0
Fail:               1
Unexpected Pass:    0
Skipped:            0
Timeout:            0
```
</details>

superm1, 2022-09-23:
Any chance that you could add your environment to our CI system?  I feel like every single release there is something specific to your environment that is breaking.  It would be nice to be pro-active about it.

kloczek, 2022-09-23:
Organizing lxc zone is not that difficult.

Above is from my devel system (which is as well in lxc zone) however in that system I have a bit more installed than it is only in spec file BuildRequires, and I just trealised that it this looks like another gvfs related issue (look on line with `GLib-GIO-DEBUG: _g_io_module_get_default: Found default implementation gvfs (GDaemonVfs) for ‘gio-vfs’`).
Will try to repeat that on system without gvfs packages installed.

superm1, 2022-09-23:
BTW - I tried to run your exact build command in a debian stable LXC container (with an upgraded meson) and I can't reproduce it.  In any case I think if you can't connect due to your environment, we should just skip those tests.  So I've sent a PR to that effect.

closed 2022-09-23 by superm1

kloczek, 2022-09-23:
I have no enough knowledge about that are.
Q: do you knoiw why this unit in tuis case may be failing and/or what I can try to diagnose potentnial cause?

superm1, 2022-09-23:
Maybe dbus activation doesn't work in your environment?  I'm not sure.  But the commit should avoid it and skip the test that isn't going to work.

commit 534da23e (2022-09-23): trivial: skip self tests for fwupd_client_remotes_func if G_DBUS_ERROR_NAME_HAS_NO_OWNER
Fixes: #5073
PR #5074 (2022-09-23): trivial: skip self tests for fwupd_client_remotes_func if G_DBUS_ERRO…
…R_NAME_HAS_NO_OWNER

Fixes: #5073

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 7be0c033 (2022-09-23): trivial: skip self tests for fwupd_client_remotes_func if G_DBUS_ERROR_NAME_HAS_NO_OWNER
Fixes: #5073