fwupd/fwupd #5108: Build of ccgx plugin fails with fu-hash.h: No such file or directory

flanfly, 2022-09-30:
**Describe the bug**
Building main fails with `fu-hash.h: No such file or directory` while compiling the ccgx plugin.

**Steps to Reproduce**
On Ubuntu 20.04

```
git checkout main
meson build \
  -Dintrospection=disabled \
  -D{libxmlb,libjcat,gcab}:introspection=false \
  -Dgcab:docs=false -Dlibxmlb:gtkdoc=false \
  -Dplugin_redfish=enabled -Dcurl=enabled \
  -Dgusb=enabled --prefix $HOME/.local
ninja -C build -j1
```

A multithreaded build gets past it sometimes.

**Expected behavior**
The project builds w/o error

**fwupd version information**
56adf2473b

**fwupd device information**
N/A
**Additional questions**

Full error:

```
ninja: Entering directory `build'
[1/575] Compiling C object plugins/ccgx/libfu_plugin_ccgx.a.p/fu-ccgx-hid-device.c.o
FAILED: plugins/ccgx/libfu_plugin_ccgx.a.p/fu-ccgx-hid-device.c.o 
cc -Iplugins/ccgx/libfu_plugin_ccgx.a.p -Iplugins/ccgx -I../plugins/ccgx -I. -I.. -Ilibfwupd -I../libfwupd -Ilibfwupdplugin -I../libfwupdplugin -Isubprojects/libxmlb/src -I../subprojects/libxmlb/src -I/usr/include/libmount -I/usr/include/blkid -I/usr/include/glib-2.0 -I/usr/lib/x86_64-linux-gnu/glib-2.0/include -I/usr/include/gio-unix-2.0 -I/usr/include/p11-kit-1 -I/usr/include/gusb-1 -I/usr/include/libusb-1.0 -I/usr/include/json-glib-1.0 -fdiagnostics-color=always -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -std=c99 -O0 -g -Waggregate-return -Wunused -Warray-bounds -Wcast-align -Wclobbered -Wdeclaration-after-statement -Wdiscarded-qualifiers -Wduplicated-branches -Wduplicated-cond -Wempty-body -Wformat=2 -Wformat-nonliteral -Wformat-security -Wformat-signedness -Wignored-qualifiers -Wimplicit-function-declaration -Winit-self -Wlogical-op -Wmaybe-uninitialized -Wmissing-declarations -Wmissing-format-attribute -Wmissing-include-dirs -Wmissing-noreturn -Wmissing-parameter-type -Wmissing-prototypes -Wnested-externs -Wno-cast-function-type -Wno-address-of-packed-member -Wno-unknown-pragmas -Wno-missing-field-initializers -Wno-strict-aliasing -Wno-suggest-attribute=format -Wno-unused-parameter -Wold-style-definition -Woverride-init -Wpointer-arith -Wredundant-decls -Wreturn-type -Wshadow -Wsign-compare -Wstrict-aliasing -Wstrict-prototypes -Wswitch-default -Wtype-limits -Wundef -Wuninitialized -Wunused-but-set-variable -Wunused-variable -Wvla -Wwrite-strings -fstack-protector-strong -DFWUPD_COMPILATION -D_DEFAULT_SOURCE -DFWUPD_DISABLE_DEPRECATED -D_BSD_SOURCE -D__BSD_VISIBLE -D_XOPEN_SOURCE=700 -D_GNU_SOURCE -D_DARWIN_C_SOURCE=900000 -fPIC -pthread '-DG_LOG_DOMAIN="FuPluginCcgx"' -MD -MQ plugins/ccgx/libfu_plugin_ccgx.a.p/fu-ccgx-hid-device.c.o -MF plugins/ccgx/libfu_plugin_ccgx.a.p/fu-ccgx-hid-device.c.o.d -o plugins/ccgx/libfu_plugin_ccgx.a.p/fu-ccgx-hid-device.c.o -c ../plugins/ccgx/fu-ccgx-hid-device.c
In file included from ../libfwupdplugin/fu-usb-device.h:19,
                 from ../libfwupdplugin/fu-hid-device.h:9,
                 from ../libfwupdplugin/fwupdplugin.h:47,
                 from ../plugins/ccgx/fu-ccgx-hid-device.h:10,
                 from ../plugins/ccgx/fu-ccgx-hid-device.c:10:
../libfwupdplugin/fu-plugin.h:35:10: fatal error: fu-hash.h: No such file or directory
   35 | #include "fu-hash.h"
      |          ^~~~~~~~~~~
compilation terminated.
ninja: build stopped: subcommand failed.
```

[meson-log.txt](https://github.com/fwupd/fwupd/files/9682961/meson-log.txt)

hughsie, 2022-09-30:
Ya, mea cupla. I just pushed a fix to main -- can you try that pls.

flanfly, 2022-09-30:
Damn. 1 min turnaround time. 👍

closed 2022-09-30 by flanfly