fwupd/fwupd #5111: ThinkPad Universal USB-C Dock fails to update

bdaase, 2022-09-30:
**Describe the bug**
Today, the ThinkPad Universal USB-C Dock appeared for me. However ,fwupd fails to perform the update.

The firmware seems to be flashed onto the dock since its LED is blinking after starting the update and stopping again. However, the update does not seem to complete successfully and `fwupdmgr get-devices` shows 
```shell
Problems:           • Device is waiting for the update to be applied
```
for the dock (even after 6 hours now).

Interestingly, a second `fwupdmgr update` when is this state seems to break  `fwupdmgr update -v` midway.

```shell
➜  ~ fwupdmgr update -v
(fwupdmgr:34219): GLib-DEBUG: 19:01:39.845: setenv()/putenv() are not thread-safe and should not be used after threads are created
(pkttyagent:34221): GLib-GIO-DEBUG: 19:01:39.848: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:34219): GLib-GIO-DEBUG: 19:01:39.854: _g_io_module_get_default: Found default implementation dconf (DConfSettingsBackend) for ‘gsettings-backend’
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.854: watch_fast: "/system/proxy/" (establishing: 0, active: 0)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.855: watch_fast: "/system/proxy/http/" (establishing: 0, active: 0)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.855: watch_fast: "/system/proxy/https/" (establishing: 0, active: 0)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.855: watch_fast: "/system/proxy/ftp/" (establishing: 0, active: 0)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.855: watch_fast: "/system/proxy/socks/" (establishing: 0, active: 0)
(fwupdmgr:34219): GLib-GIO-DEBUG: 19:01:39.855: _g_io_module_get_default: Found default implementation gnome (GProxyResolverGnome) for ‘gio-proxy-resolver’
(fwupdmgr:34219): GLib-GIO-DEBUG: 19:01:39.855: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:34219): GLib-GIO-DEBUG: 19:01:39.855: Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.857: watch_established: "/system/proxy/" (establishing: 1)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.857: watch_established: "/system/proxy/http/" (establishing: 1)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.857: watch_established: "/system/proxy/https/" (establishing: 1)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.857: watch_established: "/system/proxy/ftp/" (establishing: 1)
(fwupdmgr:34219): dconf-DEBUG: 19:01:39.857: watch_established: "/system/proxy/socks/" (establishing: 1)
(fwupdmgr:34219): Fwupd-DEBUG: 19:01:39.883: Emitting ::status-changed() [idle]
Devices with no available firmware updates: 
 • Integrated Camera
ThinkPad Universal USB-C Dock is not currently updatable: (null)
```

**Steps to Reproduce**
`fwupdmg update`

**Expected behavior**
Update works succesfully.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
➜  ~ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.fwupd-efi     1.3
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
compile   com.hughsie.libjcat           0.1.10
runtime   org.kernel                    5.19.12-300.fc37.x86_64
compile   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.gusb          0.3.10
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): dnf

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```shell
➜  ~ fwupdmgr get-devices --show-all-devices
LENOVO 21A1S00D00
│
├─Unknown Device:
│     Device ID:          f685512aa07369c9e77742acef941d779d31e766
│     GUID:               37b440a9-2473-5087-a39b-db84f32a8ed8 ← GPIO\ID_AMDI0030:00
│   
├─Integrated Camera:
[... 204 lines cut ...]
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
└─Unifying Receiver:
      Device ID:          2dcbab59e6ef4d5f44aafbc2f023cd5b78441f73
      Summary:            Miniaturised USB wireless receiver
      Current version:    RQR12.11_B0032
      Bootloader Version: BOT01.04_B0016
      Vendor:             HIDRAW:0x046D|USB:0x046D
      Install Duration:   30 seconds
      GUIDs:              9d131a0c-a606-580f-8eda-80587250b8d6
                          279ed287-3607-549e-bacc-f873bb9838c4 ← HIDRAW\VEN_046D&DEV_C52B
                          fcf55bf5-767b-51ce-9c17-f6f538c4ee9f ← HIDRAW\VEN_046D&DEV_C52B&REV_00
      Device Flags:       • Updatable
                          • Signed Payload
```

**Additional questions**

- Operating system and version: Fedora 37
- Have you tried rebooting? Sure
- Is this a regression? Don't know

hughsie, 2022-09-30:
Do you have to remove the dock USBC and reconnect it? It might take a few mins to "come back"

hughsie, 2022-10-01:
> even though it is essential information in my opinion

I think vendors are supposed to include this in the update post-install message. I've added it on the update now and emailed the uploader. If you give it a couple of hours and then reinstall the update then it should show you a message.

closed 2022-10-01 by hughsie