fwupd/fwupd #5114: Unable to update Lenovo Z13 system firmware

vincentbernat, 2022-10-03:
**Describe the bug**
When trying to upgrade Lenovo Z13 system firmware from 0.1.21 to 0.1.22, nothing happens on reboot. `fwupdmgr` asks for the same update again and again.

**Steps to Reproduce**
```
 07:50 ❱ fwupdmgr update
Devices with no available firmware updates:
 • ELAN06A0:00 04F3:3231
 • Integrated Camera
 • Micron MTFDKCD512TFK
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Embedded Controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.21 to 0.1.22?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ CHANGES IN THIS RELEASE Version 1.22 -Important updates: Nothing -New        ║
║ functions or enhancements: Nothing -Problem fixes: Fixed an issue where      ║
║ temperature of palm rest might raising. (ThinkPad Z16 Gen 1 only)            ║
║                                                                              ║
║ 21D2CTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Waiting…                 [***************************************]
Writing…                 [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Waiting…                 [***************************************]
 • UEFI dbx

An update requires a reboot to complete. Restart now? [y|N]:
```

**Expected behavior**
The firmware should update on boot.

**fwupd version information**

```console
$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.fwupd-efi     1.0
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.19.0-2-amd64
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.gusb          0.3.10
```

Installed with apt on Debian unstable.

**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.

```console
$ fwupdmgr get-devices --show-all-devices
LENOVO 21D2CTO1WW
[... 251 lines cut ...]
$ tree /boot/efi
/boot/efi
└── EFI
    └── debian
        ├── BOOTX64.CSV
        ├── fbx64.efi
        ├── fw
        │   └── fwupd-ce5cfa7d-0ce2-4974-b866-a367f9e85481.cap
        ├── fwupdx64.efi
        ├── grub.cfg
        ├── grubx64.efi
        ├── mmx64.efi
        └── shimx64.efi

3 directories, 8 files
```

**Additional questions**

- Operating system and version: Debian unstable
- Have you tried rebooting? yes
- Is this a regression? dunno
- Are you using an NVMe disk? yes
- Is secure boot enabled? yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? I'll check, but manually selecting the Linux-Firmware-Updater entry does not help.

hughsie, 2022-10-03:
Does it work if you disable secure boot? If it does, I'll open a bug against Lenovo.

superm1, 2022-10-03:
Given you're taking shim from debian unstable, is it safe to assume you have the latest shim [15.6-1](https://packages.debian.org/source/unstable/shim), shim-signed [1+15.6+1](https://packages.debian.org/source/unstable/shim-helpers-amd64-signed) and fwupd-amd64-signed [1:1.2+3](https://packages.debian.org/source/unstable/fwupd-amd64-signed)?  Can you please double check these didn't get held back from some upgrade for some reason?

vincentbernat, 2022-10-03:
Yes, I am up-to-date. Moreover, the installed EFI binaries are in-sync.
```
 20:47 CEST ❱ md5sum fwupdx64.efi /usr/libexec/fwupd/efi/fwupdx64.efi.signed shimx64.efi /usr/lib/shim/shimx64.efi.signed

b96e88fb7789fdd09c4024689d0594e3  fwupdx64.efi
b96e88fb7789fdd09c4024689d0594e3  /usr/libexec/fwupd/efi/fwupdx64.efi.signed
cefed90870185ef689f004e686de8fce  shimx64.efi
cefed90870185ef689f004e686de8fce  /usr/lib/shim/shimx64.efi.signed
```

Disabling SecureBoot and asking to execute "Linux-Firmware-Updater" does not help. However, applying again the update works (but for some reason, I was first sent to the BIOS?).

superm1, 2022-10-03:
Disabling secure boot and just executing `Linux-Firmware-Updater` before staging it again with fwupd will use shim to chainload it.  Running the update again from fwupd will replace the `fwupdx64.efi` on the ESP with the unsigned version and chain the `Boot####` variable to no longer use shim to chainload.

I don't think we're narrowed down to which component has the problem, but it certainly seems like a shim bug..
https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=968997 isn't closed yet.  I think that's it.

vincentbernat, 2022-10-03:
Thanks. Yes, that's it. shim-unsigned is 15.6-1, but shim-signed is still 1.38+15.4-7. I don't know how it works, but that's not the first time the signed stuff is quite late compared to the unsigned stuff (we had fwupd in the same situation a year ago). So, I suppose we could close this issue.

superm1, 2022-10-03:
Ah it's also https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=995155.

You should leave some comments on those upstream bugs.  Agree, closing it here.

superm1, 2022-10-03:
BTW - as a workaround for the future, there is a "GRUB" mode for `/etc/fwupd/uefi_capsule.conf` via the key `EnableGrubChainLoad`.  It was envisioned exactly for this kind of bug where something is wrong with SHIM, but you use GRUB already so you can load from there instead.

closed 2022-10-03 by superm1

vincentbernat, 2022-10-03:
I have left a comment on the other bug. I'll just disable SecureBoot since on Debian, this is not really secure as the initramfs can be tampered with unless you enroll your own keys (and this is not really documented either).

vincentbernat, 2022-10-03:
FI, on https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=968997:

> We've had problems in submitting shim 15.6 to Microsoft for
> signing. We're working on a solution, but it's going to take a little
> longer yet.