fwupd/fwupd #5117: fwupd hangs forever after SIGTERM

real-or-random, 2022-10-03:
**Describe the bug**
fwupd often doesn't shutdown after receiving SIGTERM on system shutdown.

This seems to be related to a certain USB device, namely a USB hub on which is part of the "Acer Nitro XV282K" monitor. Its an internal hub which has an "Acer Sensor Hub" device (not a USB hub) connected to it. I think this is supposed to expose certain sensors built in the monitor to the OS but I don't know and I don't use it.

With `--verbose`, the last messages of `fwupd` (long before the SIGTERM are):
```
Sep 15 12:52:27 fwupd[5657]: 10:52:27:0242 FuMain               Called GetDevices()
Sep 15 12:52:27 fwupd[5657]: 10:52:27:0242 FuMain               Called GetDevices()
Sep 15 12:52:27 fwupd[5657]: 10:52:27:0243 FuMain               Called GetDevices()
Sep 15 12:52:27 fwupd[5657]: 10:52:27:0259 FuMain               Called GetDevices()
Sep 15 12:52:27 fwupd[5657]: 10:52:27:0677 FuDevice             using ed9a26c4705263c02531076751e1ea1b176f137f for usb:03:00:04:01:02:01:04
```
where that USB path corresponds to that device. So this is where it hangs.

But it doesn't always hang it this stage. I often see this the log after the `using ed9a26c4705263c02531076751e1ea1b176f137f for usb:03:00:04:01:02:01:04` message:
```
Sep 15 14:11:39 fwupd[2139]: 12:11:39:0928 FuEngine             failed to add device usb:03:00:04:01:02:01:04: failed to get static tool info from device: USB error on device 05e3:0610 : Pipe Error [-9]
```

As a workaround, I tried to blacklist the device using an udev rule:
```
SUBSYSTEM=="usb", ATTR{idVendor}=="05e3", ATTR{idProduct}=="0610", ACTION=="add", ATTR{authorized}="0"
```

This doesn't solve the problem but it *changes* it. Now there are no messages with `usb:03:00:04:01:02:01:04` in the log anymore, and `fwupd` doesn't hang entirely. However, it still doesn't shutdown entirely:

```
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0743 FuMain               Received SIGTERM
Oct 03 13:47:23 systemd[1]: Stopping Firmware update daemon...
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0813 FuPluginUpower       failed to query lid state
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0813 FuContext            lid state now unknown
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0813 FuPluginUpower       failed to query lid state
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0814 FuContext            battery state now unknown
Oct 03 13:47:23 fwupd[2337]: 11:47:23:0814 FuContext            battery level now 101
Oct 03 13:50:23 systemd[1]: fwupd.service: State 'stop-sigterm' timed out. Killing.
Oct 03 13:50:23 systemd[1]: fwupd.service: Killing process 2337 (fwupd) with signal SIGKILL.
Oct 03 13:50:23 systemd[1]: fwupd.service: Killing process 2353 (gdbus) with signal SIGKILL.
Oct 03 13:50:23 systemd[1]: fwupd.service: Main process exited, code=killed, status=9/KILL
Oct 03 13:50:23 systemd[1]: fwupd.service: Failed with result 'timeout'.
```

**Steps to Reproduce**
Well. I see this *sometimes* when I try to shutdown with the monitor plugged in. But I don't have a reliable way of reproducing this. 

**Expected behavior**
fwupd terminates.

**fwupd version information**

```shell
fwupdmgr --version

runtime   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.0
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.12-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.gusb          0.4.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
`pacman`

**fwupd device information**
(Leaving this out for privacy reasons. I don't think it's required. --- The suspect device is not recognized and not listed there anyway.)

```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version:
Arch Linux
- Is this a regression?
No

superm1, 2022-10-03:
As a better W/A can you maybe add the GUID to the `BlacklistDevices=` key in `/etc/fwupd/daemon.conf`?

real-or-random, 2022-10-03:
How can I get the GUID if the device isn't even recognized / shown by `fwupdmgr get-devices --show-all-devices`?

superm1, 2022-10-03:
I had assumed the GUID you posted above may work (`ed9a26c4705263c02531076751e1ea1b176f137f `)

superm1, 2022-10-03:
Btw is the kernel showing any messages about this device when you plug it in?  A snippet of the `dmesg` would be helpful to cross reference I think.

hughsie, 2022-10-04:
Is it possible to get a backtrace for all threads when it's hung? e.g. a gdb `t a a bt`?

real-or-random, 2022-10-04:
> ed9a26c4705263c02531076751e1ea1b176f137f

This is string is longer than a GUID.

> Btw is the kernel showing any messages about this device when you plug it in? A snippet of the `dmesg` would be helpful to cross reference I think.

```
Oct 03 13:51:48 kernel: usb 3-4.1.2.1.4: new high-speed USB device number 41 using xhci_hcd
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4: New USB device found, idVendor=05e3, idProduct=0610, bcdDevice=60.52
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4: Product: USB2.0 Hub
Oct 03 13:51:49 kernel: hub 3-4.1.2.1.4:1.0: USB hub found
Oct 03 13:51:49 kernel: hub 3-4.1.2.1.4:1.0: 2 ports detected
[...]
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: new full-speed USB device number 43 using xhci_hcd
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: New USB device found, idVendor=042e, idProduct=ac0a, bcdDevice= 2.08
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: Product: Sensor Hub
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: Manufacturer: Acer Inc.
Oct 03 13:51:49 kernel: usb 3-4.1.2.1.4.2: SerialNumber: 7BCD2C010000000
Oct 03 13:51:49 kernel: hid-generic 0003:042E:AC0A.0010: hiddev98,hidraw6: USB HID v1.11 Device [Acer Inc. Sensor Hub] on usb-0000:00:14.0-4.1.2.1.4.2/input0
```

> Is it possible to get a backtrace for all threads when it's hung? e.g. a gdb `t a a bt`?

I'll try to get one.

PHLAK, 2022-10-04:
I believe this is the same issue I'm having. On shutdown I often see the following message that holds up shutdown for 3 minutes.

```
A stop job is running for Firmware update daemon (42s / 3min)
```

Here's the journal logs for the session leading up to this.

```
Oct 03 08:20:49 Blurr systemd[1]: Starting Firmware update daemon...
Oct 03 08:20:49 Blurr fwupd[2085]: 15:20:49:0865 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file "/sys/class/firmware-attributes/thinklmi/attributes/TotalMemoryEncryption/type": No such file or directory)
Oct 03 08:20:53 Blurr fwupd[2085]: 15:20:53:0177 FuEngine             failed to add device /sys/devices/pci0000:00/0000:00:1f.5: failed to mmap SPIBAR: Operation not permitted
Oct 03 08:20:53 Blurr systemd[1]: Started Firmware update daemon.
Oct 04 08:09:06 Blurr fwupd[2085]: 15:09:06:0737 FuMain               Received SIGTERM
Oct 04 08:09:06 Blurr systemd[1]: Stopping Firmware update daemon...
Oct 04 08:09:06 Blurr fwupd[2085]: 15:09:06:0816 FuPluginUpower       failed to query lid state
Oct 04 08:09:06 Blurr fwupd[2085]: 15:09:06:0819 FuPluginUpower       failed to query lid state
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: State 'stop-sigterm' timed out. Killing.
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: Killing process 2085 (fwupd) with signal SIGKILL.
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: Killing process 2093 (GUsbEventThread) with signal SIGKILL.
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: Main process exited, code=killed, status=9/KILL
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: Failed with result 'timeout'.
Oct 04 08:12:06 Blurr systemd[1]: Stopped Firmware update daemon.
Oct 04 08:12:06 Blurr systemd[1]: fwupd.service: Consumed 3.692s CPU time.
```

**OS:** Arch

```
$ pacman --query fwupd
fwupd 1.8.5-1
```

hughsie, 2022-10-04:
@PHLAK a backtrace would be wonderful, the GUsbEventThread would be my suspicion although nothing has changed there for some time.

invidian, 2022-10-04:
I've also started experiencing this recently with version `1.8.5`. I'll try to get the backtrace from it.

hughsie, 2022-10-04:
>  I've also started experiencing this recently with version 1.8.5

Uh oh. I wonder if this is BOS descriptor related. What libgusb version out of interest?

hughsie, 2022-10-05:
> I wonder if this is BOS descriptor related.

Does `sudo gusbcmd save` trigger the same hang?

hughsie, 2022-10-05:
I think I just captured this on my device-test box -- it seemed to happen when I plugged in a dock with many USB devices connected:

	(gdb) t a a bt

	Thread 5 (Thread 0x7fbaf97fc6c0 (LWP 124659) "gdbus"):
	#0  0x00007fbb0b22dd2f in poll () from target:/lib64/libc.so.6
	#1  0x00007fbb0b3f833d in g_main_context_iterate.constprop () from target:/lib64/libglib-2.0.so.0
	#2  0x00007fbb0b3a220f in g_main_loop_run () from target:/lib64/libglib-2.0.so.0
	#3  0x00007fbb0b5a179a in gdbus_shared_thread_func.lto_priv () from target:/lib64/libgio-2.0.so.0
	#4  0x00007fbb0b3cc772 in g_thread_proxy () from target:/lib64/libglib-2.0.so.0
	#5  0x00007fbb0b1b9f9d in start_thread () from target:/lib64/libc.so.6
	#6  0x00007fbb0b23b5d0 in clone3 () from target:/lib64/libc.so.6

	Thread 4 (Thread 0x7fbafa7fe6c0 (LWP 124657) "GUsbEventThread"):
	#0  0x00007fbb0b23343d in syscall () from target:/lib64/libc.so.6
	#1  0x00007fbb0b3f559c in g_mutex_lock_slowpath () from target:/lib64/libglib-2.0.so.0
	#2  0x00007fbb0b0ccd24 in g_mutex_locker_new (mutex=0x55bff27e0670) at /usr/include/glib-2.0/glib/gthread.h:335
	#3  g_usb_context_hotplug_cb (ctx=<optimized out>, dev=0x7fbaf000cc00, event=LIBUSB_HOTPLUG_EVENT_DEVICE_ARRIVED, user_data=0x55bff27e0680) at ../gusb/gusb-context.c:561
	#4  0x00007fbb0a771b5c in handle_events () from target:/lib64/libusb-1.0.so.0
	#5  0x00007fbb0a773150 in libusb_handle_events_timeout_completed () from target:/lib64/libusb-1.0.so.0
	#6  0x00007fbb0b0ccdfe in g_usb_context_event_thread_cb (data=<optimized out>) at ../gusb/gusb-context.c:809
	#7  0x00007fbb0b3cc772 in g_thread_proxy () from target:/lib64/libglib-2.0.so.0
	#8  0x00007fbb0b1b9f9d in start_thread () from target:/lib64/libc.so.6
	#9  0x00007fbb0b23b5d0 in clone3 () from target:/lib64/libc.so.6

	Thread 3 (Thread 0x7fbafafff6c0 (LWP 124656) "libusb_event"):
	#0  0x00007fbb0b22dd2f in poll () from target:/lib64/libc.so.6
	#1  0x00007fbb0a76d8d3 in linux_udev_event_thread_main () from target:/lib64/libusb-1.0.so.0
	#2  0x00007fbb0b1b9f9d in start_thread () from target:/lib64/libc.so.6
	#3  0x00007fbb0b23b5d0 in clone3 () from target:/lib64/libc.so.6

	Thread 2 (Thread 0x7fbafbfff6c0 (LWP 124649) "gmain"):
	#0  0x00007fbb0b22dd2f in poll () from target:/lib64/libc.so.6
	#1  0x00007fbb0b3f833d in g_main_context_iterate.constprop () from target:/lib64/libglib-2.0.so.0
	#2  0x00007fbb0b39fec0 in g_main_context_iteration () from target:/lib64/libglib-2.0.so.0
	#3  0x00007fbb0b3a1b51 in glib_worker_main () from target:/lib64/libglib-2.0.so.0
	#4  0x00007fbb0b3cc772 in g_thread_proxy () from target:/lib64/libglib-2.0.so.0
	#5  0x00007fbb0b1b9f9d in start_thread () from target:/lib64/libc.so.6

	Thread 1 (Thread 0x7fbb0974cb00 (LWP 124643) "fwupd"):
	#0  0x00007fbb0b1b6dc0 in __lll_lock_wait () from target:/lib64/libc.so.6
	#1  0x00007fbb0b1bd282 in pthread_mutex_lock@@GLIBC_2.2.5 () from target:/lib64/libc.so.6
	#2  0x00007fbb0a772099 in libusb_lock_events () from target:/lib64/libusb-1.0.so.0
	#3  0x00007fbb0a772527 in libusb_close () from target:/lib64/libusb-1.0.so.0
	#4  0x00007fbb0b0ce482 in g_usb_device_close (self=0x55bff27dcf60, error=0x7fffd4c62530) at ../gusb/gusb-device.c:1184
	#5  0x00007fbb0b782859 in fu_device_locker_finalize (obj=0x7fbae4003980) at ../libfwupdplugin/fu-device-locker.c:46
	#6  0x00007fbb0ba82a02 in g_object_unref () from target:/lib64/libgobject-2.0.so.0
	#7  0x00007fbb0b7c43b6 in fu_usb_device_close (device=<optimized out>, error=<optimized out>) at ../libfwupdplugin/fu-usb-device.c:496
	#8  0x00007fbb0b78f90b in fu_device_close_internal (self=0x55bff60b42e0, error=0x7fffd4c62610) at ../libfwupdplugin/fu-device.c:4512
	#9  0x00007fbb0b782859 in fu_device_locker_finalize (obj=0x55bff27cf380) at ../libfwupdplugin/fu-device-locker.c:46
	#10 0x00007fbb0ba82a02 in g_object_unref () from target:/lib64/libgobject-2.0.so.0
	#11 0x00007fbb0b7ac5fe in glib_autoptr_clear_GObject (_ptr=0x55bff27cf380) at /usr/include/glib-2.0/gobject/gobject-autocleanups.h:29
	#12 glib_autoptr_clear_FuDeviceLocker (_ptr=0x55bff27cf380) at ../libfwupdplugin/fu-device-locker.h:13
	#13 glib_autoptr_cleanup_FuDeviceLocker (_ptr=<synthetic pointer>) at ../libfwupdplugin/fu-device-locker.h:13
	#14 fu_plugin_backend_device_added (error=0x7fffd4c62730, device=0x55bff28772a0, self=0x55bff2873cd0) at ../libfwupdplugin/fu-plugin.c:1556
	#15 fu_plugin_runner_backend_device_added (self=0x55bff2873cd0, device=device@entry=0x55bff2820840, error=error@entry=0x7fffd4c62730) at ../libfwupdplugin/fu-plugin.c:1636
	#16 0x00007fbb0b881354 in fu_engine_backend_device_added_run_plugin (error=0x7fffd4c62730, progress=<optimized out>, plugin_name=0x55bff27cb520 "vli", device=0x55bff2820840, self=0x55bff27aa010) at ../src/fu-engine.c:7032
	#17 fu_engine_backend_device_added_run_plugins (progress=0x55bff27a9260, device=0x55bff2820840, self=0x55bff27aa010) at ../src/fu-engine.c:7062
	#18 fu_engine_backend_device_added (self=0x55bff27aa010, device=0x55bff2820840, progress=0x55bff28a74a0) at ../src/fu-engine.c:7131
	#19 0x00007fbb0b88150b in fu_engine_backend_device_added_cb (backend=<optimized out>, device=0x55bff2820840, self=0x55bff27aa010) at ../src/fu-engine.c:7139
	#20 0x00007fbb0ba75fe0 in g_closure_invoke () from target:/lib64/libgobject-2.0.so.0
	#21 0x00007fbb0baa3d76 in signal_emit_unlocked_R.isra.0 () from target:/lib64/libgobject-2.0.so.0
	#22 0x00007fbb0ba933da in g_signal_emit_valist () from target:/lib64/libgobject-2.0.so.0
[... cut ...]
	#30 g_usb_context_add_device (self=0x55bff27e0680, dev=<optimized out>) at ../gusb/gusb-context.c:323
	#31 0x00007fbb0b0d19bc in g_usb_context_idle_hotplug_cb (user_data=<optimized out>) at ../gusb/gusb-context.c:535
	#32 0x00007fbb0b3a1c32 in g_idle_dispatch () from target:/lib64/libglib-2.0.so.0
	#33 0x00007fbb0b3a2c3f in g_main_context_dispatch () from target:/lib64/libglib-2.0.so.0
	#34 0x00007fbb0b3f83c8 in g_main_context_iterate.constprop () from target:/lib64/libglib-2.0.so.0
	#35 0x00007fbb0b3a220f in g_main_loop_run () from target:/lib64/libglib-2.0.so.0

invidian, 2022-10-05:
@hughsie I just checked poked by your notification and my daemon is stuck again, so below is my stacktrace. I'm using Thunderbolt Docking station Dell TB16 with another USB hub connected, so your theory might be right.
```
(gdb) t a a bt

Thread 5 (Thread 0x7f383f7fe6c0 (LWP 2769638) "gdbus"):
#0  0x00007f3848b49e5f in poll () from target:/usr/lib/libc.so.6
#1  0x00007f3848ea4e2f in ?? () from target:/usr/lib/libglib-2.0.so.0
#2  0x00007f3848e4dd7f in g_main_loop_run () from target:/usr/lib/libglib-2.0.so.0
#3  0x00007f38490a323c in ?? () from target:/usr/lib/libgio-2.0.so.0
#4  0x00007f3848e7bc55 in ?? () from target:/usr/lib/libglib-2.0.so.0
#5  0x00007f3848ad574d in ?? () from target:/usr/lib/libc.so.6
#6  0x00007f3848b57700 in ?? () from target:/usr/lib/libc.so.6

Thread 4 (Thread 0x7f3844ffe6c0 (LWP 2769636) "GUsbEventThread"):
#0  0x00007f3848b4f59d in syscall () from target:/usr/lib/libc.so.6
#1  0x00007f3848e9ea8d in ?? () from target:/usr/lib/libglib-2.0.so.0
#2  0x00007f3848da9337 in ?? () from target:/usr/lib/libgusb.so.2
#3  0x00007f3848176eef in ?? () from target:/usr/lib/libusb-1.0.so.0
#4  0x00007f3848178438 in libusb_handle_events_timeout_completed () from target:/usr/lib/libusb-1.0.so.0
#5  0x00007f3848da9417 in ?? () from target:/usr/lib/libgusb.so.2
#6  0x00007f3848e7bc55 in ?? () from target:/usr/lib/libglib-2.0.so.0
#7  0x00007f3848ad574d in ?? () from target:/usr/lib/libc.so.6
#8  0x00007f3848b57700 in ?? () from target:/usr/lib/libc.so.6

Thread 3 (Thread 0x7f38457ff6c0 (LWP 2769635) "libusb_event"):
#0  0x00007f3848b49e5f in poll () from target:/usr/lib/libc.so.6
#1  0x00007f3848172a95 in ?? () from target:/usr/lib/libusb-1.0.so.0
#2  0x00007f3848ad574d in ?? () from target:/usr/lib/libc.so.6
#3  0x00007f3848b57700 in ?? () from target:/usr/lib/libc.so.6

Thread 2 (Thread 0x7f38471ff6c0 (LWP 2769633) "gmain"):
#0  0x00007f3848b49e5f in poll () from target:/usr/lib/libc.so.6
#1  0x00007f3848ea4e2f in ?? () from target:/usr/lib/libglib-2.0.so.0
#2  0x00007f3848e4d0d2 in g_main_context_iteration () from target:/usr/lib/libglib-2.0.so.0
#3  0x00007f3848e4d122 in ?? () from target:/usr/lib/libglib-2.0.so.0
#4  0x00007f3848e7bc55 in ?? () from target:/usr/lib/libglib-2.0.so.0
#5  0x00007f3848ad574d in ?? () from target:/usr/lib/libc.so.6
#6  0x00007f3848b57700 in ?? () from target:/usr/lib/libc.so.6

Thread 1 (Thread 0x7f384755da00 (LWP 2769632) "fwupd"):
#0  0x00007f3848ad2306 in ?? () from target:/usr/lib/libc.so.6
#1  0x00007f3848ad4e24 in pthread_cond_timedwait () from target:/usr/lib/libc.so.6
#2  0x00007f3848177d3f in libusb_wait_for_event () from target:/usr/lib/libusb-1.0.so.0
#3  0x00007f38481783c9 in libusb_handle_events_timeout_completed () from target:/usr/lib/libusb-1.0.so.0
#4  0x00007f3848178505 in libusb_handle_events_completed () from target:/usr/lib/libusb-1.0.so.0
#5  0x00007f384817857a in ?? () from target:/usr/lib/libusb-1.0.so.0
#6  0x00007f3848178c3d in libusb_control_transfer () from target:/usr/lib/libusb-1.0.so.0
#7  0x00007f384817944f in libusb_get_string_descriptor_ascii () from target:/usr/lib/libusb-1.0.so.0
#8  0x00007f3848db0cae in g_usb_device_get_string_descriptor () from target:/usr/lib/libgusb.so.2
#9  0x00007f38492e113d in fu_usb_device_setup (device=0x564b2752b2e0, error=0x7ffcbeb83ec0) at ../fwupd-1.8.5/libfwupdplugin/fu-usb-device.c:314
#10 0x00007f38467c85ee in fu_genesys_usbhub_device_setup (device=0x564b2752b2e0, error=0x7ffcbeb83ec0) at ../fwupd-1.8.5/plugins/genesys/fu-genesys-usbhub-device.c:762
#11 0x00007f38492acb07 in fu_device_setup (self=self@entry=0x564b2752b2e0, error=error@entry=0x7ffcbeb83ec0) at ../fwupd-1.8.5/libfwupdplugin/fu-device.c:4735
#12 0x00007f38492acc5b in fu_device_open_internal (self=0x564b2752b2e0, error=0x7ffcbeb83ec0) at ../fwupd-1.8.5/libfwupdplugin/fu-device.c:4436
#13 0x00007f38492a2be4 in fu_device_locker_new_full (device=0x564b2752b2e0, open_func=0x7f38492accc0 <fu_device_open>, close_func=0x7f38492ab700 <fu_device_close>, error=0x7ffcbeb83ec0) at ../fwupd-1.8.5/libfwupdplugin/fu-device-locker.c:202
#14 0x00007f38492c8c04 in fu_plugin_backend_device_added (error=0x7ffcbeb83ec0, device=0x564b275d99c0, self=0x564b27331d50) at ../fwupd-1.8.5/libfwupdplugin/fu-plugin.c:1587
#15 fu_plugin_runner_backend_device_added (self=0x564b27331d50, device=device@entry=0x564b275d99c0, error=error@entry=0x7ffcbeb83ec0) at ../fwupd-1.8.5/libfwupdplugin/fu-plugin.c:1626
#16 0x0000564b263622a6 in fu_engine_backend_device_added_run_plugin (error=0x7ffcbeb83ec0, progress=<optimized out>, plugin_name=0x564b275f88b0 "genesys", device=0x564b275d99c0, self=0x564b27289010) at ../fwupd-1.8.5/src/fu-engine.c:7031
#17 fu_engine_backend_device_added_run_plugins (progress=0x564b275ca410, device=0x564b275d99c0, self=0x564b27289010) at ../fwupd-1.8.5/src/fu-engine.c:7061
#18 fu_engine_backend_device_added (self=self@entry=0x564b27289010, device=device@entry=0x564b275d99c0, progress=progress@entry=0x564b275ca390) at ../fwupd-1.8.5/src/fu-engine.c:7130
#19 0x0000564b26362508 in fu_engine_backend_device_added_cb (backend=<optimized out>, device=0x564b275d99c0, self=0x564b27289010) at ../fwupd-1.8.5/src/fu-engine.c:7138
#20 0x00007f3848f4a210 in g_closure_invoke () from target:/usr/lib/libgobject-2.0.so.0
#21 0x00007f3848f77e88 in ?? () from target:/usr/lib/libgobject-2.0.so.0
#22 0x00007f3848f67f35 in g_signal_emit_valist () from target:/usr/lib/libgobject-2.0.so.0
[... cut ...]
#28 0x00007f3848f681c4 in g_signal_emit () from target:/usr/lib/libgobject-2.0.so.0
#29 0x00007f3848dad8ea in ?? () from target:/usr/lib/libgusb.so.2
#30 0x00007f3848dadf47 in ?? () from target:/usr/lib/libgusb.so.2
#31 0x00007f3848e4e81b in g_main_context_dispatch () from target:/usr/lib/libglib-2.0.so.0
#32 0x00007f3848ea4ec9 in ?? () from target:/usr/lib/libglib-2.0.so.0
#33 0x00007f3848e4dd7f in g_main_loop_run () from target:/usr/lib/libglib-2.0.so.0
#34 0x0000564b263523f5 in main (argc=<optimized out>, argv=<optimized out>) at ../fwupd-1.8.5/src/fu-main.c:213
```

hughsie, 2022-10-05:
Half the fix: https://github.com/hughsie/libgusb/pull/81

hughsie, 2022-10-05:
The other half: https://github.com/hughsie/libgusb/pull/82

superm1, 2022-10-05:
This is triggered because of the BOS descriptor lookups, right?  Maybe we should gate that code on libgusb version+1 for now then instead to avoid hitting it in fwupd?

hughsie, 2022-10-05:
> This is triggered because of the BOS descriptor lookups, right

It's made worse of the BOS lookups (because they're slower), but the string g_usb_device_get_string_descriptor() can trigger it too. It's anything that can trigger a usbi_cond_timedwait

superm1, 2022-10-05:
@invidian there is a new feature that landed in 1.8.5 by this commit https://github.com/fwupd/fwupd/commit/bfebede490f95ba049ee5dee18313d154b1253d6 and various fixups for it.  That feature is causing some longer lookups and causing these hangs.

superm1, 2022-10-05:
> > This is triggered because of the BOS descriptor lookups, right
> 
> It's made worse of the BOS lookups (because they're slower), but the string g_usb_device_get_string_descriptor() can trigger it too. It's anything that can trigger a usbi_cond_timedwait

OK.. I still think we should gate the code on the newer version.  It's going to take longer for libgusb to get reviewed/released/into distros.

hughsie, 2022-10-05:
> OK.. I still think we should gate the code on the newer version

Ack, agree.

superm1, 2022-10-12:
the W/A we discussed is in 1.8.6.

bulletmark, 2022-10-12:
I'm on Arch which updated fwupd recently:
```
2022-10-08 09:40:30 fwupd 1.8.5-1 -> 1.8.6-1
```
and just got this bug as per the following 2 adjacent lines in my shutdown journal:
```
Oct 11 07:18:35 lt systemd[1]: NetworkManager.service: Consumed 11.922s CPU time.
Oct 11 07:21:34 lt systemd[1]: fwupd.service: State 'stop-sigterm' timed out. Killing.
```

kparal, 2022-10-14:
I have `fwupd-1.8.6-1.fc37.x86_64` on Fedora 37 and I believe I see exactly this issue. Every time I boot my laptop, connect my docking station and try to reboot/poweroff, the `fwupd.service` times out (a 3 minutes delay before it gets killed and the laptop finally reboots). This only happens when I boot the laptop standalone, log in, and connect the docking station **afterwards**. If I boot with the dock already attached, this problem doesn't happen.

I have Thinkpad P1 Gen 3 and Thinkpad Thunderbolt 3 Dock Gen 2, and the same problem happens even on Thinkpad P1 Gen 4 and the same docking station for my colleague.

When grepping for `fwupd` in the journal with the hanged reboot:
```
$ grep fwupd journal-fwupd-hang.txt 
říj 14 10:42:38 hydra systemd[1]: Started fwupd-refresh.timer - Refresh fwupd metadata regularly.
říj 14 10:42:48 hydra systemd[1]: Starting fwupd.service - Firmware update daemon...
říj 14 10:42:48 hydra fwupd[2174]: 08:42:48:0765 FuBiosSettings       KERNEL BUG: 'type' attribute not exported: (failed to load type: Failed to open file "/sys/class/firmware-attributes/thinklmi/attributes/SecureBoot/type": No such file or directory)
říj 14 10:42:50 hydra systemd[1]: Started fwupd.service - Firmware update daemon.
říj 14 10:42:50 hydra audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
říj 14 10:42:50 hydra gnome-software[1823]: enabled plugins: appstream, fwupd, os-release, repos, epiphany, fedora-langpacks, fedora-pkgdb-collections, flatpak, hardcoded-blocklist, modalias, packagekit, rewrite-resource, generic-updates, malcontent, provenance, icons, provenance-license
říj 14 10:43:12 hydra systemd[1]: fwupd-refresh.timer: Deactivated successfully.
říj 14 10:43:12 hydra systemd[1]: Stopped fwupd-refresh.timer - Refresh fwupd metadata regularly.
říj 14 10:43:12 hydra systemd[1]: Stopping fwupd.service - Firmware update daemon...
říj 14 10:46:13 hydra systemd[1]: fwupd.service: State 'stop-sigterm' timed out. Killing.
říj 14 10:46:13 hydra systemd[1]: fwupd.service: Killing process 2174 (fwupd) with signal SIGKILL.
říj 14 10:46:13 hydra systemd[1]: fwupd.service: Killing process 2190 (gdbus) with signal SIGKILL.
říj 14 10:46:13 hydra systemd[1]: fwupd.service: Main process exited, code=killed, status=9/KILL
říj 14 10:46:13 hydra systemd[1]: fwupd.service: Failed with result 'timeout'.
říj 14 10:46:13 hydra systemd[1]: Stopped fwupd.service - Firmware update daemon.
říj 14 10:46:13 hydra audit[1]: SERVICE_STOP pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fwupd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=failed'
říj 14 10:46:13 hydra systemd[1]: fwupd.service: Consumed 1.058s CPU time.
```

There is a `KERNEL BUG: 'type' attribute not exported` line from #5077 , not sure if it's related. You can see the service was killed after 3 minutes.

Log files:
* [system journal with fwupd hanging during reboot](https://github.com/fwupd/fwupd/files/9784879/journal-fwupd-hang.txt)
* [package versions](https://github.com/fwupd/fwupd/files/9784882/rpm-qa.txt)
* [fwupdmgr get-devices](https://github.com/fwupd/fwupd/files/9784884/fwupd-devices.txt)
* [fwupdmgr --version](https://github.com/fwupd/fwupd/files/9784889/fwupd-version.txt)

I can reproduce the problem easily, so if I should provide any addition information, I'm happy to.

```
Fedora 37
fwupd-1.8.6-1.fc37.x86_64
libgusb-0.4.1-1.fc37.x86_64
Lenovo Thinkpad P1 Gen 3 and Lenovo Thinkpad Thunderbolt 3 Dock Gen 2
```

hughsie, 2022-10-14:
@kparal can you try the libgub package in https://copr.fedorainfracloud.org/coprs/rhughes/fwupd and then restart the fwupd service please? That fixes it for me.

kparal, 2022-10-14:
@hughsie I tested with `libgusb-0.4.2-0.161.20221014git.fc37.x86_64` and that indeed fixes the problem, `fwupd.service` no longer times out. At the same time, it makes `sudo` time out 30 seconds before showing a password prompt, the journal says:
```
říj 14 13:01:17 hydra sudo[3475]: pam_fprintd(sudo:auth): GetDevices failed: Connection timed out
```
And it also makes `fprintd.service` time out (90 seconds) during shutdown. So, some work is still needed :slightly_smiling_face:

hughsie, 2022-10-14:
> it also makes fprintd.service time out (90 seconds) during shutdown

Hmm, good catch. Will try with fprintd here.

hughsie, 2022-10-17:
> So, some work is still needed

Understatement of the weekend. Can you try libgusb 0.4.2-0.179.20221017git.fc37 from my COPR please.

hughsie, 2022-10-17:
Sorry, that was aimed at @kparal -- but anyone else can try the main branch of libgusb.

kparal, 2022-10-18:
> Can you try libgusb 0.4.2-0.179.20221017git.fc37 from my COPR please.

That seems to work fine, at least in my quick testing. No issues with `fwupd.service` nor `fprintd.service`.

hughsie, 2022-10-18:
Phew, me also! I'll merge that and tag a release. Thanks for the check.

hughsie, 2022-10-18:
I've released GUsb 0.4.2 this morning and am now doing the Fedora builds. Thanks everybody.

closed 2022-10-18 by hughsie