fwupd/fwupd #5121: 'Blocked executable in the ESP, ensure grub and shim are up to date': should people who run into this problem not be given more guidance?

LinuxOnTheDesktop, 2022-10-05:
**Describe the question**
On the version of fwupd that my distro provides, I see this (after, earlier, getting a 404 error):

```
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.If the installation fails, you will     ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly.You may have to        ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/EFI/BOOT/bkpbootx64.efi Authenticode checksum [007f4c95125713b112093e21663e2d23e3c1ae9ce4b5de0d58a297332336a2d8] is present in dbx
```

The only guidance I can find is [a locked thread](https://github.com/fwupd/fwupd/issues/5035) on this the fwupd bugtracker and this [terse fwupd wiki page](https://github.com/fwupd/fwupd/wiki/Blocked-executable-in-the-ESP,-ensure-grub-and-shim-are-up-to-date). The non-expert user seems stuck between non installing updates, on the one hand, and potentially breaking her system, on the other. So more advice, or at least somewhere to discuss the issue, would be welcome.

**fwupd version information**

```shell
fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.5
compile   org.freedesktop.gusb          0.3.10
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.15.0-48-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
```

superm1, 2022-10-05:
It's a wiki page. Perhaps you can edit it for some better guidance to people as you see fit?

LinuxOnTheDesktop, 2022-10-05:
I did not realise that you would welcome such an edit. Now that I know that you do: luckily, I have found out something useful . .

closed 2022-10-05 by LinuxOnTheDesktop

superm1, 2022-10-05:
> I did not realise that you would welcome such an edit. Now that I know that you do: luckily, I have found out something useful . .

Sure no problem. @hughsie and I are engineers. It's best to translate for us sometimes :)

LinuxOnTheDesktop, 2022-10-05:
By 'I have found out something useful' I meant: something that I can put - and, indeed, now have put - on the wiki page. :)

Zugschlus, 2023-01-01:
It would be really helpful if the error message would mention WHICH executable is the one that is flagged as blocked.

superm1, 2023-01-01:
It does tell you in the error message after the `:`