fwupd/fwupd #5122: latest snap fails to install on Ubuntu 20.04

bergkatten, 2022-10-05:
**snap install fwupd --classic fails with library error**
A clear and concise description of what the bug is.

**Steps to Reproduce**
# snap install fwupd 
error: This revision of snap "fwupd" was published using classic confinement and thus may perform
       arbitrary system changes outside of the security sandbox that snaps are usually confined to,
       which may put your system at risk.

       If you understand and want to proceed repeat the command including --classic.
# snap install fwupd --classic 
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": 
-----
systemctl: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /snap/fwupd/4389/lib/x86_64-linux-gnu/liblzma.so.5)
systemctl: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /snap/fwupd/4389/lib/x86_64-linux-gnu/liblzma.so.5)
-----)

**Expected behavior**
snap install should work

**fwupd version information**
# snap info fwupd 
name:      fwupd
summary:   Firmware updates for Linux
publisher: Richard Hughes
store-url: https://snapcraft.io/fwupd
contact:   https://github.com/fwupd/fwupd
license:   LGPL-2.1+
description: |
  The full fwupd firmware updating stack for Linux.  Can be used to perform updates on all supported
  devices via https://fwupd.org
snap-id: HpOj37PuyuaMUZY0NQhtwnp7oS5P8u5R
channels:
  latest/stable:    1.8.5-10-g34dfa8689 2022-09-22 (4389) 22MB classic
  latest/candidate: ↑                                          
  latest/beta:      ↑                                          
  latest/edge:      1.8.5-68-g1b6438e8f 2022-10-04 (4395) 21MB classic

```shell
fwupdmgr --version
```
fails
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
snap  
**fwupd device information**
Please provide the output of the fwupd devices recognized in your system.
fails
```shell
fwupdmgr get-devices --show-all-devices
```

**Additional questions**

- Operating system and version: Ubuntu 20.04.5 LTS
- Have you tried rebooting? Yes
- Is this a regression? happens since 1.8.4

superm1, 2022-10-05:
I can reproduce this; I suspect it's caused by [moving to core22](https://github.com/fwupd/fwupd/commit/8bb10ffaa747019e74c465de76c976dcf33c7998).  @valentindavid as you've been working on the confined version, maybe you have seen something like this or have some ideas?  Or @kenvandine any ideas?

superm1, 2022-10-05:
I wonder if maybe we need to be bundling libc6 in the snap or referring to libc6 in core22 snap (presuming it's there)?

valentindavid, 2022-10-06:
The confined version would never use the libc from host. So that is an issue that should not happened when confined.

In classic, in theory I suppose you would even need to use the dynamic linker to call any binary. Otherwise you might get a clash if the dynamic linker from host does not work from libc from core22. Though I suppose this issue might not happen that often.

I think the proper way to is to move to use confinement.

superm1, 2022-10-06:
I think we're still a bit from being able to switch to confined.

For the short term would adding libc6 to the classic snap be the right solution?

superm1, 2022-10-06:
I had a try with this on top of main:
```
diff --git a/snap/snapcraft.yaml b/snap/snapcraft.yaml
index b102f7587..f39296bee 100644
--- a/snap/snapcraft.yaml
+++ b/snap/snapcraft.yaml
@@ -124,6 +124,7 @@ parts:
     stage-packages:
       - libgcab-1.0-0
       - libarchive13
+      - libc6
       - libcurl4
       - libcbor0.8
       - libassuan0
```

But unfortunately, that leads to:
```
$ sudo snap install fwupd_1.8.4-113-g5a1126724_amd64.snap --dangerous --classic
error: cannot perform the following tasks:
- Run install hook of "fwupd" snap if present (run hook "install": signal: illegal instruction (core dumped))
```

majamee, 2022-10-13:
Similar issue with the latest snap package on Ubuntu 18.04:

```
fwupdmgr get-updates
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libselinux.so.1: no version information available (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libgio-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libgmodule-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.28' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.32' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0)
/snap/fwupd/4400/usr/lib/x86_64-linux-gnu/glib-2.0/gio-querymodules: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.34' not found (required by /snap/fwupd/4400/usr/lib/x86_64-linux-gnu/libglib-2.0.so.0)
/snap/fwupd/4400/bin/fwupdmgr: error while loading shared libraries: libssl.so.3: cannot open shared object file: No such file or directory
```

hughsie, 2022-10-13:
Just from the sidelines; snap doesn't seem like it provides much actual abstraction... I'm really wondering if it's worth continuing with.

majamee, 2022-10-13:
@hughsie Thank you for your quick response, would be sad to miss this option on Ubuntu, but I guess I understand your reasoning.

superm1, 2022-10-13:
> Just from the sidelines; snap doesn't seem like it provides much actual abstraction... I'm really wondering if it's worth continuing with.

At the core - we have some bug in the setup for the classic snap wrappers that it's not taking certain libraries from the core22 snap and is instead using the host OS.

I don't know the right way to solve it yet, but it's _probably_ something with `LD_LIBRARY_PATH`.

When we're ready to move to the confined snap (see description of #5062 for blockers) then the snap never gets to access any libraries from the host OS at all.

stucki, 2022-11-01:
What's the plan to fix this? Right now, fwupd doesn't work on my system. Is there any workaround?

hughsie, 2022-11-01:
> Is there any workaround

Don't use the snap? Can you use the flatpak version instead?

ClausHolbechArista, 2022-11-01:
It works if you uninstall the snap and install with `apt` instead.

stucki, 2022-11-01:
Installation via apt works, thanks. Unfortunatly, it means that I'm back at version 1.7.9 instead of the current 1.8.6.

mailsanchu, 2022-12-09:
Anybody able to find a workaround?

superm1, 2022-12-16:
We don't have a good solution for this right now.  Fundamentally the issue is that we're building against a newer core snap but this is a classic snap that uses libraries from outside of the core snap.

This should be solved by moving to a confined snap, which is under way.  You can see some of the progress for that at this issue and everything else it links to: https://github.com/fwupd/fwupd/pull/5062

superm1, 2022-12-16:
If you have a need for a newer version in Ubuntu 20.04, what you should do is either use the flatpak if it works for your use case or compile by hand.  

The hand compilation stuff isn't as bad as it used to be.  Use `./contrib/setup` and it will set up all the build deps from the Ubuntu archives and pypi.  Then if one of those deps can't be found it will either get installed from meson subproject or features will be disabled.

mailsanchu, 2022-12-16:
```
svarkey@svarkey-Precision-7760:/tmp/fwupd/build/src$ ./fwupdmgr
Failed to connect to daemon: Error calling StartServiceByName for org.freedesktop.fwupd: Unit snap.fwupd.fwupd.service not found.
svarkey@svarkey-Precision-7760:/tmp/fwupd/build/src$ 

```

mailsanchu, 2022-12-16:
I compiled it by hand and got the error above. Is it familiar to you ? Any work around

superm1, 2022-12-16:
You do need to install somewhere (like the default `/usr/local`).
Before you do your `ninja  -C build install` you should remove the snap to avoid conflicts.

mailsanchu, 2022-12-16:
[image]

mailsanchu, 2022-12-16:
Snap version is not installed. See above

superm1, 2022-12-16:
The d-bus activation appears to have not been set up.  Follow the step here for it: https://github.com/fwupd/fwupd/wiki/Compilation#configure-dbus-to-let-daemon-start

mailsanchu, 2022-12-16:
Restarting dbus fixed the issue . We should add that in the build instructions

mailsanchu, 2022-12-16:
[image]

mailsanchu, 2022-12-16:
That worked as well. 
Finally I see this error
`Blocked executable in the ESP, ensure grub and shim are up to date: /boot/efi/efi.factory/boot/bootx64.efi Authenticode checksum [2ea4cb6a1f1eb1d3dce82d54fde26ded243ba3e18de7c6d211902a594fe56788] is present in dbx`

mailsanchu, 2022-12-16:
Any other solution to this?

superm1, 2022-12-16:
> Any other solution to this?

See https://bugs.launchpad.net/dell/+bug/1990179

mailsanchu, 2022-12-16:
[image]
 Is it safe to install this

superm1, 2022-12-16:
> [image] Is it safe to install this

That's for you to decide.  You can extract it and review what it does.

mailsanchu, 2022-12-16:
I dont understand one thing here 
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║

mailsanchu, 2022-12-16:
What is Microsoft package trying to update an Ubuntu system

superm1, 2022-12-16:
You can read some background on this here: 
https://eclypsium.com/2020/07/29/theres-a-hole-in-the-boot/
https://eclypsium.com/2021/04/14/boothole-how-it-started-how-its-going/

hughsie, 2023-02-19:
I think this is good now.

superm1, 2023-02-19:
Well once we release the first confined snap and snapd does the next release that is. So soon (TM).

superm1, 2023-05-10:
This particular issue is fixed by the snap in the candidate channel which has switched to confined.  Once we promote it to stable channel I think we can close this issue.

superm1, 2023-07-12:
the confined 1.9.3 snap has been released to stable channel.

closed 2023-07-12 by superm1