fwupd/fwupd #5123: fwupdmgr security: Encrypted swapfile reported as `Linux swap: invalid` (btrfs subvolume on luks)

VannTen, 2022-10-05:
**Describe the bug**
`fwupdmgr security` reports `Linux swap: Invalid` when using a swapfile located on a btrfs subvolume, where the btrfs FS is on a LUKS device.

**Steps to Reproduce**
My setup is maybe a bit particular:
I don't have an explicit fstab, except for the swapfile itself: 
```swapfile
# Static information about the filesystems.
# See fstab(5) for details.

# <file system> <dir> <type> <options> <dump> <pass>
/swapfiles/swap0 none swap defaults
```
```console
$ ls -l /swapfiles/
total 20971520
-rw------- 1 root root 21474836480 Jun 23 16:36 swap0
```
The EFI and the root partition are mapped thanks to systemd-gpt-auto-generator:
```
Disk /dev/nvme0n1: 476.94 GiB, 512110190592 bytes, 1000215216 sectors
Disk model: SAMSUNG MZVL2512HCJQ-00BL7              
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes
Disklabel type: gpt
Disk identifier: 21411084-1CE8-4BB1-8778-7B930FD14E93

Device          Start        End   Sectors   Size Type
/dev/nvme0n1p1   2048     411647    409600   200M EFI System
/dev/nvme0n1p2 411648 1000214527 999802880 476.7G Linux root (x86-64)
```
/dev/nvme0n1p2 is an encrypted LUKS container mapped to /dev/mapper/root :
```
$ mount
/dev/mapper/root on / type btrfs (rw,relatime,ssd,space_cache=v2,subvolid=256,subvol=/root)

$ sudo btrfs subvolume list / 
ID 256 gen 64862 top level 5 path root
ID 258 gen 63875 top level 256 path home
ID 260 gen 20 top level 5 path snapshots
ID 261 gen 20 top level 260 path snapshots/root
ID 262 gen 20 top level 260 path snapshots/home
ID 263 gen 550 top level 256 path var/lib/portables
ID 264 gen 550 top level 256 path var/lib/machines
ID 265 gen 64864 top level 258 path home/max
ID 266 gen 64816 top level 256 path swapfiles
```

**Expected behavior**
`fwupdmgr security` figures out the swapfiles is on an encrypted support.

**fwupd version information**
```
runtime   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.0
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.13-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.gusb          0.4.0
```
(installed from pacman)

<details>

<summary>**fwupd device information**</summary>
Please provide the output of the fwupd devices recognized in your system.
[... 440 lines cut ...]
│     Device ID:          7eb852e1a38178cb2c78b0de175c24b8c4e4b227
│     Summary:            Unmatched performance for high-speed I/O
│     GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
│     Device Flags:       • Internal device
│                         • System requires external power source
│                         • Signed Payload
│   
└─USB4 host controller:
      Device ID:          b7a693354a37ab2d3a77afd486e99f3977d2c8ec
      Summary:            Unmatched performance for high-speed I/O
      GUID:               e72e778e-94f7-5ed2-b560-1c1262ee217c ← TBT-fixed
      Device Flags:       • Internal device
                          • System requires external power source
                          • Signed Payload
```

</details>

**Additional questions**

- Operating system and version: Archlinux. `uname -a`: `Linux work-laptop-max 5.19.13-arch1-1 #1 SMP PREEMPT_DYNAMIC Tue, 04 Oct 2022 14:36:58 +0000 x86_64 GNU/Linux`
- Have you tried rebooting? Yes
- Is this a regression? No

As far as I can tell they are some similar reports ( #4969 ) but not about the swapfile precisely.

superm1, 2022-10-05:
I think conceptually it's a similar issue as #4969.  We don't really look at nested volumes to determine that what is used for swap is encrypted.  We probably need some mechanism to look at `/proc/swaps` and unroll each layer of where something is stored to make that determination

superm1, 2022-10-05:
As an unrelated side note - would you mind submitting a PR to our bug templates?  I really like how `<summary>` and `<details>` work for hiding the device information unless it's really needed to pull it apart

VannTen, 2023-01-03:
I investigated a bit on the why this didn't work:

Fwupd use the 'device number' reported by `stat` on the swapfile and try to match that with the device exposed by udisk, as far as I can tell.

The thing is, btrfs subvolume have their own device number, which is, as far as I understand, not mapped to actual device (encrypted or otherwise).

So far, I didn't find a way to map a device number taken from a file on a btrfs filesystem to a device and find out whether it's encrypted.

Not sure if that's the same story for LVM setups like in #4969.

@superm1 What would you think of a PR using mountpoints instead of device number to map a file to a device ?
Aka, walk `/proc/mounts` and use the most specific mountpoint.

superm1, 2023-01-03:
Thanks for looking into it.  The problem you end up with `/proc/mounts` is nested mounts.  Any idea how `df -a` does it?
Maybe we can use that as a fallback path when we get this result from udisks.

VannTen, 2023-01-03:
By nested mounts, you mean something like :
```
/dev/stuff on /mount/point
/dev/staff on /mount/point/specific
```
?
Would'nt that be solved by while walking the mouts list, remembering last match and have a "more specific" comparison ?

in that case for the file `/mount/point/specific/swapfile` that would go something like (very pseudo coding)

```
current_mount_point_match = NULL
for mountpoint in /proc/mounts
   if $mounpoint contains $file and len($mountpoint) > len($current_mount_point_match)
   current_mount_point_match = $mountpoint
```
Might be missing something obvious this is a brain dump.

superm1, 2023-01-03:
> Would'nt that be solved by while walking the mounts list, remembering last match and have a "more specific" comparison ?

Yeah, that's exactly what I mean. 

> in that case for the file /mount/point/specific/swapfile that would go something like (very pseudo coding)

Yeah I think that would work.  I was just suggesting to see if `df -a` had another way to do it that you can leverage.

hughsie, 2024-11-11:
I think someone needs to implement this, it's not something on my radar -- sorry. Please re-open if needed.

closed 2024-11-11 by hughsie

Thaodan, 2024-11-11:
Would it make sense to mark the issue as help wanted and keep it open?
This way potential contributors could see the issue and help.

hughsie, 2024-11-11:
>  Would it make sense to mark the issue as help wanted and keep it open?

Ohh I'm very happy to reopen if such a person comes along. What we can't have is ~140 open issues (I triaged them **all** today!) as then it's just not manageable for us dev. I think it's 100% fine to say "we don't support that system" given our limited resources -- but I'd be happy to review a pull request obviously.

diamondburned, 2024-11-11:
I think it would be better to close this as Not planned rather than Completed.

Thaodan, 2024-11-11:
> &gt;  Would it make sense to mark the issue as help wanted and keep it open?
>
> Ohh I&#39;m very happy to reopen if such a person comes along. What we
> can&#39;t have is ~140 open issues (I triaged them **all** today!) as
> then it&#39;s just not manageable for us dev. I think it&#39;s 100%
> fine to say &quot;we don&#39;t support that system&quot; given our
> limited resources -- but I&#39;d be happy to review a pull request
> obviously.

I understand. Thanks for your work in any case fwupd has been fantastic.