fwupd/fwupd #5138: Flatpak app prompts error message

Overcraft90, 2022-10-08:
**Describe the bug**
When opened for the first time after booting into the system the `Flatpack Software App` shows an error message related to _**fwupd**_.

**Steps to Reproduce**
Log into the account, press on the app icon Software (Flatpack).

**Expected behavior**
A clear screen which allows me to navigate the app and search for system updates.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
/snap/fwupd/4397/bin/fwupdmgr: error while loading shared libraries: libfwupdplugin.so: cannot open shared object file: No such file or directory
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):
I installed _**fwupd**_ on my machine using the following link: https://flathub.org/apps/details/org.gnome.Firmware. 

However, today for some reason it stopped working properly. Noted, the app has been installed long ago; today, it simply happened to have this malfunction.
<details>

<summary>**fwupd device information**</summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
/snap/fwupd/4397/bin/fwupdmgr: error while loading shared libraries: libfwupdplugin.so: cannot open shared object file: No such file or directory
```

</details>

**Additional questions**

- Operating system and version: UBUNTU 22.04 "Jammy Jellyfish"
- Have you tried rebooting? Yes
- Is this a regression? No

See attachment
[image]

superm1, 2022-10-08:
This is actually a bug with the snap it seems to me not the flatpak.

Overcraft90, 2022-10-08:
@superm1 Thanks for the clarification, thing is how can I address it?

superm1, 2022-10-08:
I think we need to dig into it to see what went wrong with the snap build. If you don't have a need for the snap I suggest you uninstall it and use the distro package instead.

Overcraft90, 2022-10-08:
Thanks again, how can I do so? In fact if I remove the Flatpak app the one installed in Snap gets deleted too and vice-versa..., or at least that is what happened during my 'experimentation' in order to fix this issue. 
Let me know, thank you!

Overcraft90, 2022-10-08:
@superm1 Never mind I did it! Just wonder when the actual app from your side will be ready. It is great to have it and very convenient.

superm1, 2022-10-09:
>  0x000000000000001d (RUNPATH)            Library runpath: [/lib/x86_64-linux-gnu/fwupd-1.8.7]

It looks like the rpath in the snap is not correct..   I'll have to read up on what we can do about that.

superm1, 2022-10-09:
I'm guessing what we need to do is patch the snap build process to update the RUNPATH  to a relative one based on `$ORIGIN`.  Something like `$ORIGIN/../lib`

Overcraft90, 2022-10-09:
Hi again @superm1,

Actually after going back to the distro package the command line was working perfectly. Then I though why not trying to get the software app for Flatpak.

This time around it works, it might have been that I was missing the distro? Has it always have to be present for any other higher level `fwupd` software to work? Well, anyway I think we can close for now.

Thanks again!

superm1, 2022-10-09:
I think there's a real problem with the snap still, let's keep this open to track that.

clauded, 2022-10-09:
I have the same problem with the snap on Solus. Is there a way to patch manually the snap to test ?

> I'm guessing what we need to do is patch the snap build process to update the RUNPATH to a relative one based on `$ORIGIN`. Something like `$ORIGIN/../lib/x86_64-linux-gnu/fwupd-1.8.7`

I have the same problem with the snap on Solus. Is there a way to patch manually the snap to test this workaround?

superm1, 2022-10-09:
@clauded you can have a try with my PR idea if you want.

clauded, 2022-10-10:
@superm1 unfortunately I don't have the tools to build the snap and patching the current snap don't work.

superm1, 2022-10-10:
OK.  Well once that merges we'll have a new snap built.

clauded, 2022-10-10:
The new snap still fails on Solus:
`/snap/fwupd/4399/bin/fwupdmgr: error while loading shared libraries: libffi.so.8: cannot open shared object file: No such file or directory`
Also, this bug is not resolved on Solus: https://github.com/fwupd/fwupd/issues/4195

PhilippWendler, 2022-10-11:
With the snap in version 1.8.6-13-gbbf62261d on Ubuntu 22.04 I get:
```
$ fwupd.fwupdtool 
/snap/fwupd/4399/bin/fwupdtool: error while loading shared libraries: libprotobuf-c.so.1: cannot open shared object file: No such file or directory
```
I also cannot start the `fwupd` daemon because it fails with the same error message and thus `fwupdmgr` times out.

Just wanted to note this here since the referenced library is a different one than the one reported by clauded.

superm1, 2022-10-11:
It seems that several of these libraries were already installed on my local environment is probably why I didn't hit it.

I think they probably need to be added to stage packages phase.

commit 7b6c3250 (2022-10-09): trivial: snap: add a workaround to the build process to ignore rpath

snapcraft overrides the rpath for the snap to make it most compatible.
Our rpath decisions make this not work properly though.  Since everything
is in /snap/fwupd anyway, just move the libraries to this place.

Fixes: #5138

Link: https://snapcraft.io/blog/the-new-classic-confinement-in-snaps-even-the-classics-need-a-change
commit b6199679 (2022-10-10): trivial: snap: add a workaround to the build process to ignore rpath

snapcraft overrides the rpath for the snap to make it most compatible.
Our rpath decisions make this not work properly though.  Since everything
is in /snap/fwupd anyway, just move the libraries to this place.

Fixes: #5138

Link: https://snapcraft.io/blog/the-new-classic-confinement-in-snaps-even-the-classics-need-a-change
PR #5142 (2022-10-10): trivial: snap: add a workaround to the build process to ignore rpath
snapcraft overrides the rpath for the snap to make it most compatible. Our rpath decisions make this not work properly though.  Since everything is in /snap/fwupd anyway, just move the libraries to this place.

Fixes: #5138

Link: https://snapcraft.io/blog/the-new-classic-confinement-in-snaps-even-the-classics-need-a-change

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 9475a849 (2022-10-10): trivial: snap: add a workaround to the build process to ignore rpath

snapcraft overrides the rpath for the snap to make it most compatible.
Our rpath decisions make this not work properly though.  Since everything
is in /snap/fwupd anyway, just move the libraries to this place.

Fixes: #5138

Link: https://snapcraft.io/blog/the-new-classic-confinement-in-snaps-even-the-classics-need-a-change
closed 2022-10-11 by superm1