fwupd/fwupd #5139: Secure boot is enabled, but shim isn't installed to /run/media/root/.../EFI/arch/shimx64.efi

WhyNotHugo, 2022-10-08:
**Describe the bug**

`fwupdmgr install` (and `fwupdmgr install --force`) refuses to install because `shim` is not installed.

However, `fwupdmgr` itself is signed. The signed copy is in `/usr/lib/fwupd/efi/fwupdx64.efi.signed`, and fwupdmgr actually handles this signed binary properly. This should make it very clear that `shim` is not required in this context because the binary is signed with the SecureBoot key.

It would make sense the most for `fwupdmgr` to _not_ require shim. It is substantially less secure, and only really required on setups that dual-boot into windows, but not on regular setups. Additionally, in this particular scenario there's also a signed version of the fwupdmgr binary, so there's clear indication that it's safe to proceed.

**Steps to Reproduce**

1. Sign `fwupdx64.efi`.
2. `fwupdmgr install`

**Expected behavior**

Firmware upgrade should proceed correctly.

**fwupd version information**
Please provide the version of the daemon and client.

```shell
> fwupdmgr --version
runtime   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.fwupd-efi     1.3
compile   org.freedesktop.gusb          0.4.0
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    6.0.0-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.5
runtime   org.freedesktop.gusb          0.4.0
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc):

pacman

**System UEFI configuration**
Please provide the output of the following commands:

```shell
> efibootmgr -v

BootCurrent: 0000
Timeout: 0 seconds
BootOrder: 0000,0001
Boot0000* Linux Boot Manager	HD(1,GPT,b186c4af-01d5-b344-b278-af2b7117cd89,0x800,0x96000)/File(\EFI\systemd\systemd-bootx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 60 09 00 00 00 00 00 af c4 86 b1 d5 01 44 b3 b2 78 af 2b 71 17 cd 89 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 5c 00 73 00 79 00 73 00 74 00 65 00 6d 00 64 00 2d 00 62 00 6f 00 6f 00 74 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0001* Linux Firmware Updater	HD(1,GPT,b186c4af-01d5-b344-b278-af2b7117cd89,0x800,0x96000)/File(\EFI\arch\fwupdx64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 60 09 00 00 00 00 00 af c4 86 b1 d5 01 44 b3 b2 78 af 2b 71 17 cd 89 02 02 / 04 04 32 00 5c 00 45 00 46 00 49 00 5c 00 61 00 72 00 63 00 68 00 5c 00 66 00 77 00 75 00 70 00 64 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
```

```shell
> efivar -l | grep fw
0abba7dc-e516-4167-bbf5-4d9d1c739416-fwupd-7ceaf7a8-0611-4480-9e30-64d8de420c7c-0

```

**Additional questions**

- Operating system and version: ArchLinux
- Have you tried rebooting? Yes
- Is this a regression? I believe this worked a long time ago (over a year ago?), but might be mistaken.
- Are you using an NVMe disk? Yes
- Is secure boot enabled? Yes
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? n/a

superm1, 2022-10-10:
What you're looking for is the key `DisableShimForSecureBoot=true` in `/etc/fwupd/uefi_capsule.conf`

WhyNotHugo, 2022-10-10:
I came across that key; the issue is more about the default behaviour (having rather unusual expectations which are not very secure ones).

It's also quite hard to find this configuration key; I wonder if an interactive confirmation or a more detailed message would help, even for the current default setting.

superm1, 2022-10-10:
So the problem is what authority is enrolled into the secure boot db might not match the key they fwupd is signed with.

For the majority of the boxes running fwupd the authority is the Microsoft 3rd party CA. This is because it's extra work to remove that, create and enroll your own.

So to make this work "out of the box" on most setups this is the line in the sand.