fwupd/fwupd #5152: TPM PCR0 reconstruction is showing "Not found" even through tpm2_pcrread shows all PCR's including PCR0.

Arisa-Snowbell, 2022-10-11:
**Describe the bug**
TPM PCR0 reconstruction is showing "Not found" even through tpm2_pcrread shows all PCR's including PCR0.
Couldn't find fwupdtpmevlog binary to get the output
File /sys/kernel/security/tpm0/binary_bios_measurements doesn't exist.

**Expected behavior**
To find the PCR0

**fwupd version information**

```shell
runtime   org.freedesktop.fwupd         1.8.6
compile   org.freedesktop.gusb          0.4.0
runtime   com.dell.libsmbios            2.4
runtime   org.kernel                    5.19.13-arch1-1
runtime   com.hughsie.libjcat           0.1.11
compile   com.hughsie.libjcat           0.1.11
compile   org.freedesktop.fwupd         1.8.6
runtime   org.freedesktop.gusb          0.4.1
```

Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): pacman

**Additional questions**

- Operating system and version: Arch Linux as of 12.10.2022
- Have you tried rebooting? Yes
- Is this a regression? Unknown

hughsie, 2022-10-12:
We need /sys/kernel/security/tpm0/binary_bios_measurements if we're going to test the reconstitution. It's the source of what values get pushed into the TPM.

Arisa-Snowbell, 2022-10-12:
I have no idea why it doesn't exist, is there any way I can fix it or?

superm1, 2022-10-12:
Most likely your kernel configuration has a problem, did you compile without TPM support, or without securityfs support perhaps?

Arisa-Snowbell, 2022-10-12:
I am using stock Arch kernel, compiled by the Arch Team

superm1, 2022-10-12:
> I am using stock Arch kernel, compiled by the Arch Team

I suggest reviewing with them then to make sure the kernel configuration is appropriate.  There is nothing fwupd can do about this sysfs file missing from the kernel.

closed 2022-10-12 by superm1

Arisa-Snowbell, 2022-10-12:
I looked in the config how the kernel was compiled and found CONFIG_SECURITYFS=y and CONFIG_TCG_TPM=m (all other tpm options are y) so I tried to load it with modprobe tpm and nothing, the securityfs is already mounted on /sys/kernel/security but there is no folder tpm0.
I will try to add it to initramfs but i am not sure if i have correct name for the module, could anyone please confirm?

```output of lsmod | rg tpm at cold boot
tpm_crb                20480  0
tpm_tis                16384  0
tpm_tis_core           32768  1 tpm_tis
tpm                   102400  4 tpm_tis,trusted,tpm_crb,tpm_tis_core
rng_core               20480  2 ccp,tpm
```

```ls /lib/modules/5.19.13-arch1-1/kernel/drivers/char/tpm
st33zp24
tpm.ko.zst
tpm_atmel.ko.zst
tpm_crb.ko.zst
tpm_i2c_atmel.ko.zst
tpm_i2c_infineon.ko.zst
tpm_i2c_nuvoton.ko.zst
tpm_infineon.ko.zst
tpm_nsc.ko.zst
tpm_tis.ko.zst
tpm_tis_core.ko.zst
tpm_tis_i2c_cr50.ko.zst
tpm_tis_spi.ko.zst
tpm_vtpm_proxy.ko.zst
xen-tpmfront.ko.zst
```

jasongodev, 2025-11-18:
What is your boot loader and manager? You might be using a boot loader/manager that does not measure TPM PCR. A boot manager such as Grub and systemd-boot will do a Measured Boot that will be logged in /sys/kernel/security/tpm0/binary_bios_measurements. Similarly, a UKI that uses systemd-stub will also do a measured boot.

See my discussion and my workaround here: https://github.com/fwupd/fwupd/discussions/9568