fwupd/fwupd #5162: ALLOW_AUTHENTICATION feature flag to circumvent Polkit authentication

mgerstner, 2022-10-17:
**Describe the question**
In recent 1.8 fwupd releases a new logic in the authentication of D-Bus requests has been introduced via commit 2536bf462c3fb0cb5bd174bff5c799613ced0b16. The feature flag `FWUPD_FEATURE_FLAG_ALLOW_AUTHENTICATION` is controlled by *clients*, and allows them to simply switch off Polkit authentication for the get-bios-settings action.

This is currently the only action that evaluates this feature flag. Thus no real security issue should entail from this. Still the logic feels very strange and if this logic spreads in the future to other D-Bus methods there's a danger that people will think there is still some kind of security behind all of this, while actually there isn't.

In the current state the `get-bios-settings` Polkit action is pointless, since any client can simply circumvent it by not setting that feature flag. I would prefer this logic being dropped along with the Polkit action and simply never authenticate this D-Bus call instead of having this complexity that is confusing but doesn't really add any value with regards to security.

I stumbled over this via a [routine review of new Polkit actions in fwupd](https://bugzilla.suse.com/show_bug.cgi?id=1204026) for addition openSUSE.

You can reproduce this by calling this:

```
gdbus call -y -d org.freedesktop.fwupd -o / -m org.freedesktop.fwupd.GetBiosSettings
```

It will not request any authentication, because the client didn't set the feature flag `ALLOW_AUTHENTICATION`.

hughsie, 2022-10-17:
But the results should be different depending if you choose to authenticate or not. If you authenticate you get the current BIOS values, if you do it without auth then you don't.

superm1, 2022-10-17:
As Richard points out this behavior is different if you request authentication or not. To add to this it mirrors the kernel behavior that you can't access certain information without root.

The reason for this difference is to enable bash completion.

mgerstner, 2022-10-17:
I cannot really test this during runtime in my VM, since there I don't get any bios values whatsoever.

But from the code I cannot really make out the difference. Without the flag it returns [these attrs](https://github.com/fwupd/fwupd/blob/main/src/fu-daemon.c#L1997):

```
			g_autoptr(FuBiosSettings) attrs =
			    fu_context_get_bios_settings(fu_engine_get_context(self->engine));
			val = fu_bios_settings_to_variant(
			    attrs,
			    fu_engine_request_get_device_flags(request) &
				FWUPD_DEVICE_FLAG_TRUSTED);
			g_dbus_method_invocation_return_value(invocation, val);
```

With the flag and after successful authentication it returns [these attrs](https://github.com/fwupd/fwupd/blob/main/src/fu-daemon.c#L530):

```
	ctx = fu_engine_get_context(helper->self->engine);
	attrs = fu_context_get_bios_settings(ctx);
	val = fu_bios_settings_to_variant(attrs, TRUE);
	g_dbus_method_invocation_return_value(helper->invocation, val);
```

hughsie, 2022-10-17:
fu_bios_settings_to_variant() returns different results depending on the `trusted` boolean.

mgerstner, 2022-10-17:
Ah I see. That's the (literally) bit I missed. Are you planning on extending the use of this `ALLOW_AUTHENTICATION` differentiation? Some more comments in the code would have been helpful to better understand the intention of this. Thanks for explaining!

closed 2022-10-17 by mgerstner

superm1, 2022-10-17:
> Ah I see. That's the (literally) bit I missed. Are you planning on extending the use of this `ALLOW_AUTHENTICATION` differentiation? 
> Some more comments in the code would have been helpful to better understand the intention of this. Thanks for explaining!

Would you mind submitting a PR with the comments where you think they make sense?