fwupd/fwupd #5169: Feature request: Warning before EFI firmware updates and/or backup functionality for EFI Boot entries and EFI Secure boot certs

das-menschy, 2022-10-18:
**Is your feature request related to a problem? Please describe.**
As I described in fwupd/firmware-lenovo/issues/276 , the Lenovo System Firmware update deleted all UEFI Secure Boot certs and UEFI Boot entries. 

**Describe the solution you'd like**
It would be nice if fwupdmgr could 
* print a warning before running the firmware update urging the user to make a **backup** of all relevant UEFI Secure Boot certs and UEFI Boot entries.  
* implement a kind of **backup** functionality which makes backing up and reloading all these UEFI Secure Boot certs and UEFI Boot entries easier. 

**Describe alternatives you've considered**
At the moment, I have saved all the Secure Boot certs on the EFI system partition: 
```shell
[root@linux-on-20hr002mmx ~]$ tree /efi/EFI/keys/
/efi/EFI/keys/
├── DB.efitools.auth
├── DBX.efitools.auth
├── KEK.efitools.auth
├── MicWinProPCA2011_2011-10-19.efitools.auth
└── PK.efitools.auth
```
And I also have KeyTool.efi from efitools set up as boot entry in systemd-boot: 
```
[root@linux-on-20hr002mmx ~]$ bootctl list 
Boot Loader Entries:
         type: Boot Loader Specification Type #1 (.conf)
        title: KeyTool-signed.efi
           id: KeyTool.conf
       source: /efi/loader/entries/KeyTool.conf
...
```
... so I can reload all necessary EFI Secure Boot certs immediately after a System firmware update; because without the UEFI Boot entries from systemd (Loader-...), KeyTool will start by default (and not the Arch Linux boot entries). These UEFI Secure Boot *.auth files could be manipulated however; so it is not a safe solution. 

Additionally, I have to run ``sudo bootctl install`` after the firmware update, to make the Linux systemd-boot boot entry the default one again (and not the Windows boot entry).

hughsie, 2022-10-18:
I think we're better off just checking this never happens, i.e. https://gitlab.com/fwupd/lvfs-website/-/issues/841

das-menschy, 2022-10-18:
Sorry, https://gitlab.com/fwupd/lvfs-website/-/issues/845 is a dead link for me (404 not found). Is that the private issue with the security vulnerability (secalert) you were talking about which can be made invisible until full disclosure? 
[image]

hughsie, 2022-10-19:
Doh, it is marked confidential, sorry. I'll work on this a bit today and hopefully have something we can share.

hughsie, 2022-10-20:
https://github.com/fwupd/fwupd/pull/5178 should get us most of the way there

das-menschy, 2022-10-20:
Wow, thank you! That's cool.

closed 2023-09-23 by hughsie