fwupd/fwupd #5171: latest fwupd-1.8.6  installation failed on Ubuntu20.04

dukexinaw, 2022-10-19:
Background ：
       Fwupd - 1.8.5 cannot upgrade quectel EM1X0 PCIE MBB equipment, we learned that filed the latest modified Ivanmikh point (https://github.com/fwupd/fwupd/pull/5103), And released on FWUPD-1.8.6, I'm ready to test this latest fwupd

Issue:  latest fwupd-1.8.6  installation failed on Ubuntu20.04

duke@duke:~/Desktop/install_lvfs/fwupd$  meson build --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu -Dplugin_modem_manager=enabled
......
Header "linux/gpio.h" has symbol "GPIO_V2_LINE_FLAG_OUTPUT" : NO 

plugins/lenovo-thinklmi/meson.build:23:2: ERROR: Unknown variable "plugin_builtin_uefi_capsule".

hughsie, 2022-10-19:
Are you compiling without efivar? Can you share the complete meson output please.

dukexinaw, 2022-10-19:
Are you compiling without efivar? 
>> I'm not sure, fail log No efivar dependency message, but I have fwupd-18.5 pass installed on my local environment

Can you share the complete meson output please.
>> 
Linux duke 5.13.0-30-generic #33~20.04.1-Ubuntu SMP Mon Feb 7 14:25:10 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux

duke@duke:~/Desktop/install_lvfs/fwupd$ mmcli  --version
mmcli 1.19.1
Copyright (2011 - 2021) Aleksander Morgado
License GPLv2+: GNU GPL version 2 or later <http://gnu.org/licenses/gpl-2.0.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.

duke@duke:~/Desktop/install_lvfs/fwupd$  meson build --prefix=/usr --libdir=/usr/lib/x86_64-linux-gnu -Dplugin_modem_manager=enabled
The Meson build system
Version: 0.63.3
Source dir: /home/duke/Desktop/install_lvfs/fwupd
Build dir: /home/duke/Desktop/install_lvfs/fwupd/build
Build type: native build
Project name: fwupd
Project version: 1.8.7
C compiler for the host machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the host machine: cc ld.bfd 2.34
Host machine cpu family: x86_64
Host machine cpu: x86_64
Program git found: YES (/usr/bin/git)
WARNING: You should add the boolean check kwarg to the run_command call.
         It currently defaults to false,
         but it will default to true in future releases of meson.
         See also: https://github.com/mesonbuild/meson/issues/9300
Compiler for C supports arguments -Waggregate-return: YES 
Compiler for C supports arguments -Wunused: YES 
Compiler for C supports arguments -Warray-bounds: YES 
Compiler for C supports arguments -Wcast-align: YES 
Compiler for C supports arguments -Wclobbered: YES 
Compiler for C supports arguments -Wdeclaration-after-statement: YES 
Compiler for C supports arguments -Wdiscarded-qualifiers: YES 
Compiler for C supports arguments -Wduplicated-branches: YES 
Compiler for C supports arguments -Wduplicated-cond: YES 
Compiler for C supports arguments -Wempty-body: YES 
Compiler for C supports arguments -Wformat=2: YES 
Compiler for C supports arguments -Wformat-nonliteral: YES 
Compiler for C supports arguments -Wformat-security: YES 
Compiler for C supports arguments -Wformat-signedness: YES 
Compiler for C supports arguments -Wignored-qualifiers: YES 
Compiler for C supports arguments -Wimplicit-function-declaration: YES 
Compiler for C supports arguments -Winit-self: YES 
Compiler for C supports arguments -Wlogical-op: YES 
Compiler for C supports arguments -Wmaybe-uninitialized: YES 
Compiler for C supports arguments -Wmissing-declarations: YES 
Compiler for C supports arguments -Wmissing-format-attribute: YES 
Compiler for C supports arguments -Wmissing-include-dirs: YES 
Compiler for C supports arguments -Wmissing-noreturn: YES 
Compiler for C supports arguments -Wmissing-parameter-type: YES 
Compiler for C supports arguments -Wmissing-prototypes: YES 
Compiler for C supports arguments -Wnested-externs: YES 
Compiler for C supports arguments -Wno-cast-function-type: YES 
Compiler for C supports arguments -Wno-address-of-packed-member: YES 
Compiler for C supports arguments -Wno-unknown-pragmas: YES 
Compiler for C supports arguments -Wno-missing-field-initializers: YES 
Compiler for C supports arguments -Wno-strict-aliasing: YES 
Compiler for C supports arguments -Wno-suggest-attribute=format: YES 
Compiler for C supports arguments -Wno-unknown-warning-option: NO 
Compiler for C supports arguments -Wno-unused-parameter: YES 
Compiler for C supports arguments -Wold-style-definition: YES 
Compiler for C supports arguments -Woverride-init: YES 
Compiler for C supports arguments -Wpointer-arith: YES 
Compiler for C supports arguments -Wredundant-decls: YES 
Compiler for C supports arguments -Wreturn-type: YES 
[... 438 lines cut ...]
Configuring fwupd.test using configuration
Configuring fwupdmgr-p2p.sh using configuration
Configuring fwupd-tests.conf using configuration
Configuring fwupd-offline-update.service using configuration
Configuring fwupd.service using configuration
Configuring fwupd.shutdown using configuration
Configuring org.freedesktop.fwupd.service using configuration
Configuring fu-version.h using configuration
Compiler for C supports link arguments -Wl,--version-script,/home/duke/Desktop/install_lvfs/fwupd/libfwupdplugin/fwupdplugin.map: YES 
Configuring simple_client.py using configuration
Configuring fwupd.spec.in using configuration
Compiler for C supports arguments -fcf-protection: YES 
Checking if "__attribute__((noinline,noclone))" compiles: YES 
Run-time dependency tss2-esys found: NO (tried pkgconfig and cmake)
Run-time dependency libsmbios_c found: YES 2.4.3
Configuring dell-esrt.conf using configuration
Header "linux/gpio.h" has symbol "GPIO_V2_LINE_FLAG_OUTPUT" : NO 

plugins/lenovo-thinklmi/meson.build:23:2: ERROR: Unknown variable "plugin_builtin_uefi_capsule".

A full log can be found at /home/duke/Desktop/install_lvfs/fwupd/build/meson-logs/meson-log.txt
duke@duke:~/Desktop/install_lvfs/fwupd$ 

 
 ....

hughsie, 2022-10-19:
I see:

    Run-time dependency efiboot found: NO (tried pkgconfig and cmake)
    Run-time dependency efivar found: NO (tried pkgconfig and cmake)

@superm1 i guess we should make the thinklmi tests depend on efiboot.found() and efivar.found() -- right?

dukexinaw, 2022-10-19:
@hughsie Unfortunately, as you suggested, I added efibootmgr and efivar(apt install it), do   [meson build .......] ，  The error is the same, and it still ：
Run-time dependency efiboot found: NO (tried pkgconfig and cmake)
Run-time dependency efivar found: NO (tried pkgconfig and cmake)

dukexinaw, 2022-10-19:
Tried to apply  [fix #5171]  patch,build and test ok.
efiboot and efivar are kernel fearture cannot be passed through apt install, right?

hughsie, 2022-10-19:
If you run ./contrib/setup all the deps will install.

closed 2022-10-19 by hughsie