fwupd/fwupd #5173: Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 4705a9d0cacec299a079523d0fcd3ac7ef9b6221

Locutus64, 2022-10-19:
When I use Topgrade to do my system udates and it gets to the firmware update I get the below. Have tried a couple of things to fix it from another issue thread but nothing works. 

`Failed to update metadata for lvfs: checksum failure: failed to verify data, expected 4705a9d0cacec299a079523d0fcd3ac7ef9b6221`

On the lastest XeroLinux release (Arch based).

hughsie, 2022-10-19:
I get the same! debugging!

hughsie, 2022-10-19:
I get:

    wget https://fwupd.org/downloads/firmware.xml.gz.jcat
$ jcat-tool info firmware.xml.gz.jcat 
JcatFile:
  Version:               0.1
  JcatItem:
    ID:                  firmware-04761-stable.xml.gz
    AliasId:             firmware.xml.gz
    JcatBlob:
      Kind:              sha1
      Flags:             is-utf8
      Timestamp:         2022-10-19T12:47:27Z
      Size:              0x28
      Data:              c65dbc3fe6ef8b8f559d0fe6e00c3fad3f24645e
    JcatBlob:
      Kind:              sha256
      Flags:             is-utf8
      Timestamp:         2022-10-19T12:47:27Z
      Size:              0x40
      Data:              2072aec885473ed1afeb5f425b526b5a1b517954f363ee580784639065ad01d1
    JcatBlob:
      Kind:              gpg
      Flags:             is-utf8
      Timestamp:         2022-10-19T12:47:27Z
      Size:              0x1e8
      Data:              -----BEGIN PGP SIGNATURE-----
                         
                         iQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAmNP8d8ACgkQSKbYDkU4
                         coGq/R4JAHJZLTceBMOe/MVLw+caPQ==
                         =Nx1T
                         -----END PGP SIGNATURE-----
                         
    JcatBlob:
      Kind:              pkcs7
      Flags:             is-utf8
      Timestamp:         2022-10-19T12:47:27Z
      Size:              0x8c0
      Data:              -----BEGIN PKCS7-----
                         MIIGUgYJKoZIhvcNAQcCoIIGQzCCBj8CAQExDTALBglghkgBZQMEAgEwCwYJKoZI
                         Xzu1Rck6UyMwwS6EKs/jBnGXe/oQwMRbn1RP+jycJhmEnhpgAZ0=
                         -----END PKCS7-----
    $ wget https://fwupd.org/downloads/firmware-04761-stable.xml.gz
    sha256sum firmware-04761-stable.xml.gz
    2072aec885473ed1afeb5f425b526b5a1b517954f363ee580784639065ad01d1  firmware-04761-stable.xml.gz

..which is correct...

Locutus64, 2022-10-19:
```
░▒▓ ~  wget https://fwupd.org/downloads/firmware.xml.gz.jcat
--2022-10-19 05:58:34--  https://fwupd.org/downloads/firmware.xml.gz.jcat
Loaded CA certificate '/etc/ssl/certs/ca-certificates.crt'
Resolving fwupd.org (fwupd.org)... 35.155.169.38, 35.167.130.158
Connecting to fwupd.org (fwupd.org)|35.155.169.38|:443... connected.
HTTP request sent, awaiting response... 416 REQUESTED RANGE NOT SATISFIABLE

    The file is already fully retrieved; nothing to do.

░▒▓ ~  jcat-tool info firmware.xml.gz.jcat
JcatFile:
  Version:               0.1
  JcatItem:
    ID:                  firmware-04760-stable.xml.gz
    AliasId:             firmware.xml.gz
    JcatBlob:
      Kind:              sha1
      Flags:             is-utf8
      Timestamp:         2022-10-19T11:37:14Z
      Size:              0x28
      Data:              da258a154df4cea2a47714b1b65aa52cac896524
    JcatBlob:
      Kind:              sha256
      Flags:             is-utf8
      Timestamp:         2022-10-19T11:37:14Z
      Size:              0x40
      Data:              5bcc9b47c3f9ff56c47b9d8d5c409110bf4611a458cd2422d55f2d0b6618f5b8
    JcatBlob:
      Kind:              gpg
      Flags:             is-utf8
      Timestamp:         2022-10-19T11:37:15Z
      Size:              0x1e8
      Data:              -----BEGIN PGP SIGNATURE-----
                         
                         iQEzBAABCAAdFiEEP8a4BEEO0IQNjy+XSKbYDkU4usIFAmNP4WsACgkQSKbYDkU4
                         usLf7AgAj5U9/nHngF4h7F7osydvA4wIpnvy1p80NXe3ZYx4gLd+M6oWwBxtH2A2
                         M17/+SRefEoB9jFySG0azmGlHhzAfRb53xMfCGZiMKZWsZG0dsH1SM/MS5uWSDwv
                         R4+ZEXxrQ1uWBHY7dQ4RwTT9HU7Ea82Bf0MVQ4YrPrl4FFa+h8j/EC1F2BvC5KZ9
                         mePk/P5yVt5+p9p1wYNfhHOF5kkCh3qcN3B3uDPzMCjbXk98KUoOrbRhyQMVBo06
                         BJtSPy++rbpL/Ljl0ynen4sVAbsV7YDGZLphfGb2r1LwZvDLXv8PFt0G+Rs6LdwC
                         ITfnpaH9tlxmQD5AcZqW6Zso6AnVwg==
                         =yivx
                         -----END PGP SIGNATURE-----
                         
    JcatBlob:
      Kind:              pkcs7
      Flags:             is-utf8
      Timestamp:         2022-10-19T11:37:15Z
      Size:              0x8c0
      Data:              -----BEGIN PKCS7-----
                         MIIGUgYJKoZIhvcNAQcCoIIGQzCCBj8CAQExDTALBglghkgBZQMEAgEwCwYJKoZI
                         hvcNAQcBoIIEOjCCBDYwggKeoAMCAQICDFprhisibP88kP07YjANBgkqhkiG9w0B
                         AQsFADA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3Ig
                         RmlybXdhcmUgUHJvamVjdDAeFw0xODAxMTYwMDAwMDBaFw0yODAxMTYwMDAwMDBa
                         MBkxFzAVBgNVBAMTDlJpY2hhcmQgSHVnaGVzMIIBIjANBgkqhkiG9w0BAQEFAAOC
                         AQ8AMIIBCgKCAQEArR5nseG3+Zs+o41P9LTspiVSGVcp6ifNHhbKKxBvZZXsy0gX
                         +P/VRlsHLiKQrFulQ8GbPODytFv0o+y/0MkiJxv/fY3yEZ2bwNpsSeXFQSGHI6yz
                         jaVNeNCu8lnnDtD7kiC8UNUNHcnA4+2h/Yv4k+KPqYF+Qb6nWAEIID1ObMnjeJUb
                         dbPPvy12aasV3gZcZ+goYNFc0ua6OU/CNEuzAVVCTAJ/EpCdGpll1+6BGU5ImIUG
                         TlMTWq2xmCfCPugakHrmWA66yHWwE6LmC/U7qQDWFemsSNnmzyBB9HPkqsW1DjHr
                         +ZmNUPj3+q2UGnNwP/Cne462XbsZB569w7pnzQIDAQABo4HcMIHZMAwGA1UdEwEB
                         /wQCMAAwNwYDVR0RBDAwLoYXaHR0cHM6Ly9md3VwZC5vcmcvbHZmcy+BE3JpY2hh
                         cmRAaHVnaHNpZS5jb20wEwYDVR0lBAwwCgYIKwYBBQUHAwMwDwYDVR0PAQH/BAUD
                         AweAADAdBgNVHQ4EFgQUMcrnDWk3DM1GJCqJK7EbAycgB+UwHwYDVR0jBBgwFoAU
                         sY3q5COnfgmOte4x4GrdnjQ3ZawwKgYDVR0fBCMwITAfoB2gG4YZaHR0cDovL3d3
                         dy5md3VwZC5vcmcvcGtpLzANBgkqhkiG9w0BAQsFAAOCAYEAjoXBOtVb1qPLuE7I
                         ShdSkk66JMNmzZnODbun3BaViUF3PPhuRiJ2y7Bu7loCnxHwzKpq5Hn4Untg1WhQ
                         SsPEv6H2oc2E4HU1Gds2HEE1UL0VzulPEwaOFX0OZ37LmJ2VFvfgwfNmjAWZ+ngN
                         sjuff3PWeHNsOwbS3To0CBlHqz9LlcA7Gxpxbz4DbUEvgTTKZFbasgUS6/QmbpPy
                         880ThgD93yCg6Q002AooPMw8zW6gGUP7/2D4aKBjgM/IJDrJEicTdKDFtUacsN+d
                         CfIEcn6F1rzPgwo0B/yLYVMpbL8WzxSS7cX5dBlkreH6Nf7g+A2NTryU/6FxoFxo
                         nWJmLssNQCgx42Ywc8CgJRt1yWXxoZXOPap8zPvFAAmrfUlqfZlr2Wzcf2nxpSZJ
                         oQRzGDXDRiYNt28xjX408cJ+Wy+0oWwgnXCkfzMDnJQB8MX6+5Ah4v4mS/XjmOIi
                         Q+hKCT99Yf0Eq3TWzFhyYducqLU904FbrrUSoi07DfxXa/8RMYIB3jCCAdoCAQEw
                         SjA6MRAwDgYDVQQDEwdMVkZTIENBMSYwJAYDVQQKEx1MaW51eCBWZW5kb3IgRmly
                         bXdhcmUgUHJvamVjdAIMWmuGKyJs/zyQ/TtiMAsGCWCGSAFlAwQCAaBpMBgGCSqG
                         SIb3DQEJAzELBgkqhkiG9w0BBwEwHAYJKoZIhvcNAQkFMQ8XDTIyMTAxOTExMzcx
                         NVowLwYJKoZIhvcNAQkEMSIEIFvMm0fD+f9WxHudjVxAkRC/RhGkWM0kItVfLQtm
                         GPW4MA0GCSqGSIb3DQEBAQUABIIBACAeeyP3tChgzBjXFfoHw7DQXXgjVDNbfSxU
                         khw2NVw2H8gFjnBYU9Mu+ETKViUopEj9GnGynjJ1yUFMa7WkDilHVNvMUrmYjDtE
                         yei8QoFUQ/RxV0CiwwbRwTKJAkLJRvedyQQ1A9cfoC0uLv21QezpCVYxvMJCBnFH
                         50ST6TeceXej8n24WRZs/yOXMZ6ZGU4vYg1F7yw/VRVhTixa9BejquC6d5I6Nz3L
                         9mOYZqyImA9y/U2arHlspqc+CWnyw9Bq8cGidPqhewHBIuiSeebEr6hv8cXsLS5F
                         HTYKHU+6zWYnr8C2c+CEDdsB4OTMzpOiauX8cSw7aiwuinRNlW8=
                         -----END PKCS7-----
```

hughsie, 2022-10-19:
And this works too:

	$ jcat-tool verify firmware.xml.gz.jcat --public-keys /etc/pki/fwupd
	firmware-04761-stable.xml.gz:
	    PASSED sha1: OK
	    PASSED sha256: OK
	    PASSED gpg: 3FC6B804410ED0840D8F2F9748A6D80E4538BAC2
	    PASSED pkcs7: O=Linux Vendor Firmware Project,CN=LVFS CA

hughsie, 2022-10-19:
The client is downloading the file fine, but it's being truncated to 0xFFFFF bytes in length when being passed to the daemon.

hughsie, 2022-10-19:
Okay, I've put a workaround into the LVFS that takes the metadata size down to 800K, so if you do "fwupdmgr refresh --force" it should work now. I'll work on the proper fix in a few mins.

closed 2022-10-19 by hughsie

Locutus64, 2022-10-19:
Thanks for the quick reply and fix, very much appreciated.