fwupd/fwupd #5176: Cannot update on Debian 11 wirh Secure Boot dbx 83 => 217

Phil995511, 2022-10-20:
Gnome software tells me that a Secure Boot dbx 83 => 217 update is available but the update is impossible to perform.

- Current version available (Debian Stable) fwupd v. 1.5.7-4

- Another version available (Debian Testing) v. 1.8.6-2+b1, but this version is impossible to use, due to broken/incomplete package dependencies.

- fwupdmgr --version :

client version:	1.5.7
compile-time dependency versions
	gusb:	0.3.5

- fwupdmgr get-devices --show-all-devices :

XPS 15 9570
│
├─GP107M [GeForce GTX 1050 Ti Mobile]:
│     Device ID:          ce4c74a5188d5b9cdb1e72ed32dad2d313c1c999
│     Version actuelle:   a1
│     Fournisseur:        NVIDIA Corporation (PCI:0x10DE)
│     GUIDs:              a29c3564-3d3b-5ce2-9616-4567cb1d9670
│                         2d6a79d7-4ac2-53fd-af2d-000e2dfbdc68
│                         dc1e6866-2188-547f-b175-35e07dd7a6af
│                         04232c57-d805-551d-a1de-1bba9350d71f
│     Device Flags:       • Périphérique interne
│                         • Cryptographic hash verification is available
│   
├─Intel(R) Core™ i7-8750H CPU @ 2.20GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Version actuelle:   0x000000f0
│     Fournisseur:        Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         809a0b93-8a12-5338-a571-ad5583acf896
│                         72ec2ff3-49ff-5ec2-bdbb-525badd47543
│     Device Flags:       • Périphérique interne
│   
├─Samsung SSD 970 EVO 2TB:
│     Device ID:          f2759da7fe8e0388c5f3601cb072f837b1070b03
│     Résumé:             NVM Express Solid State Drive
│     Version actuelle:   2B2QEXE7
│     Fournisseur:        Samsung Electronics Co Ltd (NVME:0x144D)
│     GUIDs:              0b4d773a-7ac3-58c1-a541-e22ef1cdfe02
│                         c9d531ea-ee7d-5562-8def-c64d0d144813
│                         6e54c992-d302-59ab-b454-2d26ddd63e6d
│                         47335265-a509-51f7-841e-1c94911af66b
│                         4494b540-d91c-5ae9-9963-3b0f03e7fe2b
│     Device Flags:       • Périphérique interne
│                         • Updatable
│                         • System requires external power source
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─System Firmware:
│ │   Device ID:          66352e92895dc6d018071ad4ecfae249b0b8af0d
│ │   Version actuelle:   1.27.0
│ │   Version minimum:    1.27.0
│ │   Fournisseur:        Dell Inc. (DMI:Dell Inc.)
│ │   GUIDs:              e4d7bdc4-cc95-4aa0-a982-5e915fc04bf0
│ │                       230c8b18-8d9b-53ec-838b-6cfc0383493a
│ │                       127676e3-b872-5f45-83d2-de29b06d8dfd
│ │   Device Flags:       • Périphérique interne
│ │                       • Updatable
│ │                       • System requires external power source
│ │                       • Supported on remote server
│ │                       • Needs a reboot after installation
│ │                       • Cryptographic hash verification is available
│ │                       • Device is usable for the duration of the update
│ │ 
│ ├─Intel AMT [unprovisioned]:
│ │     Device ID:        e2623122c99d58220498aacbfcfdb1baebbae3c5
│ │     Résumé:           Hardware and firmware technology for remote out-of-band management
│ │     Version actuelle: 12.0.90.2072
│ │     Version du chargeur d’amorçage:12.0.90.2072
│ │     Fournisseur:      Intel Corporation
│ │     GUID:             2800f812-b7b4-2d4b-aca8-46e0ff65814c
│ │     Device Flags:     • Périphérique interne
│ │   
│ └─UEFI dbx:
│       Device ID:        362301da643102b9f38477387e2193e57abaa590
│       Résumé:           UEFI Revocation Database
│       Version actuelle: 83
│       Version minimum:  83
│       Fournisseur:      UEFI:Linux Foundation
│       Install Duration: 1 seconde
│       GUIDs:            00fe3755-a4d8-5ef7-ba5f-47979fbb3423
│                         4a6cd2cb-8741-5257-9d1f-89a275dacca7
│                         c6682ade-b5ec-57c4-b687-676351208742
│                         f8ba2887-9411-5c36-9cee-88995bb39731
│       Device Flags:     • Périphérique interne
│                         • Updatable
│                         • Supported on remote server
│                         • Needs a reboot after installation
│     
├─TPM:
│ │   Device ID:          c6a80ac3a22083423992a3cb15018989f37834d6
│ │   Version actuelle:   7.2.0.2
│ │   Fournisseur:        Nuvoton Technology (TPM:NTC|PCI:0x1028)
│ │   GUIDs:              ff71992e-52f7-5eea-94ef-883e56e034c6
│ │                       fac1c8f3-73c8-5cd6-8330-07a3690b5140
│ │                       bdb182b9-7533-5c43-b775-0c8327246042
│ │                       e9ccc1dc-960a-5e09-afe9-e59a904b776d
│ │                       31c399b3-9c72-58ce-8fc3-489e76ab1e35
│ │   Device Flags:       • Périphérique interne
│ │ 
│ └─Event Log:
│       Device ID:        58bd405f31c48e6eca290b425f530a94c91e955c
│       GUID:             a25657fe-b5dc-5be0-8b78-8b9dfec678ff
│       Device Flags:     • Périphérique interne
│     
├─UHD Graphics 630 (Mobile):
│     Device ID:          5792b48846ce271fab11c4a545f7a3df0d36e00a
│     Fournisseur:        Intel Corporation (PCI:0x8086)
│     GUIDs:              05714aa3-3a30-5606-ba13-ab4f1a48721a
│                         4d089443-bd6e-58f7-b664-d506e9ebe075
│                         12c60b98-52cd-500a-b99e-087aa6b47f0b
│                         db677673-1102-5481-a11e-6f397e123ced
│     Device Flags:       • Périphérique interne
│                         • Cryptographic hash verification is available
│   
└─USB2.0 Hub:
      Device ID:          b6bcdb236926a2452b472d3dad455c414145b52e
      Résumé:             USB 2.x Hub
      Version actuelle:   90.33
      Fournisseur:        VIA Labs, Inc. (USB:0x2109)
      Install Duration:   15 secondes
      GUIDs:              a391c569-168c-5516-ac60-207b4c597f36
                          0b3d95fe-5e53-5b5f-ad53-939138d026ff
                          32989214-b2d1-5902-a3fc-b90a5536b190
                          009c01ea-7f60-5bb9-a94b-326f460cdddf
                          aa831a2f-63a6-5689-8358-4e5d041c8d40
                          6f3b120d-c948-5a52-987a-bf212d152948
      Device Flags:       • Updatable
                          • Cryptographic hash verification is available
                          • Device stages updates
                          • Device can recover flash failures
    
________________________________________________

Devices that were not updated correctly:

 • UEFI dbx (83 → 217)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection):
0.	Do not upload reports at this time, but prompt again for future updates
1.	Do not upload reports, and never ask to upload reports for future updates
2.	Upload reports just this one time, but prompt again for future updates
3.	Upload reports this time and automatically upload reports after completing future updates

- This problem has been present for several weeks and no update has been released.

**Additional questions**

- Operating system and version : Debian 11 Stable
- Have you tried rebooting ? Yes
- Is this a regression ? No

hughsie, 2022-10-20:
I'd say the first thing to try is update the fwupd version to something newer. 1.5.7 is kinda old...

hughsie, 2022-10-20:
> Could you provide us with a new version that works in Debian Stable, Debian Backports or Debian Testing ?

Not me, sorry. @superm1 might be able to help.

Phil995511, 2022-10-20:
Would you be very kind to pass on the breakdown to him ?

Have a great day and best regards.

superm1, 2022-10-20:
Just tried it in a buster lxc container.  No problems.
[debian_testing.txt](https://github.com/fwupd/fwupd/files/9830333/debian_testing.txt)

Phil995511, 2022-10-20:
Hello superm1 and thank you for your messages.

When trying to install fwupd from Synaptic then it refuses to do so citing the presence of broken packages as the reason.

And when I perform the command line installation it works !?

`apt install -t testing fwupd`

Synaptic's behavior is strange on this one...

Thank you for everything, we can close the ticket.

closed 2022-10-20 by Phil995511