fwupd/fwupd #5185: Bind mount lead to ESP mountpoint misdetection

noraj, 2022-10-21:
**Describe the bug**

My issue seems similar to https://github.com/fwupd/fwupd/issues/75#issuecomment-266227180

My ESP is mounted to `/efi`, then I bind mounted `/efi/EFI/arch` to `/boot`.

`/etc/fstab`

```
# <file system> <dir> <type> <options> <dump> <pass>
# /dev/nvme0n1p1
UUID=2D85-D984          /efi            vfat            rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors=remount-ro   0 2

/efi/EFI/arch   /boot   none    default,bind    0       0

The issue is that `fwupd` tries to apply updates for Dell firmware update to `/boot/EFI/arch/fw` and says it's read-only.

I thought it was the ESP path misdected due to the bind mount so I decided to modify the config to tell `fwupd` the real ESP path.

`/etc/fwupd/uefi_capsule.conf`

```ini
[uefi_capsule]
OverrideESPMountPoint=/efi
```

I rebooted but now the deamon refuse to start while the correct ESP mountpoint is specified.

```
systemd[1]: Starting Firmware update daemon...
fwupd[4622]: Failed to load daemon: failed to load engine: No ESP with path /efi
systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
systemd[1]: fwupd.service: Failed with result 'exit-code'.
systemd[1]: Failed to start Firmware update daemon.
```

My ESP is as follows:

```
$ tree /efi -d
/efi
└── EFI
    ├── arch
    ├── Dell
    │   └── logs
    ├── refind
    │   ├── icons
    │   │   └── svg
    │   ├── keys
    │   ├── themes
    │   │   └── refind-theme-regular
    │   │       ├── fonts
    │   │       └── icons
    │   │           ├── 128-48
    │   │           ├── 256-96
    │   │           ├── 384-144
    │   │           └── 512-192
    │   └── vars
    └── tools

18 directories
```

**Steps to Reproduce**

See previous section. I have a [Dell Latitude 3420](https://wiki.archlinux.org/title/Dell_Latitude_3420).

**Expected behavior**

Correct ESP mountpoint detection, eitheir automatically or manually.

**fwupd version information**

tmp: I need to make the deamon work again first.

```shell
fwupdmgr --version
```

Please note how you installed it : pacman

<details>

<summary>**fwupd device information**</summary>

tmp: I need to make the deamon work again first.

```shell
fwupdmgr get-devices --show-all-devices
```

</details>

**System UEFI configuration** (there are a bunch of old entries)

```shell
efibootmgr -v
BootCurrent: 0005
Timeout: 2 seconds
BootOrder: 0005,0006,0003,0001,0002,0000,0004,0007
Boot0000* USB NIC (IPV4)        PciRoot(0x0)/Pci(0xd,0x0)/USB(1,0)/USB(0,0)/MAC(144fd7c56997,0)/IPv4(0.0.0.00.0.0.0,0,0){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 0d / 03 05 06 00 01 00 / 03 05 06 00 00 00 / 03 0b 25 00 14 4f d7 c5 69 97 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
Boot0001* ONBOARD NIC (IPV4)    PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/MAC(b445063f8bd6,0)/IPv4(0.0.0.00.0.0.0,0,0){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 1d / 01 01 06 00 00 00 / 03 0b 25 00 b4 45 06 3f 8b d6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
Boot0002* ONBOARD NIC (IPV6)    PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/MAC(b445063f8bd6,0)/IPv6([::]:<->[::]:,0,0){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 1d / 01 01 06 00 00 00 / 03 0b 25 00 b4 45 06 3f 8b d6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
Boot0003* UEFI HTTPs Boot       PciRoot(0x0)/Pci(0x1d,0x0)/Pci(0x0,0x0)/MAC(b445063f8bd6,0)/IPv4(0.0.0.00.0.0.0,0,0)/Uri(){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 1d / 01 01 06 00 00 00 / 03 0b 25 00 b4 45 06 3f 8b d6 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 18 04 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
Boot0004* USB NIC (IPV6)        PciRoot(0x0)/Pci(0xd,0x0)/USB(1,0)/USB(0,0)/MAC(144fd7c56997,0)/IPv6([::]:<->[::]:,0,0){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 0d / 03 05 06 00 01 00 / 03 05 06 00 00 00 / 03 0b 25 00 14 4f d7 c5 69 97 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0d 3c 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 40 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
Boot0005* rEFInd Boot Manager   HD(1,GPT,86e3e333-fd26-fc41-af8f-32e81bb51dd4,0x800,0x113000)/File(\EFI\refind\refind_x64.efi)
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 30 11 00 00 00 00 00 33 e3 e3 86 26 fd 41 fc af 8f 32 e8 1b b5 1d d4 02 02 / 04 04 3a 00 5c 00 45 00 46 00 49 00 5c 00 72 00 65 00 66 00 69 00 6e 00 64 00 5c 00 72 00 65 00 66 00 69 00 6e 00 64 00 5f 00 78 00 36 00 34 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
Boot0006* Windows Boot Manager  HD(1,GPT,0dcf90ad-1e3b-42d4-a5df-aeb9e58aadfe,0x800,0x96000)/File(\EFI\Microsoft\Boot\bootmgfw.efi)57494e444f5753000100000088000000780000004200430044004f0042004a004500430054003d007b00390064006500610038003600320063002d0035006300640064002d0034006500370030002d0061006300630031002d006600330032006200330034003400640034003700390035007d00000048000100000010000000040000007fff0400
      dp: 04 01 2a 00 01 00 00 00 00 08 00 00 00 00 00 00 00 60 09 00 00 00 00 00 ad 90 cf 0d 3b 1e d4 42 a5 df ae b9 e5 8a ad fe 02 02 / 04 04 46 00 5c 00 45 00 46 00 49 00 5c 00 4d 00 69 00 63 00 72 00 6f 00 73 00 6f 00 66 00 74 00 5c 00 42 00 6f 00 6f 00 74 00 5c 00 62 00 6f 00 6f 00 74 00 6d 00 67 00 66 00 77 00 2e 00 65 00 66 00 69 00 00 00 / 7f ff 04 00
    data: 57 49 4e 44 4f 57 53 00 01 00 00 00 88 00 00 00 78 00 00 00 42 00 43 00 44 00 4f 00 42 00 4a 00 45 00 43 00 54 00 3d 00 7b 00 39 00 64 00 65 00 61 00 38 00 36 00 32 00 63 00 2d 00 35 00 63 00 64 00 64 00 2d 00 34 00 65 00 37 00 30 00 2d 00 61 00 63 00 63 00 31 00 2d 00 66 00 33 00 32 00 62 00 33 00 34 00 34 00 64 00 34 00 37 00 39 00 35 00 7d 00 00 00 48 00 01 00 00 00 10 00 00 00 04 00 00 00 7f ff 04 00
Boot0007* UEFI HTTPs Boot       PciRoot(0x0)/Pci(0xd,0x0)/USB(1,0)/USB(0,0)/MAC(144fd7c56997,0)/IPv4(0.0.0.00.0.0.0,0,0)/Uri(){auto_created_boot_option}
      dp: 02 01 0c 00 d0 41 03 0a 00 00 00 00 / 01 01 06 00 00 0d / 03 05 06 00 01 00 / 03 05 06 00 00 00 / 03 0b 25 00 14 4f d7 c5 69 97 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 0c 1b 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 / 03 18 04 00 / 7f ff 04 00
    data: 4e ac 08 81 11 9f 59 4d 85 0e e2 1a 52 2c 59 b2
```

tmp: I need to make the deamon work again first.

```shell
efivar -l | grep fw
```

`/boot` = `/efi/EFI/arch`

```shell
$ tree /boot
/boot
├── initramfs-linux-fallback.img
├── initramfs-linux.img
├── initramfs-linux-lts-fallback.img
├── initramfs-linux-lts.img
├── intel-ucode.img
├── refind_linux.conf
├── vmlinuz-linux
└── vmlinuz-linux-lts
```

**Additional questions**

- Operating system and version:  ArchLinux
- Have you tried rebooting? Yes
- Are you using an NVMe disk? Yes
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? No

noraj, 2022-10-21:
I found https://wiki.archlinux.org/title/Fwupd#read-only_filesystem_error, that I missed earlier.

I unmounted the bind mout (`umount /boot`) and now the daemon starts again. I'll update my post.

superm1, 2022-10-21:
Try adding the extra paths you use to the systemd unit for fwupd.  There is a field in there for which paths to be read/write.

noraj, 2022-10-21:
> Try adding the extra paths you use to the systemd unit for fwupd. There is a field in there for which paths to be read/write.

You're talking about `OverrideESPMountPoint`?

superm1, 2022-10-21:
No, in the `/lib/systemd/units/fwupd.service` or so.  There is a key there for read only paths.  We build it dynamically at build time.

https://github.com/fwupd/fwupd/blob/main/data/meson.build#L85

If your path isn't there it won't be able to do things.

noraj, 2022-10-24:
`sudo systemctl edit fwupd.service` (`/usr/lib/systemd/system/fwupd.service` for me). I already have `ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub`

superm1, 2022-10-24:
> `sudo systemctl edit fwupd.service` (`/usr/lib/systemd/system/fwupd.service` for me). I already have `ReadWritePaths=-/boot/efi -/efi/EFI -/boot/EFI -/boot/grub`

I think you need /boot added to the list for your configuration, no?

vascorsd, 2023-01-10:
Hey all. I think I have exactly the same setup and am hitting the problem too. I knew this was working sometime ago, but for a while I hadn't seen any updates reported so started debugging and the daemon was failing.

So,

I have `/efi` mounted as the ESP in fstab, rw:
```
❯ lx /efi
drwxr-xr-x - root 13 Oct  2021  99b6a6fbc3654a27a35acb723c14c894
drwxr-xr-x - root 19 Oct  2021  EFI
drwxr-xr-x - root 10 Jan 20:30  loader

❯ lsblk
NAME                             MAJ:MIN RM   SIZE RO TYPE  MOUNTPOINTS
nvme0n1                          259:0    0 953.9G  0 disk  
├─nvme0n1p1                      259:1    0   512M  0 part  /boot
│                                                           /efi
└─nvme0n1p2                      259:2    0 953.4G  0 part  
  └─universe                     254:0    0 953.4G  0 crypt 
    ├─system--vg-archlinux--root 254:1    0   150G  0 lvm   /
    └─system--vg-linux--home     254:2    0 803.4G  0 lvm   /home

❯ bat /etc/fstab
  11   │ # /dev/nvme0n1p1
  12   │ UUID=B26D-617A                                  /efi        vfat        rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,utf8,errors
       │ =remount-ro   0 2
  13   │ 
  14   │ # custom:
  15   │ # add bind mount for /boot to be naturally put in the efi partition without additional work
  16   │ /efi/EFI/linux-archlinux                        /boot           none            defaults,bind    0 0
```

I had the uefi_capsule not updated, so after changing it to have it correctly using:
```
1    │ [uefi_capsule]
20   │ # CUSTOM / MINE - it doesn't like that I have both
21   │ # /efi and /boot at the same time and that is the conflict, so need
22   │ # to be explicit
23   │ OverrideESPMountPoint=/efi
```

it seems that restarting the daemon with `systemctl restart` actually launches it the first time, but the second time it fails. This seems to be because the daemon is copying the ESP mountpoint value to the daemon.conf file and aborts the second time when the value is already there. So, step by step:

1. clean daemon.conf from having any ESP mountpoint value defined, just reverting the file to the default.
2. start the daemon with full verbose messages and we can see the message saying that the daemon.conf file is being updated with the new esp value but constantly complains that /efi is not a valid esp mountpoint. But it still stays up.
3. restart the daemon again, which now has the esp value in the conf file that it updated itself and fails at the start by complaining about the /efi not being valid.

journalctl:
```
Jan 10 20:47:27 abra systemd[1]: Starting Firmware update daemon...
Jan 10 20:47:27 abra fwupd[6236]: 20:47:27.236 GLib-GIO             Using cross-namespace EXTERNAL authentication (this will deadlock if server is GDBus < 2.73.3)
Jan 10 20:47:27 abra fwupd[6236]: 20:47:27.247 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: crypto_LUKS
Jan 10 20:47:27 abra fwupd[6236]: 20:47:27.251 FuVolume             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
Jan 10 20:47:27 abra fwupd[6236]: Failed to load daemon: failed to load engine: No ESP with path /efi
Jan 10 20:47:27 abra systemd[1]: fwupd.service: Main process exited, code=exited, status=1/FAILURE
Jan 10 20:47:27 abra systemd[1]: fwupd.service: Failed with result 'exit-code'.
Jan 10 20:47:27 abra systemd[1]: Failed to start Firmware update daemon.
Jan 10 20:49:07 abra systemd[1]: Starting Firmware update daemon...
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.988 FuRemoteList         loading /usr/share/fwupd/metainfo
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.989 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.989 XbSilo               file: 32ebf294-094b-295f-42ed-c9a177edb5f0, current:743728b3-d1ac-1f55-3177-5be2f338ba5a, cached: (null)
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.992 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.992 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.997 FuRemoteList         loading remote from /etc/fwupd/remotes.d/fwupd-tests.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.997 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs.conf.pacnew
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.997 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.997 FuRemoteList         skipping invalid file /etc/fwupd/remotes.d/lvfs-testing.conf.pacnew
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.997 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.998 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.998 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
Jan 10 20:49:07 abra fwupd[6522]: 20:49:07.998 FuRemoteList         ignoring unfound remote fwupd
[... 543 lines cut ...]
       Status: active, boot-order
    Partition: /dev/disk/by-partuuid/e7182744-f27c-43fd-9a64-353c537b6ff5
         File: └─/EFI/systemd/systemd-bootx64.efi

        Title: Linux Firmware Updater
           ID: 0x0000
[... cut ...]
        title: Arch Linux (ZEN)
           id: archlinux-zen.conf
       source: /efi/loader/entries/archlinux-zen.conf
        linux: /EFI/linux-archlinux/vmlinuz-linux-zen
       initrd: /EFI/linux-archlinux/intel-ucode.img
               /EFI/linux-archlinux/initramfs-linux-zen.img
      options: cryptdevice=UUID=24a231f5-7bbc-43d3-a35c-04a429d630b4:universe root=/dev/system-vg/archlinux-root mem_sleep_default=deep resume=/dev/system-vg/archlinux-ro>
```

Hope this is enough information to finally fix the problem.

superm1, 2023-01-10:
I think you should just be able to update https://github.com/fwupd/fwupd/blob/main/data/daemon.conf#L66 that line in daemon.conf to help it.

vascorsd, 2023-01-10:
That line gets updated automatically when launching the daemon when it detects that the uefi_capsule configuration changed the similar option. So yes, currently, because it updated it by itself, it's pointing to the correct location that I manually defined in the uefi capsule. So, both `daemon.conf` has `EspLocation=/efi` and `uefi_capsule.conf` has `OverrideESPMountPoint=/efi`. Since that is mine ESP location, it should work, but the daemon says it doesn't like it as described above in the logs and fails to launch.

PR #5405 (2023-01-11): Respect user requested paths for ESP even if they aren't volumes
These paths may be used for bind mounts, which we can't accurately discover. Leave a message in the logs accordingly, but allow these paths.

fixes: #5185

Type of pull request:

- [ ] New plugin (Please include [new plugin checklist](https://github.com/fwupd/fwupd/wiki/New-plugin-checklist))
- [x] Code fix
- [ ] Feature
- [ ] Documentation
commit 5a24472f (2023-01-11): Respect user requested paths for ESP even if they aren't volumes

These paths may be used for bind mounts, which we can't accurately
discover. Leave a message in the logs accordingly, but allow these
paths.

fixes: #5185
closed 2023-01-11 by hughsie