fwupd/fwupd #5188: "Failed to run update on reboot" - how to diagnose?

wolfgang42, 2022-10-22:
**Describe the bug**
I have a couple of firmware updates ("Lenovo Intel Management Engine" and "Lenovo System Firmware") that are failing to install and I can't figure out why; `fwupdmgr get-upgrades` links me to a [wiki page](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot) that is no help.

Things I've tried:
* Confirmed Secure Boot was disabled in setup menu
* Confirmed Boot Order Lock disabled, ditto
* Waited a few months to see if the problem went away
* Skimmed `shim`-related bug reports, but I don't really understand how any of this stuff works
* Fiddled with `efibootmgr` to move "Linux-Firmware-Updater" to the top of the list
* Ran `sudo fwupdate -L`, which just returned:
  ```
   failed: Error opening file /sys/firmware/efi/efivars/FWUPDATE_DEBUG_LOG-0abba7dc-e516-4167-bbf5-4d9d1c739416: No such file or directory
  ```
* Opened the boot order menu when the system was starting up and told it to boot from "Linux Firmware Updater". The screen went black briefly and then dropped me back in the menu, though I've no idea what was supposed to happen.

**Steps to Reproduce**
1. `sudo fwupdmgr upgrade` and answer 'yes'
2. Reboot
3. Upgrade is still not installed

**Expected behavior**
Upgrade should be installed after reboot.

**fwupd version information**
Please provide the version of the daemon and client.
<details><summary><code>fwupdmgr --version</code></summary>
<pre>
runtime   org.freedesktop.fwupd         1.7.5
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.4
runtime   org.kernel                    5.15.0-52-generic
compile   org.freedesktop.fwupd         1.7.5
runtime   org.freedesktop.gusb          0.3.10
</pre>
</details>
Please note how you installed it (`apt`, `dnf`, `pacman`, source, etc): Came with system (KDE Neon)

<details><summary><code>i A  --\ fwupd   1.7.5-3~20.04.</code></summary>

```
  Description: Firmware update daemon

  Homepage: https://github.com/fwupd/fwupd
  Multi-Arch: foreign
  Priority: optional
  Section: admin
  Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
  Architecture: amd64
  Compressed Size: 2,423 k
  Uncompressed Size: 6,709 k
  Source Package: fwupd
  Label: Ubuntu
  Origin: Ubuntu:20.04/focal-updates [amd64]
  Origin URI: http://us.archive.ubuntu.com/ubuntu/pool/main/f/fwupd/fwupd_1.7.5-3~20.04.1_amd64.deb
```
</details>

<details><summary><strong>fwupd device information</strong></summary>

Please provide the output of the fwupd devices recognized in your system.

```shell
fwupdmgr get-devices --show-all-devices
```
```
20KHCTO1WW
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
[... 515 lines cut ...]
├── memtest86+_multiboot.bin
├── System.map-5.15.0-50-generic
├── System.map-5.15.0-52-generic
├── System.map-5.4.0-128-generic
├── System.map-5.4.0-131-generic
├── vmlinuz -> vmlinuz-5.15.0-52-generic
├── vmlinuz-5.15.0-50-generic
├── vmlinuz-5.15.0-52-generic
├── vmlinuz-5.4.0-128-generic
├── vmlinuz-5.4.0-131-generic
└── vmlinuz.old -> vmlinuz-5.4.0-131-generic

8 directories, 323 files
```

</details>

**Additional questions**

- Operating system and version: KDE Neon 5.26 (like Ubuntu 20.04)
- Have you tried rebooting? Yes, many times
- Is this a regression? No idea
- Are you using an NVMe disk? I think so? `/dev/nvme0` exists anyway
- Is secure boot enabled? No
- Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS? Lenovo yes, "Boot Order Lock" no AFAICT

hughsie, 2022-10-23:
Does it work if you disable secure boot? We don't require it, but it'll help debugging.

wolfgang42, 2022-10-23:
As I mentioned in the OP, making sure that both Secure Boot and Boot Order Lock were disabled in the F1 "BIOS Setup Utility" was the first thing I did after reading the wiki page that `fwmgr get-upgrades` linked me to:

[image]

[image]

hughsie, 2022-10-24:
I guess this is a  Lenovo ThinkPad X1 Carbon 6th -- are you trying to install the Management Engine update, the System firmware update or the EC update?

wolfgang42, 2022-10-30:
> I guess this is a Lenovo ThinkPad X1 Carbon 6th

Yep—though I'm curious how you guessed that, since I seem to have forgotten to note that in the description.

> are you trying to install the Management Engine update, the System firmware update or the EC update?

I only see two updates, "Lenovo Intel Management Engine" and "Lenovo System Firmware". I don't recall seeing an "EC update" mentioned anywhere.

hughsie, 2022-10-31:
> though I'm curious how you guessed that

I just looked up the GUID on the LVFS :)

> I only see two updates, "Lenovo Intel Management Engine" and "Lenovo System Firmware"

Can you try to install just the system firmware, then reboot -- e.g. in gnome-firmware or using the CLI and saying yes to one and no to the other. Thanks.

wolfgang42, 2022-11-07:
<details><summary><code>sudo fwupdmgr upgrade a45df35ac0e948ee180fe216a5f703f32dda163f</code></summary>

```
Devices with no available firmware updates: 
 • INTEL SSDPEKKF256G8L
 • Thunderbolt host controller
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade System Firmware from 0.1.55 to 0.1.57?                               ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ Lenovo ThinkPad X1 Carbon 6th System Firmware                                ║
║                                                                              ║
║ • Updated the Diagnostics module to version 04.25.000.                       ║
║ • Fixed an issue where an incorrect Intel Management Engine firmware         ║
║ version might be                                                             ║
║                                                                              ║
║ returned at capsule update under Linux.                                      ║
║                                                                              ║
║ Do not turn off your computer or remove the AC adapter while the update is   ║
║ in progress.                                                                 ║
║                                                                              ║
║ 20KHCTO1WW must remain plugged into a power source for the duration of the   ║
║ update to avoid damage.                                                      ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Scheduling…              [***************************************]
Decompressing…           [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware
Do not turn off your computer or remove the AC adapter while the update is in progress.
```
</details>

Then, after rebooting:

<details><summary><code>fwupdmgr get-upgrades</code></summary>

```
Devices with no available firmware updates: 
 • INTEL SSDPEKKF256G8L
 • Thunderbolt host controller
Devices with the latest available firmware version:
 • Embedded Controller
 • UEFI Device Firmware
 • UEFI dbx
________________________________________________

Devices that were not updated correctly:

 • System Firmware (0.1.55 → 0.1.57)

Uploading firmware reports helps hardware vendors to quickly identify failing and successful updates on real devices.
Upload report now? (Requires internet connection) [Y|n]:

Target:                  https://fwupd.org/lvfs/firmware/report
Payload:                 {
                           "ReportVersion" : 2,
                           "MachineId" : "38e743a2dca587ae579d33f25da0b468cdd2d27bfee05c122e32fa6963317433",
                           "Metadata" : {
                             "DistroId" : "neon",
                             "DistroVersion" : "20.04",
[... 168 lines cut ...]
  └─ThinkPad X1 Carbon 6th:
        New version:      0.1.56
        Remote ID:        lvfs
        Release ID:       13903
        Summary:          Lenovo ThinkPad X1 Carbon 6th System Firmware
        License:          Proprietary
        Size:             9.5 MB
        Created:          2016-07-08
        Urgency:          High
        Vendor:           Lenovo
        Release Flags:    • Is upgrade
        Description:      
        Lenovo ThinkPad X1 Carbon 6th System Firmware
        
        • Updated the Diagnostics module to version 04.24.000.
        • Updated the CPU microcode.
        • Fixed an issue where Intel Management Engine firmware update by LVFS is failed.
        
        Drive encryption by OPAL management software needs to be disabled before
        
        start firmware update.
        
        Do not turn off your computer or remove the AC adapter while the update is in progress.
```
</details>

wolfgang42, 2022-11-07:
> what is shown in the BIOS setup screen

I assume one of these is the number you're looking for:
```
UEFI BIOS Version:                 N23ET80W (1.55 )
UEFI BIOS Date (Year-Month-Day)    2021-11-16
Embedded Controller Version        N23HT37W (1.22 )
ME Firmware Version                11.8.90.3987
```

> Anything displayed on screen during the reboot?

Nothing out of the ordinary; just the Lenovo splash screen and then Plymouth.

mrhpearson, 2022-11-07:
Thanks.
What happens if you press F12 during the early boot and then from the boot menu choose 'Firmware updater' (or whatever it is called that is similar)?

wolfgang42, 2022-11-08:
I believe I've tried that one already:

> Opened the boot order menu when the system was starting up and told it to boot from "Linux Firmware Updater". The screen went black briefly and then dropped me back in the menu, though I've no idea what was supposed to happen.

wolfgang42, 2022-11-22:
Following up on this: Neon wants me to do a major version upgrade by the end of November, which will take me to 22.04 and presumably new versions of fwupd et al. Any other debugging you want to do before I try the upgrade?

hughsie, 2022-11-22:
Lets try with the fwupd-efi in 22.04

ffries, 2022-11-24:
I can report the same behavior with a Lenovo x280 and Debian testing. 
To upgrade the Bios, I can only use Lenovo update CD.

wolfgang42, 2022-11-30:
@ffries worth noting that your issue is covered by [the wiki page on this topic](https://github.com/fwupd/fwupd/wiki/LVFS-Triaged-Issue:-Failed-to-run-update-on-reboot) which links to some ongoing investigations into why Secure Boot causes problems. The issue I'm having is different: I have (as far as I can tell) Secure Boot disabled, and I *still* couldn't update for some reason. (And I wasn't getting any error messages, either.)

@hughsie Having upgraded to 22.04:

<details><summary>the same diagnostics as at the top of the thread</summary>

<pre>$ fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-53-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10

$ apt show fwupd
Package: fwupd
Version: 1.7.9-1~22.04.1
Priority: optional
Section: admin
Origin: Ubuntu
Maintainer: Ubuntu Developers <ubuntu-devel-discuss@lists.ubuntu.com>
Original-Maintainer: Debian EFI <debian-efi@lists.debian.org>
Bugs: https://bugs.launchpad.net/ubuntu/+filebug
Installed-Size: 6,898 kB
Provides: fwupdate
Depends: libfwupd2 (= 1.7.9-1~22.04.1), libfwupdplugin5 (= 1.7.9-1~22.04.1), libc6 (>= 2.34), libcurl3-gnutls (>= 7.63.0), libefiboot1 (>= 37), libglib2.0-0 (>= 2.70.0), libgnutls30 (>= 3.7.3), libgudev-1.0-0 (>= 165), libgusb2 (>= 0.3.8), libjcat1 (>= 0.1.4), libjson-glib-1.0-0 (>= 1.5.2), libmbim-glib4 (>= 1.26.0), libmbim-proxy, libmm-glib0 (>= 1.18), libpolkit-gobject-1-0 (>= 0.99), libqmi-glib5 (>= 1.18.0), libqmi-proxy, libsmbios-c2, libsqlite3-0 (>= 3.6.1), libsystemd0, libtss2-esys-3.0.2-0 (>= 2.3.1), libxmlb2 (>= 0.3.2), shared-mime-info
Recommends: python3, bolt, dbus, secureboot-db, udisks2, fwupd-signed
Suggests: gir1.2-fwupd-2.0
Conflicts: fwupdate-amd64-signed, fwupdate-arm64-signed, fwupdate-armhf-signed, fwupdate-i386-signed
Breaks: fwupdate (<< 12-7), gir1.2-dfu-1.0 (<< 0.9.7-1), libdfu-dev (<< 0.9.7-1), libdfu1 (<< 0.9.7-1)
Replaces: fwupdate (<< 12-7), gir1.2-dfu-1.0 (<< 0.9.7-1), libdfu-dev (<< 0.9.7-1), libdfu1 (<< 0.9.7-1)
Homepage: https://github.com/fwupd/fwupd
Task: ubuntu-desktop-minimal, ubuntu-desktop, cloud-image, ubuntu-desktop-raspi, server, ubuntu-server-raspi, kubuntu-desktop, xubuntu-core, xubuntu-desktop, lubuntu-desktop, ubuntustudio-desktop-core, ubuntustudio-desktop, ubuntukylin-desktop, ubuntu-mate-core, ubuntu-mate-desktop, ubuntu-budgie-desktop, ubuntu-budgie-desktop-raspi
Download-Size: 2,561 kB
APT-Manual-Installed: no
APT-Sources: http://us.archive.ubuntu.com/ubuntu jammy-updates/main amd64 Packages
Description: Firmware update daemon

$ fwupdmgr get-devices --show-all-devices
20KHCTO1WW
│
├─Core™ i7-8550U CPU @ 1.80GHz:
│     Device ID:          4bde70ba4e39b28f9eab1628f9dd6e6244c03027
│     Current version:    0x000000f0
│     Vendor:             Intel
│     GUIDs:              b9a2dd81-159e-5537-a7db-e7101d164d3f
│                         30249f37-d140-5d3e-9319-186b1bd5cac3
│                         561403e8-143a-5071-ab09-bf5e4c146983
│                         54bdc429-0179-5019-9cb6-8a343324c97c
│     Device Flags:       • Internal device
│   
├─Embedded Controller:
│     Device ID:          2292ae5236790b47884e37cf162dcf23bfcd1c60
│     Summary:            UEFI ESRT device
│     Current version:    0.1.22
│     Minimum Version:    0.1.22
│     Vendor:             Lenovo (DMI:LENOVO)
│     Update State:       Success
│     GUID:               3babca5f-b2bf-4f4b-a72e-2bdc84eb4019
│     Device Flags:       • Internal device
│                         • Updatable
│                         • System requires external power source
│                         • Supported on remote server
│                         • Needs a reboot after installation
│                         • Device is usable for the duration of the update
│   
├─INTEL SSDPEKKF256G8L:
│     Device ID:          04e17fcf7d3de91da49a163ffe4907855c3648be
│     Summary:            NVM Express solid state drive
[... 514 lines cut ...]
Do not turn off your computer or remove the AC adapter while the update is in progress.
</pre></details>  

And then during reboot I got “Please wait while we install a system update”, and subsequently:
<details><summary><code>sudo fwupdmgr get-upgrades</code></summary>

<pre>Devices with no available firmware updates: 
 • INTEL SSDPEKKF256G8L
Devices with the latest available firmware version:
 • Embedded Controller
 • System Firmware
 • Thunderbolt host controller
 • UEFI Device Firmware
 • UEFI dbx
________________________________________________

Devices that have been updated successfully:

 • Thunderbolt host controller (43.00 → 47.00)
 • System Firmware (0.1.55 → 0.1.58)
</pre></details>

(the other update also installed successfully, though I didn't capture output)

So it looks like from my end this was fixed by the upgrade to Focal, though I’m happy to try to capture any additional debug info you might want me to get and I'll let you decide whether to close this issue or leave it open for further discussion.

closed 2022-11-30 by hughsie