fwupd/fwupd #5193: A solution to the fwupd problem

Gaturinho, 2022-10-24:
I found a solution to the fwupd issue (reported at https://github.com/fwupd/fwupd/issues/3037) and already closed, but still not resolved. As I learned from the forums, the admins closed the discussion only because of the excess of complaints, which were disturbing the discussion and making it unproductive. However, the reported problem remains undefeated and unresolved, and continues to affect systems, even in fwupd version 1.7.5. But I found a solution for it, and it's simple. Perhaps there is interest in reopening or at least testing and/or making the solution available.
     
    I was able to solve the problem by installing a recommended package from fwupd. The excessive notifications are gone now, and it's only showing up during system boot, and only one notification, just once. It's been a little over 2 weeks since I started having "blows" of notifications with 4, 5 or even more, notifications in a row, and the outbreaks were repeated at least once an hour, all announcing a "network event", which I thought be just disconnection and reconnection events from the wifi network. At first I thought it was just my router's unstable connection to the network; but the absurd and frequent amount of notifications made me worried. Upon closer examination of the notifications, I noticed that they all came with complex code that I discovered was the addressing of my network driver on the system. Searching the internet, for pages that reported similar warnings in content ( systemd[1]: fwupd-refresh.service: Failed with result 'exit-code'. ), I came to this site here and several others, all with the same problem and all without a solution, with everyone lost and trying things that are kind of random and not always really recommendable. And instead of testing solutions like disabling or masking the service, uninstalling fwupd, or changing configuration files that do who knows what... I decided to try my approach. Even because it wasn't just anything with a problem, but the network driver (that's what it implied... although it wasn't really that), and it was clear that if I lost the network driver, it would be quite possible that I wouldn't managed to repair the system afterwards. So, the approaches cited all seemed to me something like hanging yourself in order not to be killed. So I tried my own approach, based on my vast experience in installing dependencies by fingernail. After unsuccessfully trying to reinstall fwupd, checking all fwupd dependencies and trying to reinstall them - all without success - I turned to my aforementioned vast experience, and the bizarre tricks it taught me: In my experience, sometimes installing some simple -dev package of a certain dependency fixes its bugs. But it's NOT that -dev packages have some magical power to fix buggy depes; but it just so happens that installing certain additional packages - usually -devs - causes the installation of the new package to re-install and reset the buggy package settings - logically if the problem is not in the package code, but in its settings, this might solve . So I searched Synaptic for some "fwupd-dev" or something, and there were several devs, but none seemed to be fwupd dev; but when searching, I found that there was a package recommendation (right click on the package and select from the menu the option "recommended packages"). This brings up a list, if there are recommended packages; these recommended lists usually indicate the installation of a lot of useless junk. But sometimes there's fool's gold there. In this case, there really was: "gir1.2-fwupd-2.0", a gobject package. That could be it. Installed, and got rid of the problem But working for me doesn't mean it will work for others, because my system is a Linux Mint 21 "Vanessa" all customized, with a lot of git packages and running on monstrous hardware ( ASUS, 5th/6th generation i5, 16Gb DDR5, system installed in UEFI mode on a system-only SSD, and with a dedicated graphics card), and on top of that running with Tuned in performance mode, and 3 additional fans. So, it's not uncommon for a solution that works for me, slow down other machines. I feel obliged to say this caveat. So, if you want to try it, test carefully and report if there is a loss of speed or an increase in CPU and memory consumption. Please, everyone who wants to test, contribute your experience.

Gaturinho, 2022-10-27:
I didn't have to wait another half hour. The persistent notifications kept repeating themselves, so I reinstalled gir1.2-fwupd-2.0

superm1, 2022-10-31:
This issue is fixed in Ubuntu in `1.7.9-1~22.04.1` for Ubuntu 22.04, and `1.7.9-1~20.04.1` for Ubuntu 20.04.  Please upgrade to those.