fwupd/fwupd #5194: Elan haptic pad configuration support

superm1, 2022-10-25:
> > @superm1 The touchpad firmware format is different from the haptic eprom firmware format.
> 
> But the haptic eprom, can we configure any public offsets/elements for it? sensitivity, enable/disable etc? If those could potentially be readable/writable settings from fwupd side my idea is that we can build up some other infrastructure that the plugin can hook into so a user could do something _like_ this:
> 
> ```shell
> # fwupdmgr get-firmware-setting ElanHapticsEnable
> # fwupdmgr get-firmware-setting ElanHapticsSensitivity
> # fwupdmgr set-firmware-setting ElanHapticsEnable True
> # fwupdmgr set-firmware-setting ElanHapticsSensitivity 0x128
> ```
Yes, it is much more convenient for users.
There are several questios.
1. The setting of elan hapticpad depends on different modules.
2. There must be a function to record the settings, and the settings will be restored when the device/system is turned on.
3. If there is this feature in FWUPD, can it be combined with GNOMESETTING UI?.

_Originally posted by @jinglewu in https://github.com/fwupd/fwupd/issues/5180#issuecomment-1289908722_

I've split this out as a new issue to discuss the design rather than the random PR.

> The setting of elan hapticpad depends on different modules.

OK, so this might need to be something that has a table of mapping Device GUID to what's supported then.

> There must be a function to record the settings, and the settings will be restored when the device/system is turned on.

This should be some new development from fwupd daemon side and libfwupdplugin side to accomplish this.  It would have a a few side effects we would have to keep in mind:
1. fwupd will need to query all these settings at startup and restore saved settings.
2. fwupd will need to store these settings somewhere.  We have done something like this already for BIOS settings, to let users store it to json.  I think we should do the same thing for these types of settings too.

> 3. If there is this feature in FWUPD, can it be combined with GNOMESETTING UI?.

Yes; that's my goal for BIOS settings too!    My thought is we can lump BIOS and device firmware settings together.

I think once we have it fleshed out for command line the next step will be to review a design proposal with some of the GNOME UX design guys.

superm1, 2022-10-25:
Looking at hardware I have on hand, does this happen to support haptics?  I might be able to be more useful in helping the plumbing on this if so.

```
├─ELAN06A0:00 04F3:3231:
│     Device ID:          bb1e2d8a9b7b38556e0ddcefcadd7f4cb4ad60f9
│     Summary:            Touchpad
│     Current version:    0x0007
│     Bootloader Version: 0x0001
│     Vendor:             ELAN Microelectronics (HIDRAW:0x04F3)
│     GUIDs:              1df4d514-ce17-5f53-95b4-0a320a2ebd00 ← HIDRAW\VEN_04F3&DEV_3231
│                         4ddf4f23-fb60-518c-b73e-1bb7a21fe902 ← HIDRAW\VEN_04F3&DEV_3231&REV_00
│                         3e916354-f224-51b8-bbc5-0e7cd2b34b32 ← HIDRAW\VEN_04F3&DEV_3231&MOD_000D
│                         4f9fe0ae-f26e-5c1d-8a4e-a7506fb02f96 ← ELANTP\ICTYPE_13
│                         b3c0ddb8-47a5-5cb7-854a-925289d175d1 ← ELANTP\ICTYPE_13&MOD_000D
│                         10e61acc-8837-5e39-b565-458e071d9b80 ← ELANTP\ICTYPE_13&MOD_000D&DRIVER_HID
│     Device Flags:       • Internal device
│                         • Updatable
```

jinglewu, 2022-10-26:
Yes, this TP module is hapticpad for Windows11 OS.
This hepaticpad have 6 Settings.

1. haptic feedback (use microsoft PTP spec, Haptic Intensity Feacture report).
     haptic feedback off - set intensity 0.
2.  intensity (use microsoft PTP spec, Haptic Intensity Feacture report).
Firmware declare additional haptic descriptor on win11, then win 11 will create native UI for this 2 settings.
[https://learn.microsoft.com/en-us/windows-hardware/design/component-guidelines/windows-precision-touchpad-sample-report-descriptors#sample-report-descriptor-excerpt---haptic-intensity-feature-report](url)

3 to 5 are vendor defined settings. 
(hid feacture cmd report ID 0xD, use fu_elantp_hid_device_read_cmd() fu_elantp_hid_device_write_cmd() functions read & write.)

3. Click Force (3 statges Low Medium High)
4. Use top zone as TrackPoint buttons (on/off)
5. TrackPoint button click force (3 statges Low Medium High)

superm1, 2022-10-26:
I was researching it a bit and I found that haptic support for the standard types is going to be coming in via the kernel.  Here is the most recent series:

https://lore.kernel.org/linux-input/20221007182945.1654046-1-acz@semihalf.com/

@bentiss, do you think we want the vendor defined settings (3-5 above) to come via the kernel too?  Or do we want some of this to come via userspace like fwupd to advertise to other userspace?

jinglewu, 2022-10-26:
I know this kernel patch is for chromeOS with elan hapticpad.
This is reference for hutrr63b spec. [https://www.usb.org/sites/default/files/hutrr63b_-_haptics_page_redline_0.pdf](url)
This specification is that the system/OS/kernel can control the haptic waveform of the touchpad according to the force value and there will be multiple waveform that can be applied in different scenarios.
It is different elan hapticpad supported by Windows 11.

hughsie, 2023-09-23:
@jinglewu is this something you want/need to support?

jinglewu, 2023-09-25:
I have previously developed a Hapticpad Settings tool for this touchpad.

hughsie, 2023-09-25:
Okay, lets close this one then. Thanks.

closed 2023-09-25 by hughsie

salman-farooq-sh, 2024-12-18:
Sorry to revive an old thread but is it possible to remember the settings across restarts? Is there any command line way to apply the settings manually on every boot?

Girgitt, 2026-04-27:
hid-tools based workaround does not work for Elan touchpad on xps9350 - feature 90000 is not listed there (even though mentioned in the Arch Wiki section on sensitivity and haptic intensity: https://wiki.archlinux.org/title/Dell_XPS_13_(9350)_2024#Sensitivity_and_Haptic_intensity) and only feedback level can be modified not the button click pressure treshold level. 

Is there any chance to get this re-opened?