fwupd/fwupd #5211: Upgrade UEFI dbx fails with "Too many levels of symbolic links"

briskycat, 2022-10-28:
I am unable to enroll an UEFI dbx update on a Dell Precision 5750 running Ubuntu 22.04 (with KDE neon). I also tried updating it on Ubuntu 20.04 with the same outcome.
Every time I try I get 
`Blocked executable in the ESP, ensure grub and shim are up to date: Error opening directory “/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/proc/1/task/1/cwd/sys/kernel/debug/tracing”: Too many levels of symbolic links`

**Steps to Reproduce**
```
# fwupdmgr update
Devices with no available firmware updates: 
 • DELL0990:00 04F3:311C
 • SSD 970 PRO 1TB
 • TPM 2.0
 • Thunderbolt host controller
 • Thunderbolt host controller
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Micron 2300 NVMe 1024GB
 • System Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 83 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.If the installation fails, you will     ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly.You may have to        ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
Downloading…             [***************************************]
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Blocked executable in the ESP, ensure grub and shim are up to date: Error opening directory “/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/dev/fd/16/proc/1/task/1/cwd/sys/kernel/debug/tracing”: Too many levels of symbolic links
```
**fwupd version information**

```shell
# fwupdmgr --version
runtime   org.freedesktop.fwupd         1.7.9
runtime   com.dell.libsmbios            2.4
compile   org.freedesktop.gusb          0.3.10
runtime   org.kernel                    5.15.0-52-generic
compile   com.hughsie.libjcat           0.1.9
compile   org.freedesktop.fwupd         1.7.9
runtime   org.freedesktop.gusb          0.3.10
```
This is a stock Ubuntu-provided package installed using apt.

<details>

<summary>**fwupd device information**</summary>

```shell
# fwupdmgr get-devices --show-all-devices
[... 572 lines cut ...]
│       ├── xfs.mod
│       ├── xnu.mod
│       ├── xnu_uuid.mod
│       ├── xnu_uuid_test.mod
│       ├── xzio.mod
│       ├── zfscrypt.mod
│       ├── zfsinfo.mod
│       ├── zfs.mod
│       └── zstd.mod
├── initrd.img -> initrd.img-5.15.0-52-generic
├── initrd.img-5.15.0-52-generic
├── initrd.img.old -> initrd.img-5.15.0-52-generic
├── memtest86+.bin
├── memtest86+.elf
├── memtest86+_multiboot.bin
├── System.map-5.15.0-52-generic
├── vmlinuz -> vmlinuz-5.15.0-52-generic
├── vmlinuz-5.15.0-52-generic
└── vmlinuz.old -> vmlinuz-5.15.0-52-generic

16 directories, 327 files
```
Secure Boot is enabled.

I tried cleaning out the EFI boot manager entries in the "BIOS setup" leaving only the "neon" and "ubuntu", but they get recreated automatically.

hughsie, 2022-10-28:
@vathpela @superm1 any ideas?!

superm1, 2022-10-28:
I think the problem here is related to the use of neon instead of Ubuntu in the esp, but fwupd is not compiled with that perhaps?

superm1, 2022-10-28:
My guess was if you look at `/etc/os-release` I'm guessing the ID key is not `ubuntu`, but instead it's `neon` but there is a hardcoding `efi_os_dir` as `ubuntu`.  This needs double checking, it might not be the case anymore.

superm1, 2022-10-28:
Let's see `$ fwupdtool esp-list --verbose` output for the next step

briskycat, 2022-10-28:
@superm1 you are right, the release ID is neon:

```shell
# cat /etc/os-release
PRETTY_NAME="KDE neon 5.26"
NAME="KDE neon"
VERSION_ID="22.04"
VERSION="5.26"
VERSION_CODENAME=jammy
ID=neon
ID_LIKE="ubuntu debian"
HOME_URL="https://neon.kde.org/"
SUPPORT_URL="https://neon.kde.org/"
BUG_REPORT_URL="https://bugs.kde.org/"
PRIVACY_POLICY_URL="https://kde.org/privacypolicy/"
UBUNTU_CODENAME=jammy
```

```shell
# fwupdtool esp-list --verbose
15:58:19:0319 FuDebug              Verbose debugging enabled (on console 1)
15:58:19:0393 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: ext4
15:58:19:0399 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat
15:58:19:0402 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme1n1p1, type: 0fc63daf-8483-4772-8e79-3d69d8477de4, internal: 1, fs: ext4
15:58:19:0406 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p3, type: 0657fd6d-a4ab-43c4-84e5-0933c84b4f4f, internal: 1, fs: swap
Selected volume: /org/freedesktop/UDisks2/block_devices/nvme0n1p1
/boot/efi/EFI/neon/grubx64.efi
/boot/efi/EFI/neon/shimx64.efi
/boot/efi/EFI/neon/mmx64.efi
/boot/efi/EFI/neon/BOOTX64.CSV
/boot/efi/EFI/neon/grub.cfg
/boot/efi/EFI/neon/fwupdx64.efi
/boot/efi/EFI/BOOT/BOOTX64.EFI
/boot/efi/EFI/BOOT/fbx64.efi
/boot/efi/EFI/BOOT/mmx64.efi
/boot/efi/EFI/ubuntu/grub.cfg
/boot/efi/EFI/dell/bios/recovery/BIOS_PRE.rcv
/boot/efi/EFI/dell/bios/recovery/BIOS_CUR.rcv
/boot/efi/EFI/dell/logs/diags_previous.xml
/boot/efi/EFI/dell/logs/diags_current.xml
```

fwupd is from the official jammy repos:
```shell
# apt list --installed *fwupd*
Listing... Done
fwupd-signed/jammy,now 1.44+1.2-3 amd64 [installed]
fwupd/jammy-updates,now 1.7.9-1~22.04.1 amd64 [installed]
libfwupd2/jammy-updates,now 1.7.9-1~22.04.1 amd64 [installed]
libfwupdplugin5/jammy-updates,now 1.7.9-1~22.04.1 amd64 [installed]
```

superm1, 2022-10-28:
OK, so your ESP does seem properly detected AFAICT.  Can you please reproduce it and share the full output of
`sudo fwupdtool update --verbose`

briskycat, 2022-10-28:
@superm1 sure, no problem.

```
# fwupdtool update --verbose
16:57:31:0625 FuDebug              Verbose debugging enabled (on console 1)
16:57:31:0626 FuMain               locked /run/lock/fwupdtool
16:57:31:0688 FuConfig             trying to load config values from /etc/fwupd/daemon.conf
16:57:31:0688 FuCommon             reading /etc/fwupd/daemon.conf with 1930 bytes
16:57:31:0688 FuConfig             trying to load config values from /var/etc/fwupd/daemon.conf
16:57:31:0688 FuConfig             using autodetected max archive size 18,4 EB
16:57:31:0689 GLib-GIO             _g_io_module_get_default: Found default implementation local (GLocalVfs) for ‘gio-vfs’
16:57:31:0690 FuRemoteList         loading /usr/share/fwupd/metainfo
16:57:31:0690 XbSilo               attempting to load /var/cache/fwupd/metainfo.xmlb
16:57:31:0690 XbSilo               file: 37a8e733-93f3-1d6f-9655-e3f520a65739, current:bb813a97-d4ef-c9ae-a1e2-fa245b953377, cached: (null)
16:57:31:0696 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor.conf
16:57:31:0696 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs-testing.conf
16:57:31:0696 FuRemoteList         loading remote from /etc/fwupd/remotes.d/dell-esrt.conf
16:57:31:0696 FuRemoteList         loading remote from /etc/fwupd/remotes.d/vendor-directory.conf
16:57:31:0696 FuRemoteList         loading remote from /etc/fwupd/remotes.d/lvfs.conf
16:57:31:0697 FuRemoteList         skipping invalid file /var/lib/fwupd/remotes.d/lvfs
16:57:31:0697 FuRemoteList         ordering lvfs-testing=lvfs+1
16:57:31:0697 FuRemoteList         ignoring unfound remote fwupd
16:57:31:0697 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:57:31:0697 Jcat                 ignoring GPG-KEY-Hughski-Limited as not PKCS-7 certificate
16:57:31:0697 Jcat                 ignoring GPG-KEY-Linux-Foundation-Firmware as not PKCS-7 certificate
16:57:31:0697 Jcat                 trying to load certificate from /etc/pki/fwupd/LVFS-CA.pem
16:57:31:0697 Jcat                 reading /etc/pki/fwupd/LVFS-CA.pem with 1679 bytes
16:57:31:0698 Jcat                 loaded 1 certificates
16:57:31:0698 Jcat                 ignoring GPG-KEY-Linux-Vendor-Firmware-Service as not PKCS-7 certificate
16:57:31:0698 Jcat                 trying to load certificate from /etc/pki/fwupd-metadata/LVFS-CA.pem
16:57:31:0698 Jcat                 reading /etc/pki/fwupd-metadata/LVFS-CA.pem with 1679 bytes
16:57:31:0698 Jcat                 loaded 1 certificates
16:57:31:0698 Jcat                 ignoring GPG-KEY-Linux-Foundation-Metadata as not PKCS-7 certificate
16:57:31:0698 Jcat                 reading /var/lib/fwupd/pki/secret.key with 2455 bytes
16:57:31:0698 Jcat                 reading /var/lib/fwupd/pki/client.pem with 1383 bytes
16:57:31:0698 FuEngine             client certificate now exists: nothing to do
16:57:31:0698 FuHistory            trying to open database '/var/lib/fwupd/pending.db'
16:57:31:0699 FuHistory            got schema version of 7
16:57:31:0699 FuQuirks             loading quirks from /usr/share/fwupd/quirks.d
16:57:31:0709 FuQuirks             loading quirks from /var/lib/fwupd/quirks.d
16:57:31:0709 XbSilo               attempting to load /var/cache/fwupd/quirks.xmlb
16:57:31:0709 XbSilo               file: c3ab7cbe-52e7-0d14-d52d-bd3d636f1555, current:c3ab7cbe-52e7-0d14-d52d-bd3d636f1555, cached: (null)
16:57:31:0709 XbSilo               loading silo with file contents
16:57:31:0710 FuHwids              loading HwId overrides from /etc/fwupd/daemon.conf
16:57:31:0710 FuHwids              not loading HwId overrides from /var/lib/fwupd/daemon.conf
16:57:31:0711 FuHwids              smbios property Manufacturer=Dell Inc.
16:57:31:0711 FuHwids              smbios property EnclosureKind=a
16:57:31:0711 FuHwids              smbios property Family=Precision
16:57:31:0711 FuHwids              smbios property ProductName=Precision 5750
16:57:31:0711 FuHwids              smbios property ProductSku=0990
16:57:31:0711 FuHwids              smbios property BiosVendor=Dell Inc.
16:57:31:0711 FuHwids              smbios property BiosVersion=1.20.0
16:57:31:0711 FuHwids              smbios property BiosMajorRelease=01
16:57:31:0711 FuHwids              smbios property BiosMinorRelease=14
16:57:31:0711 FuHwids              smbios property FirmwareMajorRelease=ff
16:57:31:0711 FuHwids              smbios property FirmwareMinorRelease=ff
16:57:31:0711 FuHwids              smbios property BaseboardManufacturer=Dell Inc.
16:57:31:0711 FuHwids              smbios property BaseboardProduct=0XFTD8
16:57:31:0715 FuEngine             ignoring: Error opening directory “/var/lib/fwupd/local.d”: No such file or directory
16:57:31:0715 FuEngine             ignoring: Error opening directory “/usr/share/fwupd/local.d”: No such file or directory
16:57:31:0715 XbSilo               attempting to load /var/cache/fwupd/metadata.xmlb
16:57:31:0715 XbSilo               file: 1aa08ebf-bd99-aab5-ed1e-d545ad930604, current:1aa08ebf-bd99-aab5-ed1e-d545ad930604, cached: (null)
16:57:31:0715 XbSilo               loading silo with file contents
16:57:31:0716 FuEngine             1679 components now in silo
16:57:31:0725 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-5/libfu_plugin_upower.so)
16:57:31:0726 FuPlugin             init(/usr/lib/x86_64-linux-gnu/fwupd-plugins-5/libfu_plugin_synaptics_prometheus.so)
[... cut ...]
16:58:50:0970 FuPlugin             destroy(pci_mei)
16:58:50:0970 FuPlugin             destroy(upower)
16:58:50:0971 FuPlugin             destroy(modem_manager)
16:58:50:0971 FuPlugin             destroy(linux_lockdown)
16:58:50:0971 FuPlugin             destroy(linux_swap)
16:58:50:0972 FuPlugin             destroy(dell)
16:58:50:0972 FuPlugin             destroy(linux_tainted)
16:58:50:0972 FuPlugin             destroy(tpm)
16:58:50:0973 FuPlugin             destroy(redfish)
16:58:50:0973 FuPlugin             destroy(uefi_capsule)
```

superm1, 2022-10-28:
Is your ESP mounted anywhere besides `/boot/efi` before you run fwupdtool in this fail run?

briskycat, 2022-10-28:
@superm1 
No, it is mounted only once:
```shell
# mount | grep '/dev/nvme0n1p1'
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro,discard)

# mount | grep 'efi'
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro,discard)
```

superm1, 2022-10-28:
Hmm, I wonder if we're getting an unexpected value from https://github.com/fwupd/fwupd/blob/1.8.6/libfwupdplugin/fu-volume.c#L195..  
I don't know why this would have just blown up now though, this should have made regular capulse updates to the ESP explode too.

briskycat, 2022-10-28:
I ran `strace fwupdtool update` and got a huge output log. Looks like fwupdtool crawls over the root FS and all its mount points recursively over and over again. Here is a halfway snapshot of the strace output:
```
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd", {st_mode=S_IFDIR|0500, st_size=0, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 26
newfstatat(26, "", {st_mode=S_IFDIR|0500, st_size=0, ...}, AT_EMPTY_PATH) = 0
getdents64(26, 0x55595963d1c0 /* 29 entries */, 32768) = 696
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/0", {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x6), ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/1", {st_mode=S_IFCHR|0620, st_rdev=makedev(0x88, 0x6), ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/2", {st_mode=S_IFREG|0644, st_size=28601914, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/3", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/4", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/5", {st_mode=S_IFREG|0700, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/6", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/7", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/8", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/9", {st_mode=S_IFREG|0644, st_size=32768, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/10", {st_mode=S_IFIFO|0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/11", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/12", {st_mode=S_IFIFO|0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/13", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/14", {st_mode=S_IFSOCK|0777, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/15", {st_mode=0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/16", {st_mode=S_IFSOCK|0777, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/17", {st_mode=S_IFSOCK|0777, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/18", {st_mode=S_IFREG|0700, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/19", {st_mode=S_IFIFO|0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/20", {st_mode=S_IFIFO|0600, st_size=0, ...}, 0) = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd/21", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 27
newfstatat(27, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(27, 0x55595964ef80 /* 25 entries */, 32768) = 640
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/srv", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd/21/srv", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 28
newfstatat(28, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(28, 0x555959656fc0 /* 2 entries */, 32768) = 48
getdents64(28, 0x555959656fc0 /* 0 entries */, 32768) = 0
close(28)                               = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 28
newfstatat(28, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(28, 0x555959656fc0 /* 165 entries */, 32768) = 5600
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/udisks2", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/udisks2", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 29
newfstatat(29, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(29, 0x55595965f000 /* 3 entries */, 32768) = 88
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/udisks2/udisks2-inhibit", {st_mode=S_IFREG|0755, st_size=791, ...}, 0) = 0
getdents64(29, 0x55595965f000 /* 0 entries */, 32768) = 0
close(29)                               = 0
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/xorg", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
openat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/xorg", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 29
newfstatat(29, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(29, 0x55595965f000 /* 6 entries */, 32768) = 168
newfstatat(AT_FDCWD, "/dev/fd/21/dev/fd/21/lib/xorg/protocol.txt", {st_mode=S_IFREG|0644, st_size=25699, ...}, 0) = 0
```

hughsie, 2022-10-31:
Can you try dumping the array of strings that Mario pointed at? e.g. https://github.com/fwupd/fwupd/blob/1.8.6/libfwupdplugin/fu-volume.c#L198 -- I'm also confused why you're the only one hitting this, especially since you've tried both common distros...

superm1, 2022-10-31:
You can use something like d-feet to easily graphically navigate to that to get the values too if you want.

briskycat, 2022-10-31:
@superm1 I'm not sure I'm looking at the right interface or how to copy values from d-feet. Here is what I see:
[image]

briskycat, 2022-10-31:
Looks like `/boot/efi` to me, which is exactly what mount says:
```
$ mount | grep efi
efivarfs on /sys/firmware/efi/efivars type efivarfs (rw,nosuid,nodev,noexec,relatime)
/dev/nvme0n1p1 on /boot/efi type vfat (rw,relatime,fmask=0077,dmask=0077,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro,discard)
```

hughsie, 2022-11-01:
> I ran strace fwupdtool update and got a huge output log

Can you upload the whole log please. Also; a slightly bigger ask -- could you run `fwupdtool` under `gdb` and try to get a backtrace when it's doing the recursive filesystem walking -- that'll give us a clue what's happening. Thanks.

briskycat, 2022-11-01:
@hughsie here is the strace log: https://www.dropbox.com/s/j9dwx0rir23j0o0/fwupdtool.strace.txt.gz?dl=0
Ubuntu strips the debugging information off the fwupd package so I guess I will need to rebuild it myself with debug symbols...

superm1, 2022-11-01:
Actually you can get debug symbols served to you.

https://ubuntu.com/server/docs/service-debuginfod#:~:text=Ubuntu%20maintains%20its%20own%20debuginfod%20service%2C%20which%20regularly,serve%20source-code%20in%20the%20future.%20Using%20the%20service

superm1, 2022-11-02:
> @hughsie here is the strace log: https://www.dropbox.com/s/j9dwx0rir23j0o0/fwupdtool.strace.txt.gz?dl=0 Ubuntu strips the debugging information off the fwupd package so I guess I will need to rebuild it myself with debug symbols...

It appears to be recursing from `/`:

```
openat(AT_FDCWD, "/", O_RDONLY|O_NONBLOCK|O_CLOEXEC|O_DIRECTORY) = 19
newfstatat(19, "", {st_mode=S_IFDIR|0755, st_size=4096, ...}, AT_EMPTY_PATH) = 0
getdents64(19, 0x563f792f1eb0 /* 25 entries */, 32768) = 640
newfstatat(AT_FDCWD, "/srv", {st_mode=S_IFDIR|0755, st_size=4096, ...}, 0) = 0
```

briskycat, 2022-11-02:
@superm1 thanks for the hint.

Here is a gdb session:
```
Devices with no available firmware updates: 
 • DELL0990:00 04F3:311C
 • SSD 970 PRO 1TB
 • TPM 2.0
 • Thunderbolt host controller
 • Thunderbolt host controller
 • UEFI Device Firmware
Devices with the latest available firmware version:
 • Micron 2300 NVMe 1024GB
 • System Firmware
 • UEFI Device Firmware
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 83 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.If the installation fails, you will     ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly.You may have to        ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: 
[Detaching after fork from child process 233167]
Decompressing…           [ -                                     ][Detaching after fork from child process 233169]
[Detaching after fork from child process 233171]
[Detaching after fork from child process 233173]
[Detaching after fork from child process 233175]
[Detaching after fork from child process 233177]
[Detaching after fork from child process 233179]
[Detaching after fork from child process 233181]
[Detaching after fork from child process 233183]
[Detaching after fork from child process 233185]
[Detaching after fork from child process 233187]
[Detaching after fork from child process 233189]
[Detaching after fork from child process 233191]
[Detaching after fork from child process 233193]
Decompressing…           [***************************************]
Decompressing…           [  -                                    ]^C
Thread 1 "fwupdtool" received signal SIGINT, Interrupt.
__GI___fstatat64 (fd=fd@entry=-100, file=file@entry=0x555570db9150 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., buf=buf@entry=0x7fffffffc870, flag=flag@entry=0) at ../sysdeps/unix/sysv/linux/fstatat64.c:166
Download failed: Invalid argument.  Continuing without source file ./io/../sysdeps/unix/sysv/linux/fstatat64.c.
166     ../sysdeps/unix/sysv/linux/fstatat64.c: No such file or directory.
(gdb) bt
#0  __GI___fstatat64 (fd=fd@entry=-100, 
    file=file@entry=0x555570db9150 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., buf=buf@entry=0x7fffffffc870, flag=flag@entry=0) at ../sysdeps/unix/sysv/linux/fstatat64.c:166
#1  0x00007ffff77c4e76 in __GI___stat64 (
    file=file@entry=0x555570db9150 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., buf=buf@entry=0x7fffffffc870) at ../sysdeps/unix/sysv/linux/stat64.c:29
#2  0x00007ffff7b91cc3 in g_file_test (
    filename=filename@entry=0x555570db9150 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., test=test@entry=G_FILE_TEST_IS_DIR) at ../../../glib/gfileutils.c:471
#3  0x00007ffff7eebd89 in fu_common_get_file_list_internal (files=files@entry=0x555555761360, 
    directory=directory@entry=0x555570dad0c0 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., error=error@entry=0x7fffffffda30) at ../libfwupdplugin/fu-common.c:114
#4  0x00007ffff7eebd4e in fu_common_get_file_list_internal (files=files@entry=0x555555761360, 
    directory=directory@entry=0x555570a3df10 "/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/dev/fd/21/proc/1/task/1/cwd/var/lib/docker/overlay2/0947ac683c3e2aea70351f0d00e"..., error=error@entry=0x7fffffffda30) at ../libfwupdplugin/fu-common.c:115
#5  0x00007ffff7eebd4e in fu_common_get_file_list_internal (files=files@entry=0x555555761360, 
[... cut ...]
#74 0x000055555557a3b8 in fu_util_install_release (priv=priv@entry=0x5555555f2a00, rel=<optimized out>, error=error@entry=0x7fffffffdd30) at ../src/fu-tool.c:1367
#75 0x000055555557a8f6 in fu_util_update (priv=0x5555555f2a00, values=0x5555555d9f50, error=0x7fffffffde28) at ../src/fu-tool.c:1483
#76 0x000055555557404d in fu_util_cmd_array_run (array=array@entry=0x5555555f2c00, priv=priv@entry=0x5555555f2a00, command=0x7fffffffe659 "update", values=0x7fffffffe398, error=error@entry=0x7fffffffde28) at ../src/fu-util-common.c:631
#77 0x0000555555570a6c in main (argc=<optimized out>, argv=<optimized out>) at ../src/fu-tool.c:3755
(gdb) 
```

hughsie, 2022-11-02:
That's exactly what I needed, thanks! Debugging now.

hughsie, 2022-11-02:
The root issue is this:

    15:58:19:0393 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p2, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: ext4
    15:58:19:0399 FuCommon             device /org/freedesktop/UDisks2/block_devices/nvme0n1p1, type: c12a7328-f81f-11d2-ba4b-00a0c93ec93b, internal: 1, fs: vfat

i.e. your root filesystem is using the UUID for the EFI System partition. I have no idea how that would have happened. Two workarounds in https://github.com/fwupd/fwupd/pull/5231

briskycat, 2022-11-02:
@hughsie thanks a lot! So this `c12a7328-f81f-11d2-ba4b-00a0c93ec93b` is a partition type in GPT, not a partition ID and you look for the EFI System partition by this FS type, don't you?

Now I can also see it in parted:
```
# parted -l
Model: Micron 2300 NVMe 1024GB (nvme)
Disk /dev/nvme0n1: 1024GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size   File system     Name                  Flags
 1      1049kB  538MB   537MB  fat32           EFI System Partition  boot, esp
 2      538MB   887GB   886GB  ext4            rootfs                boot, esp
 3      887GB   1024GB  137GB  linux-swap(v1)  swap                  swap
```
It's a mystery how the root partition could have been marked ESP...
Interestingly, I cant't even see that esp "flag" on the root partition in KDE Partition Manager, let alone change it...
@hughsie @superm1 thank you and everyone else involved for your swift help!

briskycat, 2022-11-02:
Now that I removed these flags from the rootfs fwupd has installed the update with no issues.

closed 2022-11-02 by briskycat

heeen, 2023-01-24:
I have the same issue