fwupd/fwupd #5221: UEFI dbx 77 -> 217, failed to run update on reboot - but resulted in 267

dlehammer, 2022-10-31:
### **Describe the bug**
Hi fwupd gurus,

Unfortunately I'm unsure if this is a bug or some sort of inconsistent reporting to the end user, so here goes

#### 1. After a manual BIOS upgrade (v. [3802](https://www.asus.com/motherboards-components/motherboards/prime/prime-b450m-a-csm/helpdesk_bios/)),  `fwupdmgr` reports an UEFI dbx update is available, and I accept as follows

<details>
<summary>fwupdmgr update</summary>

```shell
$ fwupdmgr update
Devices with no available firmware updates:
 • DT01ACA200
 • HDWD120
 • KXG50ZNV1T02 NVMe TOSHIBA 1024GB
 • ST2000VX000-1ES164
 • System Firmware
 • WDC WD20EZAZ-00GGJB0
╔══════════════════════════════════════════════════════════════════════════════╗
║ Upgrade UEFI dbx from 77 to 217?                                             ║
╠══════════════════════════════════════════════════════════════════════════════╣
║ This updates the dbx to the latest release from Microsoft which adds         ║
║ insecure versions of grub and shim to the list of forbidden signatures due   ║
║ to multiple discovered security updates.                                     ║
║                                                                              ║
║ Before installing the update, fwupd will check for any affected executables  ║
║ in the ESP and will refuse to update if it finds any boot binaries signed    ║
║ with any of the forbidden signatures.If the installation fails, you will     ║
║ need to update shim and grub packages before the update can be deployed.     ║
║                                                                              ║
║ Once you have installed this dbx update, any DVD or USB installer images     ║
║ signed with the old signatures may not work correctly.You may have to        ║
║ temporarily turn off secure boot when using recovery or installation media,  ║
║ if new images have not been made available by your distribution.             ║
║                                                                              ║
║ UEFI dbx and all connected devices may not be usable while updating.         ║
╚══════════════════════════════════════════════════════════════════════════════╝

Perform operation? [Y|n]: y
Downloading…             [***************************************]
Decompressing…           [***************************************]
Decompressing…           [***************************************]
Authenticating…          [***************************************]
Authenticating…          [***************************************]
Restarting device…       [***************************************]
Writing…                 [***************************************]
Decompressing…           [***************************************]
Writing…                 [***************************************]
Restarting device…       [***************************************]
Waiting…                 [***************************************]
Successfully installed firmware

An update requires a reboot to complete. Restart now? [y|N]: y
```
</details>

#### 2. I reboot the machine, and attempt to determine if the upgrade was a success

<details>
<summary>fwupdmgr get-upgrades</summary>

```shell
$ fwupdmgr get-upgrades
Devices with no available firmware updates:
 • DT01ACA200
 • HDWD120
 • KXG50ZNV1T02 NVMe TOSHIBA 1024GB
 • ST2000VX000-1ES164
 • System Firmware
[... 695 lines cut ...]
Codename:       jammy
```
> - Have you tried rebooting?

Yes, 3+ times so far - the machine is usable, but consistently reports UEFI dbx 267.

> - Is this a regression?

Who am I to judge :)

> - Are you using an NVMe disk?

Yes, KXG50ZNV1T02 for boot and OS partitions

> - Is secure boot enabled?

No, see also error report for details

> - Is this a Lenovo system with 'Boot Order Lock' turned on in the BIOS

No, it's an ASUS motherboard in a desktop PC.

> ?

Even if this turns out to be a big nothing burger, I thought it would be helpful for others encountering this scenario to document my findings 🔢

superm1, 2022-10-31:
Would it be possible to try main?  I'm wondering if this is the same as https://github.com/fwupd/fwupd/commit/5fcfe7f0fc8cb836a93713cae34f5ba64af8ee69 or https://github.com/fwupd/fwupd/commit/eec6de6985f6d4a8d1d758fb7c8863c1f92b02df

But to confirm we'd need to see it on master.

dlehammer, 2022-10-31:
Hi @superm1,

Thanks for a quick reply :)

Based on the comments in the linked commits, the dbx "versioning" seems like quite the quagmire.

Regarding "version" 267, the number of checksums from the following expression seems to match.

```shell
$ mokutil --dbx | grep '  ' | grep -v \\[ | wc -l
267
```

And the [217 known checksum](https://github.com/fwupd/fwupd/commit/eec6de6985f6d4a8d1d758fb7c8863c1f92b02df#diff-b49d40e1b326a674ebd765d737e162b8aa080ca0c5929f2998306d81318b782eR218) present in eec6de6985f6d4a8d1d758fb7c8863c1f92b02df is also present.

```shell
$ mokutil --dbx | grep 90aec5c4995674a849c1d1384463f3b02b5aa625a5c320fc4fe7d9bb58a62398
  90aec5c4995674a849c1d1384463f3b02b5aa625a5c320fc4fe7d9bb58a62398
```

I have no further questions, so this can be closed as "working as intended" unless you have some change in mind :)

closed 2022-10-31 by superm1