fwupd/fwupd #5222: CI Build for Ubuntu failing

Sean-StarLabs, 2022-10-31:
Since 4f3105369ad5c6cf6815d610f87fe8febd83ce87, the CI builds fail as the helper scripts don't run successfully - notably, `git` is no longer a build-dep.

jammy:
```
+ ./contrib/ci/fwupd_setup_helpers.py test-meson
meson must be installed/upgraded
/usr/bin/python3: No module named pip
+ ./contrib/ci/fwupd_setup_helpers.py test-markdown
python3-markdown must be installed/upgraded
/usr/bin/python3: No module named pip
+ . ./contrib/ci/get_test_firmware.sh
++ '[' true = true ']'
++ rm -rf fwupd-test-firmware
++ git clone https://github.com/fwupd/fwupd-test-firmware
./contrib/ci/get_test_firmware.sh: line 6: git: command not found
```

kinetic:
```
dpkg-buildpackage: info: source package fwupd
dpkg-buildpackage: info: source version 9
dpkg-buildpackage: info: source distribution kinetic
dpkg-buildpackage: info: source changed by Sean Rhodes <sean@starlabs.systems>
 dpkg-source --before-build .
dpkg-source: info: using options from build/debian/source/options: --extend-diff-ignore=.vscode|venv|subprojects|build
dpkg-checkbuilddeps: error: Unmet build dependencies: fwupd-unsigned-dev libqrtr-glib-dev
dpkg-buildpackage: warning: build dependencies/conflicts unsatisfied; aborting
```

superm1, 2022-10-31:
I think the problem is you're missing python3-pip when you start.

Sean-StarLabs, 2022-11-01:
So move `./contrib/ci/fwupd_setup_helpers.py install-pip --yes -o ubuntu` above `./contrib/ci/fwupd_setup_helpers.py install-dependencies --yes -o ubuntu`?

hughsie, 2023-02-19:
I think it works now. Please reopen if not so.

closed 2023-02-19 by hughsie